hbldh/bleak #832: Discovery Stop Failed

amunozh, 2022-05-19:
* bleak version: 0.14.3
* Python version: 3.7.13
* Operating System: Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description
I'm developing an arm32v7 linux container that continuosly scan BLE advertisement and reads the manufacturer data. This container runs on top of a Raspberry Pi 3B, while there are other Bluetooth devices that advertises some information using the manufacturer data. 

### What I Did

To discover the devices I'm using this async method that is executed as a task (I've added some prints only for debugging):
`
async def start_scanning(self):
        print("start scanning task...")
        i = 0
        while not self.stop_ble_manager:
            try:
                print(f"scanning loop number {i}")
                scanner = BleakScanner()
                scanner.register_detection_callback(self.handler)
                await scanner.start()
                print("Scanning started")
                await asyncio.sleep(5.0)
                await scanner.stop()
                print("Scanning finished")
                i+=1
`
Sometimes, I realized the app does not found more devices around. When digging into this problem I discovered using btmon that the "Stop Discovery" command returns "Failed". After failing, everytime the app starts the discovery I can see in the "btmon" that the command "Starts Discovery" returns "Rejected". 
The main problem is that I'm not getting any error or exception on the app. 

(To solve the problem I have to reboot the Raspberry)

dlech, 2022-05-19:
If rebooting the RPi solves the problem, then the issue is probably with BlueZ rather than Bleak.

dlech, 2022-05-19:
I would try looking at D-Bus messages too. If BlueZ doesn't send the error over D-Bus, then there isn't a way for Bleak to know about the error.

amunozh, 2022-05-20:
> I would try looking at D-Bus messages too. If BlueZ doesn't send the error over D-Bus, then there isn't a way for Bleak to know about the error.

How can I sniff the D-Bus message?

amunozh, 2022-05-20:
``I achieved to sniff dbus messages while the error occured. I'm not an expert on these messages, I would appreciate some help to understand whats happening.

**D-Bus messages (destination: org.bluez)**

```
method return time=1653034702.958350 sender=org.freedesktop.DBus -> destination=:1.5 serial=119 reply_serial=1051
method call time=1653034708.037072 sender=:1.93 -> destination=org.bluez serial=5 path=/; interface=org.freedesktop.DBus.ObjectManager; member=GetManagedObjects
method call time=1653034708.202605 sender=:1.93 -> destination=org.bluez serial=6 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=SetDiscoveryFilter
   array [
      dict entry(
         string "Transport"
         variant             string "le"
      )
   ]
method return time=1653034708.203649 sender=org.freedesktop.DBus -> destination=:1.5 serial=120 reply_serial=1054
method call time=1653034708.208330 sender=:1.93 -> destination=org.bluez serial=7 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=StartDiscovery
method call time=1653034713.219605 sender=:1.93 -> destination=org.bluez serial=8 path=/org/bluez/hci0; interface=org.bluez.Adapter1; member=StopDiscovery
method return time=1653034713.221193 sender=org.freedesktop.DBus -> destination=:1.5 serial=121 reply_serial=1057
```

**D-Bus messages (sender: org.bluez)**
```
method call time=1653034841.235722 sender=:1.5 -> destination=org.freedesktop.DBus serial=1132 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type='signal',sender='org.freedesktop.DBus',path='/org/freedesktop/DBus',interface='org.freedesktop.DBus',member='NameOwnerChanged',arg0=':1.106'"
method return time=1653034841.236119 sender=:1.5 -> destination=:1.106 serial=1133 reply_serial=6
method return time=1653034841.241559 sender=:1.5 -> destination=:1.106 serial=1134 reply_serial=7
method call time=1653034846.256985 sender=:1.5 -> destination=org.freedesktop.DBus serial=1135 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=RemoveMatch
   string "type='signal',sender='org.freedesktop.DBus',path='/org/freedesktop/DBus',interface='org.freedesktop.DBus',member='NameOwnerChanged',arg0=':1.106'"
method return time=1653034846.258144 sender=:1.5 -> destination=:1.106 serial=1136 reply_serial=8
method return time=1653034851.348820 sender=:1.5 -> destination=:1.107 serial=1137 reply_serial=5
   array [
      dict entry(
         object path "/org/bluez"
         array [
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.bluez.AgentManager1"
               array [
               ]
            )
            dict entry(
               string "org.bluez.ProfileManager1"
               array [
               ]
            )
            dict entry(
               string "org.bluez.HealthManager1"
               array [
               ]
            )
         ]
      )
      dict entry(
         object path "/org/bluez/hci0"
         array [
            dict entry(
               string "org.freedesktop.DBus.Introspectable"
               array [
               ]
            )
            dict entry(
               string "org.bluez.Adapter1"
               array [
                  dict entry(
                     string "Address"
                     variant                         string "B8:27:EB:C3:CD:C3"
[... 4704 lines cut ...]
                     variant                         array [
                           dict entry(
                              uint16 6
                              variant                                  array of bytes [
                                    01 09 20 02 e4 d5 80 1b 32 5c a3 a8 da ad
                                    2d 84 e3 86 79 5d 5d 1d 8c 77 18 04 d8
                                 ]
                           )
                        ]
                  )
                  dict entry(
                     string "ServicesResolved"
                     variant                         boolean false
                  )
               ]
            )
            dict entry(
               string "org.freedesktop.DBus.Properties"
               array [
               ]
            )
         ]
      )
   ]
```

dlech, 2022-05-20:
It looks like the log is cut short and we don't see the replies from BlueZ after the long list of properties. It would be helpful to log both at the same time so we can easily see the replies in context with the requests.

Maybe something like this?

    sudo dbus-monitor --system "path_namespace='/org/bluez'" "sender='org.bluez'"

dlech, 2022-09-12:
Closing since the problem appears to be with BlueZ and not Bleak. Upgrading BlueZ may help solve the problem.

closed 2022-09-12 by dlech