hbldh/bleak #837: Unable to write/notify on mac M1

arnaud-palgen, 2022-06-02:
* bleak version: 0.14.3
* Python version: 3.10.4
* Operating System: macOS Montery Version 12.4
* BlueZ version (`bluetoothctl -v`) in case of Linux: / 

### Description

I recently switched to a macbook pro with an Apple M1 Pro and have been having problems with Bleak ever since:

- impossible to notify to a characteristic
- Impossible to write on a characteristic
- I don't feel like I have any problems reading

### What I Did

I work with a sensorTag. So I tried the script in the examples. And the test made at the end of the code confirms that it is impossible to write a value in a characteristic.

To be sure that the problem does not come from the sensorTag, I ran the same example on a raspberry pi and everything works. 
(The python version on the raspberry is 3.9.2.)

Here are the logs of the execution of the `sensortag.py` example on my mac:

```
2022-06-02 12:56:40,594 bleak.backends.corebluetooth.client DEBUG: Retrieving services...
2022-06-02 12:56:40,649 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverServices_
2022-06-02 12:56:40,650 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Services discovered
2022-06-02 12:56:40,650 bleak.backends.corebluetooth.client DEBUG: Retrieving characteristics for service 180A
2022-06-02 12:56:40,651 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverCharacteristicsForService_error_
2022-06-02 12:56:40,651 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Characteristics discovered
2022-06-02 12:56:40,651 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A23
2022-06-02 12:56:40,651 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,652 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 10
2022-06-02 12:56:40,652 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A24
2022-06-02 12:56:40,652 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,652 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 12
2022-06-02 12:56:40,653 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A25
2022-06-02 12:56:40,653 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,653 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 14
2022-06-02 12:56:40,653 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A26
2022-06-02 12:56:40,653 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,653 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 16
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A27
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 18
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A28
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 20
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A29
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,654 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 22
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A2A
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 24
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A50
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 26
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.client DEBUG: Retrieving characteristics for service 180F
2022-06-02 12:56:40,655 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverCharacteristicsForService_error_
2022-06-02 12:56:40,656 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Characteristics discovered
2022-06-02 12:56:40,656 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic 2A19
2022-06-02 12:56:40,656 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,656 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 29
2022-06-02 12:56:40,657 bleak.backends.corebluetooth.client DEBUG: Retrieving characteristics for service F000AA00-0451-4000-B000-000000000000
2022-06-02 12:56:40,657 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverCharacteristicsForService_error_
2022-06-02 12:56:40,657 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Characteristics discovered
2022-06-02 12:56:40,657 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic F000AA01-0451-4000-B000-000000000000
2022-06-02 12:56:40,658 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
2022-06-02 12:56:40,658 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Descriptor discovered 35
2022-06-02 12:56:40,658 bleak.backends.corebluetooth.client DEBUG: Retrieving descriptors for characteristic F000AA02-0451-4000-B000-000000000000
2022-06-02 12:56:40,658 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didDiscoverDescriptorsForCharacteristic_error_
[... 57 lines cut ...]
2022-06-02 12:56:40,904 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2022-06-02 12:56:40,904 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Read characteristic value
2022-06-02 12:56:40,904 bleak.backends.corebluetooth.client DEBUG: Read Characteristic 00002a19-0000-1000-8000-00805f9b34fb : bytearray(b'S')
Battery Level: 83%
2022-06-02 12:56:40,933 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2022-06-02 12:56:40,933 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Read characteristic value
2022-06-02 12:56:40,934 bleak.backends.corebluetooth.client DEBUG: Read Characteristic f000aa65-0451-4000-b000-000000000000 : bytearray(b'\x7f')
I/O Data Pre-Write Value: bytearray(b'\x7f')
2022-06-02 12:56:40,934 bleak.backends.corebluetooth.client DEBUG: Write Characteristic f000aa65-0451-4000-b000-000000000000 : bytearray(b'\xa0')
2022-06-02 12:56:40,963 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_
2022-06-02 12:56:40,963 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Read characteristic value
2022-06-02 12:56:40,963 bleak.backends.corebluetooth.client DEBUG: Read Characteristic f000aa65-0451-4000-b000-000000000000 : bytearray(b'\x7f')
I/O Data Post-Write Value: bytearray(b'\x7f')
2022-06-02 12:56:40,963 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDisconnectPeripheral_error_
2022-06-02 12:56:40,963 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Peripheral Device disconnected!
Traceback (most recent call last):
  File "path/to/bleak-repo/bleak/examples/sensortag.py", line 154, in <module>
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "path/to/bleak-repo/bleak/examples/sensortag.py", line 146, in main
    assert value == write_value
AssertionError```

dlech, 2022-06-02:
From the logs, we can see that writing and reading in Bleak are working. I'm not familiar with this device, so I'm not sure if the value read from this characteristic is always supposed to match the value written or not.

arnaud-palgen, 2022-06-02:
> From the logs, we can see that writing and reading in Bleak are working. I'm not familiar with this device, so I'm not sure if the value read from this characteristic is always supposed to match the value written or not.

what is the meaning of error lines like this one ?
`2022-06-02 12:56:40,933 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateValueForCharacteristic_error_`

According to the documentation, the value read from this characteristic is supposed to match the value written

arnaud-palgen, 2022-06-07:
I did some tests with the `sensortag.py` example and the application I'm working on which uses bleak. 

On raspberry pi no problem.
But on my mac, I used the _packetLogger_ application and I notice that no packet is sent for writing when `response=False`. On the other hand when `response=True` the example and my application works (i.e. the packet to write a data is sent).

There is no problem for reading on my mac and on the raspberry pi.

dlech, 2022-06-07:
> But on my mac, I used the _packetLogger_ application and I notice that no packet is sent for writing when `response=False`.

Interesting... in CoreBluetooth, there is a [canSendWriteWithoutResponse](https://developer.apple.com/documentation/corebluetooth/cbperipheral/2891512-cansendwritewithoutresponse) API that we haven't made use of so far. I wonder if it returns false in your case and that is why the packet is not sent. 

Could you try making the change below to your local install of Bleak on your Mac to see if this is the case?

```diff
diff --git a/bleak/backends/corebluetooth/client.py b/bleak/backends/corebluetooth/client.py
index 4431a37..aa9a84c 100644
--- a/bleak/backends/corebluetooth/client.py
+++ b/bleak/backends/corebluetooth/client.py
@@ -315,6 +315,9 @@ class BleakClientCoreBluetooth(BaseBleakClient):
         if not characteristic:
             raise BleakError("Characteristic {} was not found!".format(char_specifier))
 
+        if not self._delegate.peripheral.canSendWriteWithoutResponse():
+            raise RuntimeError("just as we suspected")
+
         value = NSData.alloc().initWithBytes_length_(data, len(data))
         await self._delegate.write_characteristic(
             characteristic.obj,
```

arnaud-palgen, 2022-06-08:
So I added this test and the exception is not launched. (I also displayed the value of `self._delegate.peripheral.canSendWriteWithoutResponse()` and it is `True`. 

To be sure of my manipulations and to bring more information, here are the commands that I made followed by the packet capture in the case of `response=False` and `response=True`:

# `sensortag.py` with `response=False`
## 1. console
```console
$ cd bleak
$ git pull
Already up to date.
$ git diff
diff --git a/bleak/backends/corebluetooth/client.py b/bleak/backends/corebluetooth/client.py
index 4431a37..aa9a84c 100644
--- a/bleak/backends/corebluetooth/client.py
+++ b/bleak/backends/corebluetooth/client.py
@@ -315,6 +315,9 @@ class BleakClientCoreBluetooth(BaseBleakClient):
         if not characteristic:
             raise BleakError("Characteristic {} was not found!".format(char_specifier))
 
+        if not self._delegate.peripheral.canSendWriteWithoutResponse():
+            raise RuntimeError("just as we suspected")
+
         value = NSData.alloc().initWithBytes_length_(data, len(data))
         await self._delegate.write_characteristic(
             characteristic.obj,
$ pew mktmpenv
created virtual environment CPython3.10.4.final.0-64 in 184ms
This is a temporary environment. It will be deleted when you exit
$ pip install .
$ pip freeze | grep bleak
bleak @ file:///path/to/bleak/repo
$ cd example
$ python sensortag.py <DEVICE_ADDR>
python sensortag.py <DEVICE_ADDR>
2022-06-08 09:35:28,551 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManagerDidUpdateState_
2022-06-08 09:35:28,551 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
2022-06-08 09:35:28,552 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
2022-06-08 09:35:28,584 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 09:35:28,584 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 09:35:28,584 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device <ADDR>: None @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600001cd0400>
2022-06-08 09:35:28,584 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device <ADDR>: None @ RSSI: -75 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataTimestamp', 'kCBAdvDataManufacturerData', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataRxPrimaryPHY'])>) and Central: <CBCentralManager: 0x600001cd0400>
2022-06-08 09:35:28,595 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 09:35:28,596 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device <ADDR>: 310 @ RSSI: -94 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600001cd0400>
2022-06-08 09:35:28,900 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 09:35:28,904 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device <ADDR>: 310 @ RSSI: -98 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600001cd0400>
2022-06-08 09:35:29,491 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 09:35:29,492 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 09:35:29,493 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device <DEVICE_ADDR>: CC2650 SensorTag @ RSSI: -47 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600001cd0400>
2022-06-08 09:35:29,494 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device <DEVICE_ADDR>: CC2650 SensorTag @ RSSI: -47 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600001cd0400>
2022-06-08 09:35:29,494 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
2022-06-08 09:35:29,495 bleak.backends.corebluetooth.client DEBUG: CentralManagerDelegate  at <CentralManagerDelegate: 0x126726c50>
2022-06-08 09:35:29,495 bleak.backends.corebluetooth.client DEBUG: Connecting to BLE device @ <DEVICE_ADDR>
2022-06-08 09:35:29,676 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didConnectPeripheral_
[... cut ...]
Jun 08 09:43:55.866  HCI Event        0x005E  <DEVICE_ADDR>  Number Of Completed Packets - Handle: 0x005E - Packets: 0x0001    
Jun 08 09:43:58.913  ATT Send         0x005E  <DEVICE_ADDR>  Write Request - Handle:0x004D - Value: 0000  
Jun 08 09:43:58.949  ATT Receive      0x005E  <DEVICE_ADDR>  Write Response  
Jun 08 09:43:58.953  HCI Command      0x005E  <DEVICE_ADDR>  Disconnect - Connection Handle: 0x005E - Reason: Remote User Terminated Connection  
Jun 08 09:43:58.966  HCI Event        0x005E  <DEVICE_ADDR>  Disconnection Complete - Success  
```

dlech, 2022-06-08:
Thanks for all of the info. We could just make that change to the example and call it good. But if we really want to understand what is going on some things I am curious about:

- If we add a delay after calling the write method, does it eventually send a `Write Command`?
- Does the characteristic being written to have both property flags set for `write` and `write-without-response`? We can see these flags by running `python examples/service_explorer.py <device id>` where `<device id>` is the UUID listed for the device by `python examples/detection_callback.py`.

orlyprofili, 2024-03-26:
@dlech 
I encountered this issue, works on Windows, fails on Mac (Sonoma 14.4.1, M2 silicon).  Could receive from device, but a send could not be observed.

I changed from response=False to True and it worked:

`await client.write_gatt_char(tx_size, s, response=True)`

below is the result of service_explorer.py for this device, does this reveal anything?

```
 py examples/service_explorer.py --address 779196B7-2C82-636C-68CA-1B0957D631D2
2024-03-25 22:44:52,915 __main__ INFO: starting scan...
2024-03-25 22:44:53,828 __main__ INFO: connecting to device...
2024-03-25 22:44:55,241 __main__ INFO: connected
2024-03-25 22:44:55,241 __main__ INFO: [Service] 0000fe59-0000-1000-8000-00805f9b34fb (Handle: 14): Nordic Semiconductor ASA
2024-03-25 22:44:55,241 __main__ INFO:   [Characteristic] 8ec90003-f315-4f60-9fb8-838830daea50 (Handle: 15): Buttonless DFU (write,indicate)
2024-03-25 22:44:55,299 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration, Value: bytearray(b'')
2024-03-25 22:44:55,299 __main__ INFO: [Service] 00001623-1212-efde-1523-785feabcd124 (Handle: 18): Unknown
2024-03-25 22:44:55,359 __main__ INFO:   [Characteristic] 00001625-1212-efde-1523-785feabcd124 (Handle: 19): Unknown (read,notify), Value: bytearray(b'\x08\x01\x92\x02\t\x085\x10\x01\x1d=\x8aq@')
2024-03-25 22:44:55,419 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 21): Client Characteristic Configuration, Value: bytearray(b'')
2024-03-25 22:44:55,479 __main__ INFO:   [Characteristic] 00001627-1212-efde-1523-785feabcd124 (Handle: 22): Unknown (read,notify), Value: bytearray(b'\x0e\x00\x00\x00\x00\x00\x00\x00')
2024-03-25 22:44:55,538 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 24): Client Characteristic Configuration, Value: bytearray(b'')
2024-03-25 22:44:55,597 __main__ INFO:   [Characteristic] 00001626-1212-efde-1523-785feabcd124 (Handle: 25): Unknown (read,write), Value: bytearray(b'\x05\x00\x00\x00')
2024-03-25 22:44:55,657 __main__ INFO:   [Characteristic] 00001624-1212-efde-1523-785feabcd124 (Handle: 27): Unknown (read,write), Value: bytearray(b'\x08\x01\x8a\x02\x00')
2024-03-25 22:44:55,657 __main__ INFO: disconnecting...
2024-03-25 22:44:55,658 __main__ INFO: disconnected
```

dlech, 2024-03-26:
> does this reveal anything?

I only see characteristics with the `write` property, so it is expected that `response=True` works and `response=False` doesn't work.