hbldh/bleak #840: Failed to read characteristic 18

snowuyl, 2022-06-08:
* bleak version: 0.15.0a1
* Python version: 3.9.12
* Operating System: macOS 12.4 M1 (ARM64)

### Description
[main.zip](https://github.com/hbldh/bleak/files/8857802/main.zip)

% python3 main.py
ANCS1
2956B0F4-976D-4E07-CE53-BCA1F475F44A
main.py:102: FutureWarning: is_connected has been changed to a property. Calling it as an async method will be removed in a future version
  x = await client.is_connected()
HRV manual
M. Measure
S. Stop measure
R. Reading result
Q. Quit

Enter an option: M
HRV manual
M. Measure
S. Stop measure
R. Reading result
Q. Quit

Enter an option: S
HRV manual
M. Measure
S. Stop measure
R. Reading result
Q. Quit

Enter an option: R
HRV manual
M. Measure
S. Stop measure
R. Reading result
Q. Quit

Enter an option: Failed to read characteristic 18: Error Domain=CBATTErrorDomain Code=2 "Reading is not permitted." UserInfo={NSLocalizedDescription=Reading is not permitted.} q

dlech, 2022-06-08:
contents of main.zip:

```python
import asyncio
import logging
import platform
import sys
import threading
import time
from bleak       import BleakClient
from bleak       import BleakClient
from bleak       import BleakScanner
from bleak       import _logger as logger
from bleak.uuids import uuid16_dict
from state       import State
from strings     import Strings

BATTERY_LEVEL_UUID = "00002a19-0000-1000-8000-00805f9b34fb" # battery level characteristic
UART_TX_UUID       = "6e400002-b5a3-f393-e0a9-e50e24dcca9e" # Nordic NUS characteristic for TX
UART_RX_UUID       = "6e400003-b5a3-f393-e0a9-e50e24dcca9e" # Nordic NUS characteristic for RX

data_flag    = False #global flag to check for new data
quit_flag    = False
connect_flag = False
command      = ""
state        = State.STATE_UNKNOWN

def display_manual():
    print("""HRV manual
M. Measure
S. Stop measure
R. Reading result
Q. Quit
""")

def measure(bt_mac_address):
    global command
    command = "START"
    #thread  = threading.Thread(target=thread_function, args=(bt_mac_address, ))
    #thread.start()

def stop_measure(bt_mac_address):
    global command
    command = "STOP"
    #time.sleep(0.01)
    #thread = threading.Thread(target=thread_function, args=(bt_mac_address, ))
    #thread.start()

def read_result(bt_mac_address):
    global command
    command = "READ"
    #thread = threading.Thread(target=thread_function, args=(bt_mac_address, ))
    #thread.start()

def quit_program():
    global quit_flag
    quit_flag = True
    sys.exit(0)

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    #print("{0}: {1}".format(sender, data))
    global command
    if (command == "READ"):
        file = open('ppg_gsensor.dat', 'ab')
        file.write(data)
        file.close()
        global data_flag
        data_flag = True

[... 123 lines cut ...]
    while True:
            display_manual()
            choice = input("Enter an option: ")
            choice = choice.upper()
            if choice == "M":
                measure(bt_mac_address)
            elif choice == "S":
                stop_measure(bt_mac_address)
            elif choice == "R":
                read_result(bt_mac_address)
            elif choice == "Q":
                quit_program()
            else:
                print("{0} is not a valid choice".format(choice))

    #loop = asyncio.new_event_loop()
    #asyncio.set_event_loop(loop)
    #loop.run_until_complete(run_uart(bt_mac_address, loop))

    #thread = threading.Thread(target=thread_function, args=(bt_mac_address,))
    #thread.start()
    

    thread.join()
```

dlech, 2022-06-08:
> Enter an option: Failed to read characteristic 18: Error Domain=CBATTErrorDomain Code=2 "Reading is not permitted." UserInfo={NSLocalizedDescription=Reading is not permitted.} q

This is an error that comes from the OS. Presumably it is telling the truth and you are trying to read from a characteristic that does not have the `read` property flag set. So the solution is probably to just not read this characteristic.

closed 2025-05-16 by dlech