hbldh/bleak #841: Problem with start_notify, bleak

Egidio92, 2022-06-08:
* bleak version: 0.14.3
* Python version: 3.9.12
* Operating System: macOS 10.15.7

 Hi! I have to read the weight from a Mi Smart Scale 2 scale via Mac.

The weight is encoded in a BLE characteristic. However, sometimes it reads the weight, other times it doesn't.

I'm using the code found in the Bleak repository examples and this is my code:

import sys
import asyncio
import platform

from bleak import BleakClient

CHARACTERISTIC_UUID = "00002a9d-0000-1000-8000-00805f9b34fb"
ADDRESS = (
    "70:87:9e:0f:f8:7d"
    if platform.system() != "Darwin"
    else "775A199B-A238-4AB7-86EB-4EFF8C404F9A"
)

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    data1 = list(data)
    data2 = ((data1[1]+data1[2]*256)*0.005)
    print("{0}: {1}".format(sender, data2))

async def main(address, char_uuid):
    async with BleakClient(address) as client:
        print(f"Connected: {client.is_connected}")

       
        await client.start_notify(char_uuid, notification_handler)
        await asyncio.sleep(30.0)
        await client.stop_notify(char_uuid)
        

if __name__ == "__main__":
    asyncio.run(
        main(
            sys.argv[1] if len(sys.argv) > 1 else ADDRESS,
            sys.argv[2] if len(sys.argv) > 2 else CHARACTERISTIC_UUID,
        )
    )

dlech, 2022-06-08:
Have you tried the troubleshooting recommendations in the Bleak docs (enable logging, capturing Bluetooth packets)?

Egidio92, 2022-06-08:
Thanks for the quick response. I tried to install Xcode to acquire bluetooth packets but I don't have enough memory space ... Instead from the logs I get:

2022-06-08 23:55:11,255 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManagerDidUpdateState_
2022-06-08 23:55:11,256 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Bluetooth powered on
2022-06-08 23:55:11,308 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: 'isScanning' changed
2022-06-08 23:55:11,467 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:11,468 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device A70B7C91-2959-4105-9A26-2E7E5875B04E: None @ RSSI: -93 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:11,484 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:11,485 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:11,486 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 69D6FD30-01D0-4310-AEDF-B7D91EB02850: None @ RSSI: -91 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:11,486 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 69D6FD30-01D0-4310-AEDF-B7D91EB02850: None @ RSSI: -91 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:11,773 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:11,774 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 8D1CE81E-1EBD-4BBA-A0DD-B4DA13AE7A01: None @ RSSI: -94 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:12,650 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:12,651 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:12,652 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device ACF7BD93-CAD1-427E-B056-A5CE4CEAA28F: None @ RSSI: -36 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:12,652 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device ACF7BD93-CAD1-427E-B056-A5CE4CEAA28F: None @ RSSI: -36 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:12,949 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:12,951 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 61F94717-1906-4C8A-BED8-6C9017021397: None @ RSSI: -36 (kCBAdvData <nsdict_keys(['kCBAdvDataServiceData', 'kCBAdvDataChannel', 'kCBAdvDataServiceUUIDs', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:12,962 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:12,962 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:12,963 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device D7888FE2-72E6-4407-94F3-2EB5CC43F761: None @ RSSI: -90 (kCBAdvData <nsdict_keys(['kCBAdvDataAppleMfgData', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData', 'kCBAdvDataTxPowerLevel', 'kCBAdvDataIsConnectable'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:12,964 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device D7888FE2-72E6-4407-94F3-2EB5CC43F761: None @ RSSI: -92 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:14,201 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
2022-06-08 23:55:14,202 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Discovered device 2F11A1E4-E957-4A02-9A4D-03BBA7E6D23A: None @ RSSI: -92 (kCBAdvData <nsdict_keys(['kCBAdvDataIsConnectable', 'kCBAdvDataChannel', 'kCBAdvDataManufacturerData'])>) and Central: <CBCentralManager: 0x7fedf9cc8180>
2022-06-08 23:55:14,204 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDiscoverPeripheral_advertisementData_RSSI_
[... cut ...]
2022-06-08 23:55:18,613 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Character Notify Update
2022-06-08 23:55:48,644 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: peripheral_didUpdateNotificationStateForCharacteristic_error_
2022-06-08 23:55:48,644 bleak.backends.corebluetooth.PeripheralDelegate DEBUG: Character Notify Update
2022-06-08 23:55:48,687 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: centralManager_didDisconnectPeripheral_error_
2022-06-08 23:55:48,687 bleak.backends.corebluetooth.CentralManagerDelegate DEBUG: Peripheral Device disconnected!

Egidio92, 2022-06-08:
On the other hand, if I launch from spyder without enable logging, I get:

Connected: True
Traceback (most recent call last):
  File "/Users/Egidio/Documents/Python/Tesi/BLE/BLE17.py", line 42, in <module>
    asyncio.run(
  File "/Users/Egidio/opt/anaconda3/envs/BLE/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Users/Egidio/opt/anaconda3/envs/BLE/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/Users/Egidio/Documents/Python/Tesi/BLE/BLE17.py", line 37, in main
    await client.stop_notify(char_uuid)
  File "/Users/Egidio/opt/anaconda3/envs/BLE/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 406, in stop_notify
    await self._delegate.stop_notifications(characteristic.obj)
  File "/Users/Egidio/opt/anaconda3/envs/BLE/lib/python3.9/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 230, in stop_notifications
    await future
bleak.exc.BleakError: disconnected
logout
Saving session...
...copying shared history...
...saving history...truncating history files...

Egidio92, 2022-06-08:
I think the problem is related to caches or in general to the saving session... What do you think about it? 
Thanks again

dlech, 2022-06-08:
> tried to install Xcode to acquire bluetooth packets but I don't have enough memory space

You don't need to install XCode, just the hardware utilities from the apple developer website

> I think the problem is related to caches or in general to the saving session... What do you think about it?

If the device is listed in the Bluetooth system settings, try removing it there.

paulosergiosantos, 2022-08-26:
Hi @Egidio92 ... I'm with the same problem, the code run on windows but on mac os, when I call the start notify for the connect BleClient, I receive the same error that you: ... await Future ... 

Did you discover or had a workaround for that error?

Thanks in an advance...

alexanderturner, 2023-09-23:
start_notify appears to try to open a SPP in MacOS instead of subscription to a GATT notification. What I'm seeing on an ESP32

```I (66891) BLE: ESP_GATTS_CONNECT_EVT, conn_id = 0
I (66892) BLE: 8c 85 90 7c 61 27
I (66966) BLE: ESP_GATTS_MTU_EVT, MTU 500
I (67126) BLE: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
W (68125) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (68126) BLE: ESP_GATTS_DISCONNECT_EVT, reason = 0x13
I (68132) BLE: Advertising start successfully
```

and bleak
```
bleak.exc.BleakError: Failed to update the notification status for characteristic 58: Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}
```

julio-liriano, 2024-08-28:
I dusted off an old nRF dongle and confirmed that it gets notification updates from my ESP32 dev board. This problem also occurs with the simpleblepy module. Must be a Mac OS thing.

DeHuilendeHacker, 2025-06-16:
I would like to add that this problem is reproducable with the example:
(https://github.com/hbldh/bleak/blob/develop/examples/enable_notifications.py)
when connecting to an MCUMGR characteristic (that does report notify when printing out the characteristics).

However I'm not on MacOS, I'm on Ubuntu 25.04 with BlueZ 5.79.
Let me know if I should provide more info than the traceback, then I will update the issue here.

`/home/jeroen/pythonbletool/bleak_enable_notifications.py DA2E7828-FBCE-4E01-AE9E-261174997C48 --name TEST_DEVICE
2025-06-16 12:13:08,546 __main__ INFO: starting scan...
2025-06-16 12:13:08,668 __main__ INFO: connecting to device...
2025-06-16 12:13:10,182 __main__ INFO: Connected
Traceback (most recent call last):
  File "/home/jeroen/pythonbletool/bleak_enable_notifications.py", line 109, in <module>
    asyncio.run(main(args))
    ~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/lib/python3.13/asyncio/runners.py", line 195, in run
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "/usr/lib/python3.13/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/lib/python3.13/asyncio/base_events.py", line 719, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/home/jeroen/pythonbletool/bleak_enable_notifications.py", line 61, in main
    await client.start_notify(args.characteristic, notification_handler)
  File "/home/jeroen/pythonbletool/.venv/lib/python3.13/site-packages/bleak/__init__.py", line 844, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "/home/jeroen/pythonbletool/.venv/lib/python3.13/site-packages/bleak/backends/bluezdbus/client.py", line 951, in start_notify
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/home/jeroen/pythonbletool/.venv/lib/python3.13/site-packages/dbus_fast/aio/message_bus.py", line 398, in call
    await future
  File "src/dbus_fast/aio/message_reader.py", line 23, in dbus_fast.aio.message_reader._message_reader
  File "src/dbus_fast/_private/unmarshaller.py", line 868, in dbus_fast._private.unmarshaller.Unmarshaller._unmarshall
  File "src/dbus_fast/_private/unmarshaller.py", line 732, in dbus_fast._private.unmarshaller.Unmarshaller._read_header
  File "src/dbus_fast/_private/unmarshaller.py", line 453, in dbus_fast._private.unmarshaller.Unmarshaller._read_to_pos
  File "src/dbus_fast/_private/unmarshaller.py", line 393, in dbus_fast._private.unmarshaller.Unmarshaller._read_sock_with_fds
EOFError`

dlech, 2025-06-16:
`EOFError` on Linux sounds like #409 or #1151. This issue is about macOS, so can't be the same issue.