hbldh/bleak #842: Notification callback is not called continuously

jaekwonKim, 2022-06-10:
* bleak version: 0.14.3
* Python version: 3.8
* Operating System: Window 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

I try to get notification continuously. This means stop_notify not called. Only called start_notify.
Device send every 100ms to buffered data(It mostly 150 bytes).
Most of times callback is not called. In this case, try to start notification again, callback is called some of times and stop again.

Device MCU is nRF52833 so I can check notification by nRF52840 USB Dongle. Device send all notification.

How can I get notification continuously??

dlech, 2022-06-10:
I would try logging Bluetooth packets to see if the notifications are getting to the Bluetooth stack on Windows. See the troubleshooting section of the docs for instructions on how to do this.

Xin-Lee, 2024-08-08:
Hi, have you solved the problem? I also find this problem and I don't know what to do.