hbldh/bleak #843: Can't connect to Bluetooth device before nRF Connect connecting and then disconnecting Bluetooth device

snowuyl, 2022-06-10:
* bleak version: 0.15.0a1
* Python version: 3.10.5
* Operating System: Windows 10

### Description
python3 .\main.py
Waiting connecting to ANCS1
E8:20:9F:75:90:08
Exception in thread Thread-1 (thread_function):
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\lib\asyncio\locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\lib\asyncio\tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\lib\threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "D:\main.py", line 173, in thread_function
    loop.run_until_complete(run_uart(bt_mac_address, loop))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "D:\main.py", line 99, in run_uart
    async with BleakClient(address, loop=loop) as client:
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 268, in connect
    await asyncio.wait_for(event.wait(), timeout=timeout)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\lib\asyncio\tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
Traceback (most recent call last):
  File "D:\andy\workspace_Python3_Win10\HrvPy\main.py", line 190, in <module>
    time.sleep(0.01)
KeyboardInterrupt

After using nRF connect APP to connect and then disconnct Bluetooth device. Bleak can connect to Bluetooth device.
PS D:\andy\workspace_Python3_Win10\HrvPy> python3 .\main.py
Waiting connecting to ANCS1
E8:20:9F:75:90:08
D:\andy\workspace_Python3_Win10\HrvPy\main.py:102: FutureWarning: is_connected has been changed to a property. Calling it as an async method will be removed in a future version
  x = await client.is_connected()
client.is_connected() True

snowuyl, 2022-06-10:
[main.zip](https://github.com/hbldh/bleak/files/8875441/main.zip)

snowuyl, 2022-06-13:
The following is log from Nordic nRF52832 device.
0> <debug> nrf_ble_gatt: Requesting to update ATT MTU to 185 bytes on connection 0x0.
 0> <info> app: Connected.
 0> <info> app: Disconnected.
 0> <debug> nrf_ble_gatt: Requesting to update ATT MTU to 185 bytes on connection 0x0.
 0> <info> app: Connected.
 0> <debug> nrf_ble_gatt: Peer on connection 0x0 requested an ATT MTU of 527 bytes.
 0> <debug> nrf_ble_gatt: Updating ATT MTU to 185 bytes (desired: 185) on connection 0x0.
 0> <debug> nrf_ble_gatt: ATT MTU updated to 185 bytes on connection 0x0 (response).
 0> <debug> nrf_ble_gatt: Peer on connection 0x0 requested a data length of 251 bytes.
 0> <debug> nrf_ble_gatt: Updating data length to 27 on connection 0x0.
 0> <debug> nrf_ble_gatt: Data length updated to 27 on connection 0x0.
 0> <debug> nrf_ble_gatt: max_rx_octets: 27
 0> <debug> nrf_ble_gatt: max_tx_octets: 27
 0> <debug> nrf_ble_gatt: max_rx_time: 2120
 0> <debug> nrf_ble_gatt: max_tx_time: 2120
 0> <info> app: Disconnected.

dlech, 2022-10-04:
> OSError: [WinError -2147483629] 該物件已關閉。

This is probably a duplicate of #1061

closed 2022-10-04 by dlech