hbldh/bleak #844: Can't connect to Bluetooth device before nRF Connect connecting and then disconnecting Bluetooth device on Ubuntu 20.04

snowuyl, 2022-06-11:
* bleak version: 0.15.0a1
* Python version: 3.8.10
* Operating System: Ubuntu 20.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description
python3 main.py
BleDeviceStruct(name='Nordic_UART', address='EF:CC:A0:02:16:69', rssi=-44)
Traceback (most recent call last):
  File "/home/andyueng/samba/workspace_Python3/HrvPyQt6/worker.py", line 37, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/andyueng/samba/workspace_Python3/HrvPyQt6/main_window.py", line 114, in _connect_device
    self.ble_obj.ble_uart_thread_function(ble_device)
  File "/home/andyueng/samba/workspace_Python3/HrvPyQt6/ble.py", line 178, in ble_uart_thread_function
    loop.run_until_complete(self._run_uart(ble_device, loop))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/andyueng/samba/workspace_Python3/HrvPyQt6/ble.py", line 101, in _run_uart
    async with BleakClient(ble_device.address, loop=loop) as client:
  File "/usr/local/lib/python3.8/dist-packages/bleak-0.15.0a1-py3.8.egg/bleak/backends/client.py", line 61, in __aenter__
    await self.connect()
  File "/usr/local/lib/python3.8/dist-packages/bleak-0.15.0a1-py3.8.egg/bleak/backends/bluezdbus/client.py", line 278, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.8/dist-packages/bleak-0.15.0a1-py3.8.egg/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort

After nRF Connect APP connecting and then disconnecting.
BleDeviceStruct(name='Nordic_UART', address='EF:CC:A0:02:16:69', rssi=-44)
Connected: True

dlech, 2022-07-26:
I do not understand the issue. Can you give a fully reproducible test case?

snowuyl, 2022-07-26:
[PyCmd.zip](https://github.com/hbldh/bleak/files/9185824/PyCmd.zip)
Unzip PyCmd.zip
python3 main.py
This issue isn't occur every time.

closed 2025-11-30 by github-actions[bot]