hbldh/bleak #846: Android app crashes because don't found characteristic

Egidio92, 2022-06-13:
* bleak version: 0.14.3
* Python version: 3.9.12
* Operating System: Android 6.0.1, API 23

I am trying to get this android app to work which is supposed to connect to a Mi Smart Scale 2 scale via low energy Bluetooth. However, even if the app crashes from the logs I see that it connects to the phone.

The app once connected to the phone should read the feature with the start_notify operation but this does not happen.

This is the error that I get, from logcat in android studio:

022-06-13 17:34:33.657 12975-12986/org.test.demo12 D/BluetoothGatt: onClientConnParamsChanged() - Device=70:87:9E:0F:F8:7D interval=39 status=0
2022-06-13 17:34:33.847 12975-13001/org.test.demo12 I/python: Connected: True
2022-06-13 17:34:33.847 12975-13001/org.test.demo12 D/BluetoothGatt: cancelOpen() - device: 70:87:9E:0F:F8:7D
2022-06-13 17:34:33.847 1944-1978/? D/BtGatt.GattService: clientDisconnect() - address=70:87:9E:0F:F8:7D, connId=10
2022-06-13 17:34:33.847 1944-2370/? E/bt_btm: btm_ble_get_search_if search_if=4
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0016
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x0016
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x0016
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=8 connected=0 conn_id=8 reason=0x0016
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=9 connected=0 conn_id=9 reason=0x0016
2022-06-13 17:34:33.847 1944-2370/? W/bt_btif: bta_gattc_conn_cback() - cif=10 connected=0 conn_id=10 reason=0x0016
2022-06-13 17:34:33.857 1944-2267/? D/bt_vendor: op for 7
2022-06-13 17:34:33.857 1944-2267/? D/bt_upio: upio_set : pio 0 action 2, polarity 1
2022-06-13 17:34:33.857 1944-2267/? D/bt_upio: BT_WAKE is asserted already
2022-06-13 17:34:33.857 1944-2370/? E/bt_btif: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 70:87:9e:0f:f8:7d
2022-06-13 17:34:33.857 1944-2113/? D/BtGatt.GattService: onDisconnected() - clientIf=10, connId=10, address=70:87:9E:0F:F8:7D
2022-06-13 17:34:33.857 12975-12987/org.test.demo12 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=70:87:9E:0F:F8:7D
2022-06-13 17:34:33.857 844-2748/? V/AlarmManager:  remove PendingIntent] PendingIntent{c6946a9: PendingIntentRecord{96e10 com.android.bluetooth broadcastIntent}}
2022-06-13 17:34:33.857 844-1752/? V/AlarmManager:  remove PendingIntent] PendingIntent{c10c42e: PendingIntentRecord{96e10 com.android.bluetooth broadcastIntent}}
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 D/BluetoothGatt: close()
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 D/BluetoothGatt: unregisterApp() - mClientIf=10
2022-06-13 17:34:33.867 1944-7183/? D/BtGatt.GattService: unregisterClient() - clientIf=10
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python:  Traceback (most recent call last):
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python:    File "/content/.buildozer/android/app/main.py", line 225, in <module>
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python:    File "/content/.buildozer/android/platform/build-armeabi-v7a/build/other_builds/python3/armeabi-v7a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 616, in run_until_complete
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python:    File "/content/.buildozer/android/app/main.py", line 218, in get_the_weight_from_the_mi_smart_scale_2
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python:    File "/content/.buildozer/android/platform/build-armeabi-v7a/build/python-installs/demo12/armeabi-v7a/bleak/backends/p4android/client.py", line 486, in start_notify
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python:  bleak.exc.BleakError: Characteristic with UUID 00002a9d-0000-1000-8000-00805f9b34fb could not be found!
2022-06-13 17:34:33.867 12975-13001/org.test.demo12 I/python: Python for android ended.
2022-06-13 17:34:33.897 1944-2370/? I/bt_btm_sec: btm_sec_disconnected clearing pending flag handle:64 reason:22

This is the code I converted with Colab:

import asyncio

from kivy.app import App
from kivy.lang.builder import Builder
from kivy.uix.screenmanager import ScreenManager, Screen
from kivy.properties import ObjectProperty

from bleak import BleakClient

kv = '''
WindowManager:
    
    MainWindow:
    BluetoothWindow:
    LoginWindow:
    
        
     
    
<LoginWindow>:
    name: "login"

    email: email
    password: password

    FloatLayout:
[... 171 lines cut ...]
   
    global notification_handler
    def notification_handler(sender, data):
        """Simple notification handler which prints the data received."""
        data1 = list(data)
        global data2
        data2 = ((data1[1]+data1[2]*256)*0.005)
        print("{0}: {1}".format(sender, data2))

    async def get_the_weight_from_the_mi_smart_scale_2(self):
        #da modificare
        '''This method is also run by the asyncio loop and periodically prints
        something.
        '''
        async with BleakClient("70:87:9E:0F:F8:7D") as client:
            print(f"Connected: {client.is_connected}")
            
            await client.start_notify("00002a9d-0000-1000-8000-00805f9b34fb", notification_handler)
            await asyncio.sleep(30.0)
            await client.stop_notify("00002a9d-0000-1000-8000-00805f9b34fb")   

if __name__ == '__main__':
    loop = asyncio.get_event_loop()
    loop.run_until_complete(AsyncApp().app_func())
    loop.close()

Egidio92, 2022-06-13:
I don't understand where the mistake is ... Thanks in advance for your help

Egidio92, 2022-06-13:
Finally, I add that the feature that the app does not read and therefore crashes is the one relating to the weight acquisition, in fact the device I want to connect is a Mi Smart Scale 2 scale.

The error relate to Characteristic:

bleak.exc.BleakError: Characteristic with UUID 00002a9d-0000-1000-8000-00805f9b34fb could not be found!

dlech, 2022-06-13:
Does the device actually have this characteristic? Does this error only happen on Android?

Egidio92, 2022-06-13:
Yes, the feature is there. 
No, this error also happened to me on my mac, in fact I had opened an issue a few days ago.

Egidio92, 2022-06-13:
Thanks again

dlech, 2022-06-13:
Was the output in https://github.com/hbldh/bleak/issues/846#issuecomment-1154151293 generated on Android or another OS?

Egidio92, 2022-06-13:
Yes exactly. It would practically be part of a project. But the error that is generated is the same. What I want to do is develop an app with kivy for android (Samsung Galaxy5, Android 6.0.1, API 23) and to acquire data from a device with low energy bluetooth with bleak. 

When I load the app on the phone from the logs in Android studio I see that it connects but it does not read the characteristic linked to start_notify.

Egidio92, 2022-06-13:
This output is generated on my mac. But it is the same program before conversion with buildozer on Colab

dlech, 2022-06-13:
Can you run the same example program on Android? I am wondering if the characteristics UUIDs are lower case and `str` type on Android like they are on macOS.

Egidio92, 2022-06-14:
Thanks for the suggestion I'll try. Anyway, I tried replacing the feature with upper case in characteristics UUID but it doesn't change anything. 
Thanks again.

closed 2022-07-13 by dlech