hbldh/bleak #850: ISSUE! -> PC (on which bleak example is running) is not able to successfully receive couple of first packets.

ASabovic, 2022-06-16:
* bleak version: 
* Python version: Python 3.8.3
* Operating System: Windows 10 Pro 
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I enabled the BLE communication between my PC and the Arduino Nano 33 BLE Sense board. I am trying to send a simple message just for the first tests (something like "Person is detected" or integer 0, 1, etc.). I can receive data on the PC side (as it acts as a central node) but it is not able to immediately start receiving data. It always misses receiving the first couple of packets, and after that, it works okay. Is there any way to enable the PC to immediately start receiving packets after the connection is established? How to remove this delay? For me, this is very important as I want to send an image from the Arduino to my PC and then restructure it there. Is there also a way to receive an image on the PC side or it is only capable to receive 1 packet per transmission? 
 
### What I Did

Here you can see the code that I am using: 

`# -*- coding: utf-8 -*-
"""
Notifications
-------------
Example showing how to add notifications to a characteristic and handle the responses.
Updated on 2019-07-03 by hbldh <henrik.blidh@gmail.com>
"""

import sys
import logging
import asyncio
import platform
import traceback

from bleak import BleakClient
from bleak import _logger as logger

CHARACTERISTIC_UUID = "ble_characteristic_uuid"  # <--- Change to the characteristic you want to enable notifications from.
ADDRESS = (
    "arduino_ble_device_address"  # <--- Change to your device's address here if you are using Windows or Linux
    if platform.system() != "Windows"
    else "arduino_ble_device_address"  # <--- Change to your device's address here if you are using macOS
)
if len(sys.argv) == 3:
    ADDRESS = sys.argv[1]
    CHARACTERISTIC_UUID = sys.argv[2]

def interpret(data):
	
	data = data.decode("utf-8")
	
	return data

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    dataMap = interpret(data)
    print("{0}: {1}".format(sender, dataMap))

    if dataMap=="1":
        data_to_send = "Message 1!"
        print(data_to_send)
    else:
        data_to_send = "Message 2!"
        print(data_to_send)

async def main(address, debug=False):
    
    if debug:
        import sys

        l = logging.getLogger("asyncio")
        l.setLevel(logging.DEBUG)
        h = logging.StreamHandler(sys.stdout)
        h.setLevel(logging.DEBUG)
        l.addHandler(h)
        logger.addHandler(h)

    async with BleakClient(address) as bleakClient:
        logger.info(f"Connected: {bleakClient.is_connected}")

        await bleakClient.start_notify(CHARACTERISTIC_UUID, notification_handler)
        await asyncio.sleep(2000.0)   # <--- seconds to keep running.
        # await bleakClient.stop_notify(CHARACTERISTIC_UUID)
        while True:
            await asyncio.sleep(10.0)

if __name__ == "__main__":
    import os

    os.environ["PYTHONASYNCIODEBUG"] = str(1)
    # loop = asyncio.get_event_loop()
    # loop.set_debug(True)
    # loop.run_until_complete(main(ADDRESS, True))
    try:
        asyncio.run(main(ADDRESS, True))
    except KeyboardInterrupt:
        print(' Shutdown requested. Exiting...')
	except Exception:
        traceback.print_exc(file=sys.stdout)
    
    try:
        sys.exit(0)
    except SystemExit:
        os._exit(0)
		

`

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

ASabovic, 2022-06-17:
Do you mean that I need to configure my Arduino board to wait until notifications are enabled? Also, does it work better on Ubuntu or the same issue is present? And also, is it possible to receive an image using bleak library?

dlech, 2022-06-17:
> Do you mean that I need to configure my Arduino board to wait until notifications are enabled?

Yes.

> Also, does it work better on Ubuntu or the same issue is present?

It is technically possible to start listening for notifications before connecting to the device with BlueZ. I don't remember if there is a way to do this with Bleak or not.

> And also, is it possible to receive an image using bleak library?

Sure. I've used Bleak for transferring other large files like firmware for a device. It just has to be split up into small pieces.

ASabovic, 2022-06-17:
Do you maybe have an example of how to enable this on an Arduino board using the BLE library? I will take a look to see if this part is implemented with Bleak (to start listening for notifications before connecting to the device). And for the last part, I already have an array of bytes (gotten from my image) and I can easily send the data by creating a for loop that will go through the array and send the data to the PC, but how to collect them on PC and wait until all of them is collected? Do you maybe have an example for enabling this?

ASabovic, 2022-06-30:
> Yes.
>
>
> It is technically possible to start listening for notifications before connecting to the device with BlueZ. I don't remember if there is a way to do this with Bleak or not.
>
>
> Sure. I've used [...]

> Yes.
>
>
> It is technically possible to start listening for notifications before connecting to the device with BlueZ. I don't remember if there is a way to do this with Bleak or not.
>
>
> Sure. I've used [...]

I am stuck as I do not know how to start with BlueZ on Ubuntu. Which repo can I use for examples? Or to be more specific, where can I find some basic examples just to see how it works and how it can be run on my PC? With bleak it was easy, just install bleak and run existing or your own examples, but now I am a bit confused because I saw that people used the same examples as with Bleak with some small modifications. Can you give me some guidelines please as I am really new in this :)

closed 2025-11-30 by github-actions[bot]