hbldh/bleak #853: Keep getting Error: bleak.exc.BleakError: Could not read characteristic handle 58: Protocol Error 0x02: Read Not Permitted

Boydenzzz, 2022-06-22:
* bleak version: 0.14.3
* Python version: 3.9.9
* Operating System: Windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I'm writing a program on windows in python using the bleak module to communicate with the RN4020 BLE module.

I can read and write to every UUID, but can not read data or receive data from Microchip Serial Port with UUID: 00035B03-58E6-07DD-021A-08123A0003FF.

It works with the app on android and on iOS, so it should work.

The error I got is: bleak.exc.BleakError: Could not read characteristic handle 58: Protocol Error 0x02: Read Not Permitted.

I also tried it with notify, but the characteristic does not support notifications or indications.

### What I Did

```
import asyncio
from bleak import BleakClient

address = "60:8A:10:88:EA:CF"

TX_UUID = "00035B03-58E6-07DD-021A-08123A000301"

RX_UUID = "00035B03-58E6-07DD-021A-08123A0003FF"

async def main(address):
    
    client = BleakClient(address)
    
    try:
        await client.connect()
        
        services = await client.get_services()
        
        for ser in services:
            print(ser.uuid)
            
        for x in range(10):
            await client.write_gatt_char(TX_UUID, b'Hello, world!!!\r\n', response=True)
        
        while(RX != b'\r'):
            RX = await client.read_gatt_char(RX_UUID, response=True)
            print(RX)
            await asyncio.sleep(0.001)
    
    except Exception as e:
        print(e)
    
    finally:
        await client.disconnect()
        
        

asyncio.run(main(address))

```
OUTPUT Python script:
Battery: 0
Could not get GATT characteristics for 00035b03-58e6-07dd-021a-08123a0003ff: ProtocolError (Error: 0x02: Unknown Connection Identifier)

OUTPUT UART TERMINAL:

Connected
ConnParam:000C,0064,04E2
MLDP
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
Hello, world!!!
CMD
Connection End
```

dlech, 2022-06-22:
> ProtocolError (Error: 0x02: Unknown Connection Identifier)

A quick web search for "Unknown Connection Identifier" reveals this is an error at the Bluetooth protocol level. I would log Bluetooth packets and if you control the software on the remote device, try to fix the problem there.

Boydenzzz, 2022-06-22:
Hi

I found a way around it using your example code.
I still think it's weird, but here is my solution:

```

import asyncio
import sys

from bleak import BleakScanner, BleakClient
from bleak.backends.device import BLEDevice

address = "60:8A:10:88:EA:CF"

UART_RX_TX_CHAR_UUID = "00035B03-58E6-07DD-021A-08123A000301"

UART_SAFE_SIZE = 20

async def uart_terminal(address):
    
    def handle_rx(_: int, data: bytearray):
        print("received:", data)

    async with BleakClient(address) as client:
        
        await client.start_notify(UART_RX_TX_CHAR_UUID, handle_rx)
        
        print("Connected, start typing and press ENTER...")

        
        loop = asyncio.get_running_loop()

        while True:
            
            data = await loop.run_in_executor(None, sys.stdin.buffer.readline)

            if not data:
                break

            
            await client.write_gatt_char(UART_RX_TX_CHAR_UUID, data)
            print("sent:", data)
            

if __name__ == "__main__":
    try:
        asyncio.run(uart_terminal(address))
    except asyncio.CancelledError:
        
        pass

```

closed 2022-06-23 by Boydenzzz