hbldh/bleak #859: Ran any example on Raspberry PI OS returns Operation already in progress

marcelomanzo, 2022-06-30:
* bleak version:  0.14.3
* Python version: 3.9.2
* Operating System: linux - raspbian latest version - raspberry pi 4
* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.55

### Description

Any example from the examples folder returns me error bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress:

### What I Did

For example:
https://github.com/hbldh/bleak/blob/develop/examples/discover.py

ssh://srsusr@192.168.0.6:22/home/srsusr/srsapp/venv/bin/python3 -u /tmp/pycharm_project_701/basic_example.py
Traceback (most recent call last):
  File "/tmp/pycharm_project_701/basic_example.py", line 27, in <module>
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/tmp/pycharm_project_701/basic_example.py", line 22, in main
    device = await BleakScanner.find_device_by_address(address)
  File "/home/srsusr/srsapp/venv/lib/python3.9/site-packages/bleak/backends/scanner.py", line 220, in find_device_by_address
    return await cls.find_device_by_filter(
  File "/home/srsusr/srsapp/venv/lib/python3.9/site-packages/bleak/backends/scanner.py", line 253, in find_device_by_filter
    return None
  File "/home/srsusr/srsapp/venv/lib/python3.9/site-packages/bleak/backends/scanner.py", line 99, in __aexit__
    await self.stop()
  File "/home/srsusr/srsapp/venv/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 177, in stop
    assert_reply(reply)
  File "/home/srsusr/srsapp/venv/lib/python3.9/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress

Process finished with exit code 1

marcelomanzo, 2022-06-30:
Tried to downgrade bleank to 0.13.0 and same issue

dlech, 2022-06-30:
The error comes from BlueZ, so you would need to look into the BlueZ source code to see what could cause this error in the `StopDiscovery` method or upgrade BlueZ to the latest version to see if it is a known bug that has been fixed. 

The [BlueZ docs](https://github.com/bluez/bluez/blob/1d6cfb8e625a944010956714c1802bc1e1fc6c4f/doc/adapter-api.txt#L31-L43) don't list this as a known error for this method.

marcelomanzo, 2022-06-30:
@dlech thanks, I manually updated the bluetoothctl to 5.64
srsusr@srs:~ $ bluetoothctl -v
bluetoothctl: 5.64

By using the instructions here:
https://scribles.net/updating-bluez-on-raspberry-pi-from-5-43-to-5-50/

and it works!

closed 2022-06-30 by marcelomanzo