hbldh/bleak #865: client.start_notify results in an exception

MrYsLab, 2022-07-03:
* bleak version: 0.14.3
* Python version: 3.10
* Operating System: Ubuntu 22.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description
I am connecting to a micro:bit and then trying to associate bleuart with notifying my program of any incoming data.

Here is a snippet of the code I am using:
```
        self.client = BleakClient(self.ble_mac_address)
        try:
            await self.client.connect()
        except bleak.exc.BleakDBusError:
            raise KeyboardInterrupt
        self.connected = True
        print('Connection successful')

        # associate the notification handler with incoming data
        await self.client.start_notify(self.UART_TX_UUID, self.notification_handler)
```
UART_TX_UUID is defined as "6e400003-b5a3-f393-e0a9-e50e24dcca9e"

I am using similar code with an ESP32 and that code works.

### What I Did

Running my application in a debugger, there are no issues until client_start_notify is called.

I verified that the connection is successful. 

I also placed a break-point on line 23 of utils.py and looked at the reply structure.
The reply body contains: 'Operation is not supported'
The reply error name contains: 'org.bluez.Error.NotSupported'

I have turned on debugging. The console output including exception traceback is below:
```
/home/afy/PycharmProjects/telemetrix-microbit/venv/bin/python /home/afy/PycharmProjects/telemetrix-microbit/experiments/loop_back_ble_aio.py
TelemetrixAioEsp32 Version: 1.3
Copyright (c) 2022 Alan Yorinks All rights reserved.

Retrieving BLE Mac Address of Ble Device. Please wait...
2022-07-03 17:55:09,866 bleak.backends.bluezdbus.scanner DEBUG: cached devices: {'/org/bluez/hci0/dev_30_AE_A4_3B_E6_DE': {'Address': '30:AE:A4:3B:E6:DE', 'AddressType': 'public', 'Name': 'Telemetrix4ESP32BLE', 'Alias': 'Telemetrix4ESP32BLE', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '6e400001-b5a3-f393-e0a9-e50e24dcca9e'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}}
2022-07-03 17:55:10,003 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
2022-07-03 17:55:10,023 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_FD_B4_7E_F3_89_39', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', FD:B4:7E:F3:89:39)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Name': <dbus_next.signature.Variant ('s', BBC micro:bit [vugeg])>, 'Alias': <dbus_next.signature.Variant ('s', BBC micro:bit [vugeg])>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -64)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-07-03 17:55:10,277 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_62_36_C0_E0_67_3A', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 62:36:C0:E0:67:3A)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 62-36-C0-E0-67-3A)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -99)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {76: <dbus_next.signature.Variant ('ay', b'\x10\x06\x03\x1a\xda\x7f\xfb\x8c')>})>, 'TxPower': <dbus_next.signature.Variant ('n', 7)>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-07-03 17:55:10,281 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_36_52_53_80_5E_8A', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 36:52:53:80:5E:8A)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 36-52-53-80-5E-8A)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -100)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', ['0000fd6f-0000-1000-8000-00805f9b34fb'])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ServiceData': <dbus_next.signature.Variant ('a{sv}', {'0000fd6f-0000-1000-8000-00805f9b34fb': <dbus_next.signature.Variant ('ay', b'\xbd"\\%\x8b\xc3\xa4m"J\x1a\xf3\xe8\xf7C;\xba\xe9\xc5\x91')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-07-03 17:55:10,300 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FD_B4_7E_F3_89_39): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -65)>}, []]
2022-07-03 17:55:10,551 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_62_36_C0_E0_67_3A): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -100)>}, []]
2022-07-03 17:55:10,552 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_36_52_53_80_5E_8A): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -100)>}, []]
2022-07-03 17:55:10,576 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FD_B4_7E_F3_89_39): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -65)>}, []]
2022-07-03 17:55:10,824 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_36_52_53_80_5E_8A): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -100)>}, []]
2022-07-03 17:55:10,843 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_62_36_C0_E0_67_3A): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -100)>}, []]
2022-07-03 17:55:10,897 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FD_B4_7E_F3_89_39): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -65)>}, []]
2022-07-03 17:55:11,116 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_62_36_C0_E0_67_3A): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -100)>}, []]
2022-07-03 17:55:11,118 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_FD_B4_7E_F3_89_39): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -64)>}, []]
2022-07-03 17:55:11,384 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_62_36_C0_E0_67_3A): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -100)>}, []]
[... cut ...]
  File "/home/afy/PycharmProjects/telemetrix-microbit/venv/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 931, in start_notify
    assert_reply(reply)
  File "/home/afy/PycharmProjects/telemetrix-microbit/venv/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.NotSupported] Operation is not supported

Process finished with exit code 1
```

dlech, 2022-07-04:
What is the output of `examples/service_explorer.py` for this device? Maybe it doesn't have the property set that indicates this property supports notifications.

MrYsLab, 2022-07-04:
@dlech Thanks, that was extremely useful. The output from the micro:bit for Nordic UART Service, TX characteristic is:
```
INFO:__main__:	[Characteristic] 6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 41): Nordic UART TX (write-without-response,write), Value: None
```

And on a CircuitPlayGround Bluefruit device, that does not generate the exception, the output is:
```
INFO:__main__:	[Characteristic] 6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 34): Nordic UART TX (notify), Value: None
```

It appears that micro:bit does not support notifications as currently configured, so this is not a bug in bleak. I am closing the issue and will now dig into why the micro:bit is being configured this way.

closed 2022-07-04 by MrYsLab

MrYsLab, 2022-07-04:
For those who may come across this same issue, referring to this issue in the micro:bit repo, **Nordic UART service bug: should use notifications not indications** #259, the micro:bit folks flipped the TX and RX characteristics. I flipped them in my code, and it now compiles. Hopefully, I will be able to transmit and receive.