hbldh/bleak #868: AttributeError: 'CentralManagerDelegate' object has no attribute 'central_manager' on macOS

snowuyl, 2022-07-06:
* bleak version: 0.15.0a1
* Python version:  3.9.12
* Operating System: macOS 12.4
I write a Ble class to use bleak to scan BLE devices. But  The following error occured.
AttributeError: 'CentralManagerDelegate' object has no attribute 'central_manager'.

### What I Did
python3 main.py 
Traceback (most recent call last):
  File "/Users/snowuyl/samba/workspace_Python3/bleak_scan_devices/main.py", line 4, in <module>
    bleObj = Ble()
  File "/Users/snowuyl/samba/workspace_Python3/bleak_scan_devices/ble.py", line 6, in __init__
    self.scanner                = BleakScanner()
  File "/opt/homebrew/lib/python3.9/site-packages/bleak/backends/corebluetooth/scanner.py", line 44, in __init__
    self._manager = CentralManagerDelegate.alloc().init()
  File "/opt/homebrew/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 61, in init
    self.event_loop = asyncio.get_running_loop()
RuntimeError: no running event loop
Exception ignored in: <CentralManagerDelegate objective-c instance 0x0>
Traceback (most recent call last):
  File "/opt/homebrew/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 102, in __del__
    self.central_manager.removeObserver_forKeyPath_(self, "isScanning")
AttributeError: 'CentralManagerDelegate' object has no attribute 'central_manager'
Exception ignored in: <function CentralManagerDelegate.__del__ at 0x103feaee0>

snowuyl, 2022-07-06:
[bleak_scan_devices.zip](https://github.com/hbldh/bleak/files/9052009/bleak_scan_devices.zip)

dlech, 2022-07-06:
The `AttributeError` is a side-effect of the main error: `RuntimeError: no running event loop`. Bleak can only be used in function running inside of an asyncio event loop.

snowuyl, 2022-07-06:
Do you have example of how to use it?

snowuyl, 2022-07-07:
Thanks for your support! It works with  asyncio event loop.

closed 2022-07-07 by snowuyl