hbldh/bleak #874: missing data in advertisement data

rty813, 2022-07-08:
* bleak version: 0.14.3/0.14.3
* Python version: 3.8(aarch64)/3.10(x86)
* Operating System: ubuntu16.04(aarch64)/ubuntu20.04(x86)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64/5.64

### Description

I have two computers. One is X86 architecture, the other is aarch64. When I use these two computers to scan the same Bluetooth device, I get different advertisement data. There are no "locan name" and "manufacturer data" in the scanning result of aarch64. 

The scanning result of X86:
[image]

The scanning result of aarch64:
[image]

Then I turned on debugging mode by `export BLEAK_LOGGING=1`, the log of aarch64 is :
[image]

scan code:
```python
import asyncio
import time
from bleak import BleakScanner, BleakClient

def detection_callback(device, data):
    if device.address != '80:4B:50:38:F3:2F':
        return

    print(device, data)

async def main():
    global connected
    print('查找一波')
    scanner = BleakScanner()
    scanner.register_detection_callback(detection_callback)
    await scanner.start()
    while True:
        await asyncio.sleep(1)

asyncio.run(main())

```

---

However, if I use bluepy on aarch64, I can get the local name and manufacturer data.
[image]

rty813, 2022-07-08:
I can get local name and manufacturer data of the ble device from "EFR Connect" app on my phone.
[image]

dlech, 2022-07-08:
Since the aarch64 system is using Ubuntu 16.04, I assume that it is using an older Linux kernel and this is what makes the difference.

rty813, 2022-07-09:
> 

But why can bluepy get localname and manufacturer data?

dlech, 2022-07-09:
Does bluepy use d-bus?

rty813, 2022-07-10:
I am not sure. This is its source code, it seems to use bluetooth library directly
[image]

rty813, 2022-07-10:
> bluepy is a Python module which allows communication with Bluetooth Low Energy devices. The current implementation runs on Linux (internally it uses code from the BlueZ project), although it can be ported to other platforms.

rty813, 2023-04-18:
> Does bluepy use d-bus?

Any suggestion?

closed 2025-11-30 by github-actions[bot]