hbldh/bleak #875: [BlueZ] Scanner fails to close sockets, which ain't that great

leycec, 2022-07-09:
* bleak version: **0.14.3**
* Python version: **3.10..5**
* Operating System: **Gentoo Linux,** of course!
* BlueZ version (`bluetoothctl -v`) in case of Linux: **5.64**

### Description

Bleak's BlueZ-backed scanner implementation fails to close sockets in a timely manner – like, *ever.* That's badness incarnate, because sockets are extremely scarce OS resources. It's definitely best not to squander them. Sadly, bleak squanders them.

### What I Did

Minimal length example or it didn't happen, so:

```bash
$ python3.10 -X dev -c '
from asyncio import run
from bleak import BleakScanner

run(BleakScanner.discover())
'
sys:1: ResourceWarning: unclosed <socket.socket fd=6, family=AddressFamily.AF_UNIX, type=SocketKind.SOCK_STREAM, proto=0, raddr=/run/dbus/system_bus_socket>
```

:open_mouth: 

Note that the `-X dev` is essential here. By default, Python hides many critical things that it *really* shouldn't. This includes deprecation and resource warnings. Sad, yet true.

Above, `-X dev` is disclosing that the `bleak.backends.bluezdbus.scanner.BleakScannerBlueZDBus.start()` method is failing to clean up after itself. Interestingly, the companion `bleak.backends.bluezdbus.client.BleakClientBlueZDBus.connect()` method *does* clean up after itself: e.g.,

```
# In "bleak.backends.bluezdbus.client":
class BleakClientBlueZDBus(BaseBleakClient):
    ...

    async def connect(self, **kwargs) -> bool:
        try:
            ...
        except BaseException:
            self._cleanup_all()
            raise
```

The `BleakClientBlueZDBus._cleanup_all()` method, in turn, correctly deallocates and closes all currently open resources. The `BleakScannerBlueZDBus` class could benefit from a similar `_cleanup_all()` method, which the `BleakScannerBlueZDBus.start()` method should then call from a similar `try: ...; except BaseException: ...` branch.

## And... We Done Here

Thanks so much (as always!) for this *phenomenal* framework. Bleak enables our entire tech stack. We wouldn't be here without you. Here's to you, breathtaking dev team. :clinking_glasses:

elupus, 2022-07-09:
Well. It should be used as a contex manager.

leycec, 2022-07-09:
But... it's not a context manager? I don't think, anyway. I actually tried that first but received no joy. In any case, I'll let you talk the helm here. Thanks for the attention!

dlech, 2022-07-11:
Also probably the cause of #805

leycec, 2022-07-12:
**Yup.** You're almost certainly right on both accounts. Sadly, `python-dbus-next` development is *really* moribund; [their last commit was over six months ago](https://github.com/altdesktop/python-dbus-next/commits/master). It's not quite dead, but it doesn't exactly inspire confidence either.

On the other hand, `bleak` (probably) shouldn't violate privacy encapsulation by [manually calling `self._bus._sock.close()` to manually close open D-Bus sockets](https://github.com/altdesktop/python-dbus-next/issues/112#issue-1125936487). I mean, we *could* – but we (probably) shouldn't.

Thanks again for all the investigatory legwork, @dlech. This a surprising pickle we have here, but I'm sure something will eventually turn up. Perhaps someone will fork and maintain `python-dbus-next` for us? A GitHub dev can dream. :smile:

hbldh, 2022-07-12:
It is an internal attribute on the `MessageBus`, but given that the real problems that this causes I would argue that is is reasonable to include some ugly handling if it is simple and non-intrusive.

This is my "at-least-we-can-try"-attempt that I do not think (but haven't verified) will introduce any noticable delays or errors.

```patch
From 3a4dbcc6ddcfbff0ef160dcf20f5126cde3c3004 Mon Sep 17 00:00:00 2001
From: Henrik Blidh <henrik.blidh@nedomkull.com>
Date: Tue, 12 Jul 2022 09:31:18 +0200
Subject: [PATCH] Attempt to remedy #875

---
 bleak/backends/bluezdbus/client.py  | 6 ++++++
 bleak/backends/bluezdbus/scanner.py | 7 +++++++
 2 files changed, 13 insertions(+)

diff --git a/bleak/backends/bluezdbus/client.py b/bleak/backends/bluezdbus/client.py
index f8053e7..05921c4 100644
--- a/bleak/backends/bluezdbus/client.py
+++ b/bleak/backends/bluezdbus/client.py
@@ -363,6 +363,12 @@ class BleakClientBlueZDBus(BaseBleakClient):
         # Try to disconnect the System Bus.
         try:
             self._bus.disconnect()
+            # An attempt to remedy the failure to close the DBus socket, see
+            # https://github.com/hbldh/bleak/issues/875
+            try:
+                self._bus._sock.close()
+            except:  # noqa
+                pass
         except Exception as e:
             logger.error(
                 f"Attempt to disconnect system bus failed ({self._device_path}): {e}"
diff --git a/bleak/backends/bluezdbus/scanner.py b/bleak/backends/bluezdbus/scanner.py
index d241c02..3db0ed4 100644
--- a/bleak/backends/bluezdbus/scanner.py
+++ b/bleak/backends/bluezdbus/scanner.py
@@ -187,6 +187,13 @@ class BleakScannerBlueZDBus(BaseBleakScanner):
         except Exception as e:
             logger.error("Attempt to disconnect system bus failed: {0}".format(e))
 
+        # An attempt to remedy the failure to close the DBus socket, see
+        # https://github.com/hbldh/bleak/issues/875
+        try:
+            self._bus._sock.close()
+        except:  # noqa
+            pass
+
         self._bus = None
 
     def set_scanning_filter(self, **kwargs):
-- 
2.37.0.windows.1
```

It is also available in the [`issue-875`](https://github.com/hbldh/bleak/tree/issue-875) branch. I would be grateful if you could try it out in your environment to see if it actually alleviates the problem for now.

dlech, 2022-07-12:
FYI, I'm currently working on some changes that will sidestep the issue for the scanner.

hbldh, 2022-07-12:
Very good, I will not make a PR of the temporary fix above then! 😁

dlech, 2022-07-26:
Closing as duplicate of #805.

closed 2022-07-26 by dlech