hbldh/bleak #883: Android app crash after lunch

Egidio92, 2022-07-13:
* bleak version: 0.14.3
* Python version: Python 3.7.13 (Colab)
* Operating System: Samsung SM-G900F Android 6.0.1, API 23

### Description #######################################################

Hello! I am trying to implement an application on an android mobile. This app needs to connect via low-energy bluetooth BLE to a scale (Mi Smart Scale 2) and take the weight. To create the apk of the application I use Colab.
The problem is that when I launch the application it crashes immediately after starting and I cannot read the weight.

### What I Did ########################################################

import asyncio

from kivy.app import App
from kivy.lang.builder import Builder
from kivy.uix.screenmanager import ScreenManager, Screen
from kivy.properties import ObjectProperty

from bleak import BleakClient

kv = '''
WindowManager:
    
    MainWindow:
    BluetoothWindow:
    LoginWindow:
    
        
     
    
<LoginWindow>:
    name: "login"

    email: email
    password: password

    FloatLayout:

        Label:
            text:"Email: "
            font_size: (root.width**2 + root.height**2) / 13**4
            pos_hint: {"x":0.1, "top":0.9}
            size_hint: 0.35, 0.15

        TextInput:
            id: email
            font_size: (root.width**2 + root.height**2) / 13**4
            multiline: False
            pos_hint: {"x": 0.45 , "top":0.9}
            size_hint: 0.4, 0.15

        Label:
            text:"Password: "
            font_size: (root.width**2 + root.height**2) / 13**4
            pos_hint: {"x":0.1, "top":0.7}
            size_hint: 0.35, 0.15

        TextInput:
            id: password
            font_size: (root.width**2 + root.height**2) / 13**4
            multiline: False
            password: True
            pos_hint: {"x": 0.45, "top":0.7}
            size_hint: 0.4, 0.15

        Button:
            pos_hint:{"x":0.2,"y":0.05}
            size_hint: 0.6, 0.2
            font_size: (root.width**2 + root.height**2) / 13**4
            text: "Login"
[... 219 lines cut ...]
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0016
2022-07-13 15:53:26.367 1983-2417/? D/bt_upio: upio_start_stop_timer : timer_settime success
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0016
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x0016
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x0016
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=8 connected=0 conn_id=8 reason=0x0016
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=9 connected=0 conn_id=9 reason=0x0016
2022-07-13 15:53:26.367 1983-2417/? D/bt_upio: upio_set: proc btwrite assertion, buffer: 1, timer_armed 1 1
2022-07-13 15:53:26.367 1983-2477/? W/bt_btif: bta_gattc_conn_cback() - cif=10 connected=0 conn_id=10 reason=0x0016
2022-07-13 15:53:26.367 1983-2477/? E/bt_btif: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 70:87:9e:0f:f8:7d
2022-07-13 15:53:26.367 1983-2204/? D/BtGatt.GattService: onDisconnected() - clientIf=10, connId=10, address=70:87:9E:0F:F8:7D
2022-07-13 15:53:26.367 14122-14157/org.test.tesi4 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=70:87:9E:0F:F8:7D
2022-07-13 15:53:26.367 14122-14150/org.test.tesi4 D/BluetoothGatt: close()
2022-07-13 15:53:26.367 14122-14150/org.test.tesi4 D/BluetoothGatt: unregisterApp() - mClientIf=10
2022-07-13 15:53:26.377 1983-2505/? D/BtGatt.GattService: unregisterClient() - clientIf=10
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:  Traceback (most recent call last):
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:    File "/content/.buildozer/android/app/main.py", line 228, in <module>
2022-07-13 15:53:26.377 836-1510/? V/AlarmManager:  remove PendingIntent] PendingIntent{317de15: PendingIntentRecord{83cd4dc com.android.bluetooth broadcastIntent}}
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:    File "/content/.buildozer/android/platform/build-armeabi-v7a/build/other_builds/python3/armeabi-v7a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 616, in run_until_complete
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:    File "/content/.buildozer/android/app/main.py", line 216, in get_the_weight_from_the_mi_smart_scale_2
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:    File "/content/.buildozer/android/platform/build-armeabi-v7a/build/python-installs/tesi4/armeabi-v7a/bleak/backends/p4android/client.py", line 486, in start_notify
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:  bleak.exc.BleakError: Characteristic with UUID 00002a9d-0000-1000-8000-00805f9b34fb could not be found!
2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python: Python for android ended.
2022-07-13 15:53:26.387 1983-2477/? I/bt_btm_sec: btm_sec_disconnected clearing pending flag handle:64 reason:22
```

Egidio92, 2022-07-13:
And this is the buildozer.spec file:

[app]

# (str) Title of your application
title = Demo3

# (str) Package name
package.name = demo3

# (str) Package domain (needed for android/ios packaging)
package.domain = org.test

# (str) Source code where the main.py live
source.dir = .

# (list) Source files to include (let empty to include all the files)
source.include_exts = py,png,jpg,kv,atlas

# (list) List of inclusions using pattern matching
#source.include_patterns = assets/*,images/*.png

# (list) Source files to exclude (let empty to not exclude anything)
#source.exclude_exts = spec

# (list) List of directory to exclude (let empty to not exclude anything)
#source.exclude_dirs = tests, bin, venv

# (list) List of exclusions using pattern matching
# Do not prefix with './'
#source.exclude_patterns = license,images/*/*.jpg

# (str) Application versioning (method 1)
version = 0.1

# (str) Application versioning (method 2)
# version.regex = __version__ = ['"](.*)['"]
# version.filename = %(source.dir)s/main.py

# (list) Application requirements
# comma separated e.g. requirements = sqlite3,kivy
requirements = python3,kivy,pillow,kivymd,asyncio,bleak

# (str) Custom source folders for requirements
# Sets custom source for any requirements with recipes
# requirements.source.kivy = ../../kivy

# (str) Presplash of the application
#presplash.filename = %(source.dir)s/data/presplash.png

# (str) Icon of the application
#icon.filename = %(source.dir)s/data/icon.png

# (str) Supported orientation (one of landscape, sensorLandscape, portrait or all)
orientation = portrait

# (list) List of service to declare
#services = NAME:ENTRYPOINT_TO_PY,NAME2:ENTRYPOINT2_TO_PY

#
# OSX Specific
#

#
# author = © Copyright Info

# change the major version of python used by the app
osx.python_version = 3

# Kivy version to use
[... 331 lines cut ...]
#    This can be translated into:
#
#[app:source.exclude_patterns]
#license
#data/audio/*.wav
#data/images/original/*
#

#    -----------------------------------------------------------------------------
#    Profiles
#
#    You can extend section / key with a profile
#    For example, you want to deploy a demo version of your application without
#    HD content. You could first change the title to add "(demo)" in the name
#    and extend the excluded directories to remove the HD content.
#
#[app@demo]
#title = My Application (demo)
#
#[app:source.exclude_patterns@demo]
#images/hd/*
#
#    Then, invoke the command line with the "demo" profile:
#
#buildozer --profile demo android debug

dlech, 2022-07-13:
> 2022-07-13 15:53:26.377 14122-14150/org.test.tesi4 I/python:  bleak.exc.BleakError: Characteristic with UUID 00002a9d-0000-1000-8000-00805f9b34fb could not be found!

Either there is a bug with characteristic discovery, the android os hides this characteristic or the device does not have this characteristic. If we can rule out the last two, then we can look for a bug in Bleak.

Egidio92, 2022-07-13:
Thanks for the quick reply. The device has this characteristic (i'm 100% sure) because by launching the same code from Mac it works. What do you advise me to do?

Egidio92, 2022-07-13:
Of course! I'll answer in about half an hour. I build the apk with Colab. Thank you.

dlech, 2022-07-13:
> 2022-07-13 18:12:41.390 21906-21935/org.test.tesi6 I/python: {0: <bleak.backends.p4android.characteristic.BleakGATTCharacteristicP4Android object at 0x9a2009e8>}

This shows there is only one characteristic with handle `0` which doesn't seem right. I think we assumed that `getInstanceId()` returned the handle, but I think that it just returns a number that is unique per characteristics with the same UUID. That is, it returns 0 for most characteristics and 1 for the second instance of a characteristic with the same UUID as an already seen characteristic.

So, to fix this, I think we need to create a mapping between a tuple containing the UUID and instance id to a "handle" integer value to be compatible with other Bleak backends.

Egidio92, 2022-07-13:
Thanks for the replies. However I am trying to solve the problem. 😄

Egidio92, 2022-07-18:
How can I do to find the one I need? 
That is: "00002a9d-0000-1000-8000-00805f9b34fb" ...Which is the one relating to the weight of the scale passed by BLE

Egidio92, 2022-07-18:
These days I have done various tests also setting the code differently but I always find only this unique characteristic "00002a00-0000-1000-8000-00805f9b34fb", which should be linked to the name of the device

Egidio92, 2022-07-18:
Thanks for the reply. I just used the nRF Connect app and my android phone sees that characteristic and the others characteristics too.

Egidio92, 2022-08-31:
[characteristic.txt](https://github.com/hbldh/bleak/files/9461339/characteristic.txt)
[client.txt](https://github.com/hbldh/bleak/files/9461348/client.txt)
[descriptor.txt](https://github
[service.txt](https://github.com/hbldh/bleak/files/9461355/service.txt)
.com/hbldh/bleak/files/9461354/descriptor.txt)

Egidio92, 2022-08-31:
[characteristic.txt](https://github.com/hbldh/bleak/files/9461357/characteristic.txt)

[client.txt](https://github.com/hbldh/bleak/files/9461358/client.txt)

[descriptor.txt](https://github.com/hbldh/bleak/files/9461361/descriptor.txt)

[service.txt](https://github.com/hbldh/bleak/files/9461363/service.txt)

Egidio92, 2022-08-31:
By editing these four files and replacing them in the p4android folder I was able to remap the handles. I don't know if it's a clean result but it worked for me.

Egidio92, 2022-08-31:
For example, in the characteristic.py file I made sure that an index was added to the value "self.obj.getInstanceId()" for each new service found. Because "self.obj.getInstanceId()" always returns zero for every service, characteristic handle. I don't know if it's correct but it works for me now

Egidio92, 2022-09-22:
what do you think of my solution?

dlech, 2022-09-22:
In #888, we discussed relaxing the `int` requirement on the handle property. I think it would be more straight-forward to make the "handle" on Android a tuple of the UUID string and the instance ID.

Egidio92, 2022-09-22:
Ok perfect I will continue to follow the updates. thank you :)

vabt-igd, 2024-01-30:
I just now ran into similar issues. 
I have one Ubuntu VM that I have been using for about a year, which is able to build a kivy app for Android that makes heavy usage of bleak. 
Now, to move my dev to a new computer, I created a new Ubuntu VM based on the same version as previous. After compiling the Android app on that VM using the exact same code base as before, the app fails to connect to my BLE devices due to “missing” services and characteristics. Merging @Egidio92 workaround into bleaks current develop branch did help slightly, as to now the app at least is able to find the corresponding services and characteristics, but start_notify does not seem to work. The notify callbacks never get triggered.
I'll try to further investigate the issue...
@dlech regarding #888, is there anything planned regarding the __handle attribute refactoring?

dlech, 2024-01-31:
No definite plans.