hbldh/bleak #889: Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

ClusterM, 2022-07-17:
* bleak version:
bleak-0.14.3
* Python version:
Python 3.9.2
* Operating System:
Linux lovepoint 5.15.48-sunxi64 #22.05.3 SMP Wed Jun 22 07:33:24 UTC 2022 aarch64 GNU/Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:
bluetoothctl: 5.55
### Description
Can't execute `bleak/examples/scanner.py`:

```
cluster@lovepoint:~/bleak/examples$ sudo python3 scanner.py
Traceback (most recent call last):
  File "/home/cluster/bleak/examples/scanner.py", line 31, in <module>
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/cluster/bleak/examples/scanner.py", line 26, in main
    device = await BleakScanner.find_device_by_address(address)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/scanner.py", line 220, in find_device_by_address
    return await cls.find_device_by_filter(
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/scanner.py", line 249, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/scanner.py", line 95, in __aenter__
    await self.start()
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/scanner.py", line 155, in start
    assert_reply(reply)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist
```

ClusterM, 2022-07-17:
I commented out all `assert_reply(reply)` in scanner.py. It doesn't throw exception now but can't find any device.

dlech, 2022-07-17:
> sudo python3 scanner.py

Sudo is not required and should not be used with Bleak.

> bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.UnknownObject] Method "SetDiscoveryFilter" with signature "a{sv}" on interface "org.bluez.Adapter1" doesn't exist

It this the same problems as #693?

ClusterM, 2022-07-17:
> It this the same problems as #693?

Seems like yes.

dlech, 2022-07-17:
We should probably try to make a better error message for this. Also #524 could help avoid the problem in some cases.

PR #900 (2022-07-25): bluezdbus/manager: add adapter check and error
This adds a better error message when an adapter is missing on the BlueZ backend. Instead of getting a D-Bus method not found error, it will actually say what the real problem is - the adapter is not present.

Fixes #889.
commit a5b98c7a (2022-07-29): v0.15.0

Added
-----

* Added new ``assigned_numbers`` module and ``AdvertisementDataType`` enum.
* Added new ``bluez`` kwarg to ``BleakScanner`` in BlueZ backend.
* Added support for passive scanning in the BlueZ backend. Fixes #606.
* Added option to use cached services, characteristics and descriptors in WinRT backend. Fixes #686.
* Added ``PendingDeprecationWarning`` to use of ``address_type`` as keyword argument. It will be moved into the
  ``winrt`` keyword instead according to #623.
* Added better error message when adapter is not present in BlueZ backend. Fixes #889.

Changed
-------

 [... cut]
PR #911 (2022-07-29): Release/v0.15.0

Added
-----

* Added new ``assigned_numbers`` module and ``AdvertisementDataType`` enum.
* Added new ``bluez`` kwarg to ``BleakScanner`` in BlueZ backend.
* Added support for passive scanning in the BlueZ backend. Fixes #606.
* Added option to use cached services, characteristics and descriptors in WinRT backend. Fixes #686.
* Added ``PendingDeprecationWarning`` to use of ``address_type`` as keyword argument. It will be moved into the
  ``winrt`` keyword instead according to #623.
* Added better error message when adapter is not present in BlueZ backend. Fixes #889.

Changed
- [... cut]
closed 2022-07-25 by dlech