hbldh/bleak #890: He

emileberhard, 2022-07-18:
* bleak version: 0.14.3
* Python version: 3.9
* Operating System: MacOS 12.4 Monterey

### Description
I have written a short python script that uses bleak to control a bluetooth LE device. The script works exactly as intended when I run it manually in pycharm or terminal, however I want to run it at specific intervals using crontab.

I've searched for a solution and tried adding absolute paths as well as logging errors. I have found that the script throws an error when run from crontab but I don't quite understand how to fix it.

### What I Did

Script:
```
#!/usr/bin/python
from bleak import BleakClient
import asyncio

address = "DFB18A2C-6BAB-439E-AF05-485FB4259B75"
MODEL_NBR_UUID = "99FA0002-338A-1024-8A49-009C0215F78A"

async def main(address):
    async with BleakClient(address) as client:
        for i in range(3):
            await client.write_gatt_char(MODEL_NBR_UUID, bytes.fromhex("0800"))
            time.sleep(0.5)

asyncio.run(main(address))
```

This is what my crontab looks like:
```
emileberhard@Emils-MacBook-Air ~ % sudo crontab -l
0 0 * * * python /Users/emileberhard/PycharmProjects/BedMove/main.py >> /Users/emileberhard/out2.txt 2>&1
```

Here is the error I see in crontab log:
```
Traceback (most recent call last):
  File "/Users/emileberhard/PycharmProjects/BedMove/main.py", line 19, in <module>
    asyncio.run(main(address))
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/emileberhard/PycharmProjects/BedMove/main.py", line 13, in main
    async with BleakClient(address) as client:
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/client.py", line 61, in __aenter__
    await self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/corebluetooth/client.py", line 78, in connect
    device = await BleakScannerCoreBluetooth.find_device_by_address(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/scanner.py", line 220, in find_device_by_address
    return await cls.find_device_by_filter(
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/scanner.py", line 249, in find_device_by_filter
    async with cls(detection_callback=apply_filter, **kwargs):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/corebluetooth/scanner.py", line 44, in __init__
    self._manager = CentralManagerDelegate.alloc().init()
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 88, in init
    raise BleakError("Bluetooth device is turned off")
bleak.exc.BleakError: Bluetooth device is turned off
Exception ignored in: <CentralManagerDelegate objective-c instance 0x0>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 102, in __del__
IndexError: NSRangeException - Cannot remove an observer <CentralManagerDelegate 0x7fa1c7f7cae0> for the key path "isScanning" from <CBCentralManager 0x600000fa4400> because it is not registered as an observer.
Exception ignored in: <function CentralManagerDelegate.__del__ at 0x7fa1b03c4ca0>
```

Does anyone know what this error means?
Thanks in advance!

dlech, 2022-07-18:
Maybe you can't use Bluetooth as root on macOS?

emileberhard, 2022-07-18:
@dlech I get the same error regardless of using root crontab or not though

dlech, 2022-07-18:
I guess try some other way of scheduling besides crontab.

emileberhard, 2022-07-19:
Yes launchd worked better thanks!

closed 2022-07-19 by dlech