hbldh/bleak #892: Crash at enumerate bluetooth.le.ScanSettings$Builder "setPhy"

navr32, 2022-07-24:
* bleak version: 0.14.3 or master
* Python version: 3.10
* Operating System:  Android build on Manjaro Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:  bluetoothctl: 5.64

#####
I run the kivy demo on linux mode ok  "python main.py" ! 
 The scanner started work and find bluetooth...and so.
After i  Build with buildozer android debug deploy run logcat ...the build is ok ..and the deploy  ok..the app start..
But after start show screen message  "scanning"  but  the app crash.

The final message on debug log is 

File "/bleak-0.14.3/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/python-installs/bleakdemo/arm64-v8a/bleak/backends/p4android/scanner.py", line 108, in start
07-24 18:33:21.169  5707  5850 I python  :  AttributeError: 'android.bluetooth.le.ScanSettings$Builder' object has no attribute 'setPhy'

So i have search in sanner.py  and the setPhy attribute is here at some line under 108...very surprised of this error and i don't find
anything about this to have a trick..i have search on p4android forum too..because this the android backend ..but any subject about this..
perhaps a missing dependencies ? 

***************************************************************************************************************************************
crashback:

# Search for Git (git)
#  -> found at /usr/bin/git
# Search for Cython (cython)
#  -> found at /home/nico/kivy/bleak_old/bleak/examples/kivy/bin/cython
# Search for Java compiler (javac)
#  -> found at /usr/lib/jvm/java-11-openjdk/bin/javac
# Search for Java keytool (keytool)
#  -> found at /usr/lib/jvm/java-11-openjdk/bin/keytool
# Run '/home/nico/.buildozer/android/platform/android-sdk/platform-tools/adb logcat *:S python:D '
# Cwd /home/nico/.buildozer/android/platform
--------- beginning of crash
--------- beginning of system
--------- beginning of main
07-24 14:51:20.732 23944 24296 I python  : Initializing Python for Android
07-24 14:51:20.732 23944 24296 I python  : Setting additional env vars from p4a_env_vars.txt
07-24 14:51:20.732 23944 24296 I python  : Changing directory to the one provided by ANDROID_ARGUMENT
07-24 14:51:20.732 23944 24296 I python  : /data/user/0/com.github.hbldh.bleakdemo/files/app
07-24 14:51:20.739 23944 24296 I python  : Preparing to initialize python
07-24 14:51:20.739 23944 24296 I python  : _python_bundle dir exists
07-24 14:51:20.739 23944 24296 I python  : calculated paths to be...
07-24 14:51:20.739 23944 24296 I python  : /data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/stdlib.zip:/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/modules
07-24 14:51:20.746 23944 24296 I python  : set wchar paths...
07-24 14:51:20.860 23944 24296 I python  : Initialized python
07-24 14:51:20.860 23944 24296 I python  : AND: Init threads
07-24 14:51:20.862 23944 24296 I python  : testing python print redirection
07-24 14:51:20.864 23944 24296 I python  : Android path ['.', '/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/stdlib.zip', '/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/modules', '/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/site-packages']
07-24 14:51:20.864 23944 24296 I python  : os.environ is environ({'PATH': '/sbin:/vendor/bin:/system/sbin:/system/bin:/system/xbin', 'ANDROID_BOOTLOGO': '1', 'ANDROID_ROOT': '/system', 'ANDROID_ASSETS': '/system/app', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'EXTERNAL_STORAGE': '/sdcard', 'ASEC_MOUNTPOINT': '/mnt/asec', 'BOOTCLASSPATH': '/system/framework/core-oj.jar:/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar:/system/framework/mediatek-common.jar:/system/framework/mediatek-framework.jar:/system/framework/mediatek-telephony-common.jar', 'SYSTEMSERVERCLASSPATH': '/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar', 'ANDROID_SOCKET_zygote': '8', 'ANDROID_ENTRYPOINT': 'main.pyc', 'ANDROID_ARGUMENT': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'ANDROID_APP_PATH': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'ANDROID_PRIVATE': '/data/user/0/com.github.hbldh.bleakdemo/files', 'ANDROID_UNPACK': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'PYTHONHOME': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'PYTHONPATH': '/data/user/0/com.github.hbldh.bleakdemo/files/app:/data/user/0/com.github.hbldh.bleakdemo/files/app/lib', 'PYTHONOPTIMIZE': '2', 'P4A_BOOTSTRAP': 'SDL2', 'PYTHON_NAME': 'python', 'P4A_IS_WINDOWED': 'True', 'P4A_ORIENTATION': 'portrait', 'P4A_NUMERIC_VERSION': 'None', 'P4A_MINSDK': '21', 'LC_CTYPE': 'C.UTF-8'})
07-24 14:51:20.864 23944 24296 I python  : Android kivy bootstrap done. __name__ is __main__
07-24 14:51:20.864 23944 24296 I python  : AND: Ran string
07-24 14:51:20.864 23944 24296 I python  : Run user program, change dir and execute entrypoint
07-24 14:51:21.244 23944 24296 I python  : [WARNING] [Config      ] Older configuration version detected (0 instead of 24)
07-24 14:51:21.245 23944 24296 I python  : [WARNING] [Config      ] Upgrading configuration in progress.
07-24 14:51:21.256 23944 24296 I python  : [INFO   ] [Logger      ] Record log in /data/user/0/com.github.hbldh.bleakdemo/files/app/.kivy/logs/kivy_22-07-24_0.txt
07-24 14:51:21.257 23944 24296 I python  : [INFO   ] [Kivy        ] v2.1.0
07-24 14:51:21.258 23944 24296 I python  : [INFO   ] [Kivy        ] Installed at "/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/site-packages/kivy/__init__.pyc"
07-24 14:51:21.259 23944 24296 I python  : [INFO   ] [Python      ] v3.9.9 (main, Jul 22 2022, 18:36:06) 
07-24 14:51:21.259 23944 24296 I python  : [Clang 12.0.8 (https://android.googlesource.com/toolchain/llvm-project c935d99d
07-24 14:51:21.260 23944 24296 I python  : [INFO   ] [Python      ] Interpreter at ""
07-24 14:51:21.262 23944 24296 I python  : [INFO   ] [Logger      ] Purge log fired. Processing...
07-24 14:51:21.264 23944 24296 I python  : [INFO   ] [Logger      ] Purge finished!
07-24 14:51:23.653 23944 24296 I python  : [INFO   ] [Factory     ] 189 symbols loaded
07-24 14:51:24.140 23944 24296 I python  : [INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2 (img_pil, img_ffpyplayer ignored)
07-24 14:51:24.202 23944 24296 I python  : [INFO   ] [Text        ] Provider: sdl2
07-24 14:51:25.088 23944 24296 I python  : [DEBUG  ] [Using selector] EpollSelector
07-24 14:51:25.091 23944 24296 I python  : [DEBUG  ] [App         ] Loading kv <./example.kv>
07-24 14:51:25.093 23944 24296 I python  : [DEBUG  ] [App         ] kv <./example.kv> not found
07-24 14:51:25.185 23944 24296 I python  : [INFO   ] [Window      ] Provider: sdl2
[... 9 lines cut ...]
07-24 14:51:25.245 23944 24296 I python  : [DEBUG  ] [Shader      ] Vertex compiled successfully
07-24 14:51:25.270 23944 24296 I python  : [DEBUG  ] [ImageSDL2   ] Load </data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/site-packages/kivy/data/glsl/default.png>
07-24 14:51:25.284 23944 24296 I python  : [INFO   ] [Window      ] auto add sdl2 input provider
07-24 14:51:25.289 23944 24296 I python  : [INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
07-24 14:51:25.474 23944 24296 I python  : [DEBUG  ] [Resource    ] add </system/fonts> in path list
07-24 14:51:25.476 23944 24296 I python  : [DEBUG  ] [Resource    ] add </data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/site-packages/kivy/data/fonts> in path list
07-24 14:51:25.480 23944 24296 I python  : [DEBUG  ] [Base        ] Create provider from android
07-24 14:51:25.481 23944 24296 I python  : [WARNING] [Base        ] Unknown <android> provider
07-24 14:51:25.482 23944 24296 I python  : [INFO   ] [Base        ] Start application main loop
07-24 14:51:25.485 23944 24296 I python  : [INFO   ] [example     ]scanning
07-24 14:51:25.486 23944 24296 I python  : [DEBUG  ] Starting BTLE scan
07-24 14:51:25.701 23944 24296 I python  : [INFO   ] [GL          ] NPOT texture support is available
07-24 14:51:26.171 23944 24296 I python  : [INFO   ] [Base        ] Leaving application in progress...
[... cut ...] in __aenter__
07-24 14:51:26.183 23944 24296 I python  :    File "/home/nico/kivy/bleak_old/bleak/examples/kivy/.buildozer/android/platform/build-arm64-v8a/build/python-installs/bleakdemo/arm64-v8a/bleak/backends/p4android/scanner.py", line 108, in start
07-24 14:51:26.184 23944 24296 I python  :  AttributeError: 'android.bluetooth.le.ScanSettings$Builder' object has no attribute 'setPhy'
07-24 14:51:26.184 23944 24296 I python  : Python for android ended.
***************************************

dlech, 2022-07-24:
What is the Android version? This API was introduced in Android 8.0.

navr32, 2022-07-24:
The Android Version is 7.0

navr32, 2022-07-25:
Any help ?  I have  seen and test the project https://github.com/kai-morich/SimpleBluetoothLeTerminal and the app on the appstore this app work at first time with my bluetooth LE device and my Android phone on Android 7.0 i have scroll the code and i found any use of Setphy ...? Could be possible to have the same method for Bleak to be able to use phone with olders Android
with Bleak ?  many thanks.

dlech, 2022-07-25:
Android 8.0 is already 5 years old, so that seems like a reasonable minimum system requirement to me. But if someone wants to take responsibility for ensuring compatibility of older versions of Android, I wouldn't be opposed to a pull request that adds runtime version checks around newer apis.

navr32, 2022-07-26:
Perhaps this will be more nice to add a Legacy Backend for old Android. Because your code is pretty good i think this will be very annoying to add many runtime version checks ?  But problems if the app must run on anothers versions of android phone...how to manage on the same app the use of two API;....so the only solution perhaps will be runtime version check ...so very hard work..again..i haven't time for all this now..

closed 2025-12-28 by dlech