hbldh/bleak #908: Feature request: BleakGATTCharacteristic.description should honour 2901 descriptor

jackjansen, 2022-07-29:
Currently `BleakGATTCharacteristic.description` comes from the characteristic uuid, mapped through `uuidstr_to_str()`.

But if the characteristic has a `2901` descriptor that could also be used to obtain the description.

Is this a good idea? If it is I'll implement it and create a pull request.

I would be tempted to use the 2901 if it is available and the mapping as fallback, but it might be better to do it the other way around (only look at 2901 if `uuidstr_to_str()` doesn't know the uuid). Let me know.

dlech, 2022-07-29:
We seem to be on the same wavelength - or maybe it is because I just read a book about BLE. I was thinking about this too. My preference would be to prefer the descriptor if available and fall back to the well-known lookup if not.

jackjansen, 2022-07-30:
> or maybe it is because I just read a book about BLE. 

Reference please, assuming its a good book? The official documentation is baroque, and there's all this classic bluetooth legacy mixed in. And then every implementation decides to hide all sorts of things for "convenience" so that doesn't help either. I have the feeling that at the bottom there's only handles, and everything else is built on top of that. But I can't get my head around it...

dlech, 2022-07-30:
> Reference please, assuming its a good book?

*Bluetooth Low Energy: The Developer's Handbook* by Robin Heydon. It's a bit old (2012) so it doesn't cover developments since BLE was first released but it covers the fundamentals so 95% of the info is still relevant.