hbldh/bleak #913: Wild idea: use self-hosted runners and dedicated BLE device to test actual implementations

jackjansen, 2022-08-01:
I could volunteer to create a BLE device (esp32 board with fairly simple Nimble-Arduino based firmware) that has a couple of services and characteristics that are predefined, with semantics that are also predefined. Some characteristics would be read-only and return known values, some would be read-write, some would allow notify or indicate, etc.

Such a device could then be used to do actual tests of the bleak stacks (presuming the device is in range when running the tests).

And we could conceivably use self-hosted GitHub action runners to run these tests on macOS/windows/linux machines where we know that the BLE device is physically located right next to the runner machine. But even if this is a bridge too far the BLE device is still worth it: interested developers could get a $10 esp32 board and burn the firmware onto it and run the tests from their development machine.

Is this worth the effort?

Note that I'm volunteering to create the device, but not volunteering to write the tests, so someone else would need to do that...

dlech, 2022-08-01:
There is an older discussion about this topic that didn't end up with any results: https://github.com/hbldh/bleak/issues/207

A test device that can do "everything" to at least do a smoke test before releases would be helpful.

jackjansen, 2022-08-01:
Very interesting, and pretty much along the lines I was thinking. @bsiever are you still interested in this topic?