hbldh/bleak #917: Bleak on Android with Chaquopy Android sdk

satishgaurav, 2022-08-02:
* bleak version: 0.15.0
* Python version: 3.8.13 (default, May 10 2022, 19:17:45)
* Operating System: Android (Python SDK for Android) https://chaquo.com/chaquopy/

### Description
I am trying to discover active Bluetooth devices using the following script from an Android app. 

`import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())`

got the following error: 

[image]

satishgaurav, 2022-08-02:
android app repo: 
https://github.com/satishgaurav/bleak-chaquopy

dlech, 2022-08-02:
The current Android backend uses python-for-android. A new backend would need to be written for Chaquopy.

timrid, 2025-07-31:
I just want to document, that I am working on this topic right now.

The Android backends for `python-for-android` and `chaquopy` are so similar, that I want to mix them in one Backend (Android) with two Frameworks (Python-for-Android, Chaquopy). The Chaquopy Backend is working already, but currently I am trying to get the P4A Framework running... But unfortunately P4A is very slow when compiling (like 20min for one build) and poorly documented. That makes working with it very frustrating... Would it be possible to completely remove Python-For-Android support and switch to Chaquopy only?

My plan for the Chaquopy backend is to be able to test it completely in CI via integration tests. Here, the bleak code runs with the tests in an Android app in an Android emulator in a GitHub CI. Virtual Bluetooth devices are created in the Android emulator using `bumble` (see https://google.github.io/bumble/platforms/android.html). I can then use these virtual BLE devices in the integration tests. Basically, this already works, I just need to add more tests.

As soon as I have a version ready, I will create a PR.

dlech, 2025-07-31:
Yes, I think dropping P4A is reasonable. There are lots of unfixed bugs in that backend because it is difficult to work with and no one with enough spare time to work on it. And anyone who was actually using it can continue to use older versions of Bleak.

timrid, 2025-07-31:
Thanks, that's what I wanted to hear :) That saves me from spending more days struggling with Python for Android.

closed 2026-09-13 by dlech