hbldh/bleak #924: Pip installs Bleak 0.14.2 rather than 0.15.1

zapta, 2022-08-07:
I am running the pip install command on Windows 10 and am getting Bleak 0.14.2.  The latest version here is 0.15.1.   Which one is the latest stable version and if it's not 0.14.2, how can I get it?

```
$ pip install bleak
Requirement already satisfied: bleak in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (0.14.2)
Requirement already satisfied: bleak-winrt>=1.1.1 in c:\users\user\appdata\local\programs\python\python39\lib\site-packages (from bleak) (1.1.1)

$ pip show bleak
Name: bleak
Version: 0.14.2
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: c:\users\user\appdata\local\programs\python\python39\lib\site-packages
Requires: bleak-winrt
Required-by:
```

hbldh, 2022-08-07:
Since Bleak is already installed, pip will regard it as a situation where it does not have do anything. Use

```python
pip install -U bleak
```

to tell pip to upgrade if possible.

closed 2022-08-07 by hbldh

zapta, 2022-08-08:
Thanks Hendrik. I learned something new today.