hbldh/bleak #926: corebluetooth read_characteristic timeout is too low

bdraco, 2022-08-08:
Seen in https://github.com/Jc2k/aiohomekit/issues/136

Devices with a low MTU that talks HAP over bluetooth will fail to pair because the timeout is too lower

e1c4ced242bf2f054948b7baf367465dc9a16347 raised the timeout, but unfortunately it wasn't high enough.

I'm going to do some experiments to see if 15 is enough or if 20 is needed.  I suspect we are just close enough now.

BlueZ does not have the issue

bdraco, 2022-08-08:
I'm planning on working on a solution for this

bdraco, 2022-08-08:
```
2022-08-08 09:00:23.207 DEBUG (MainThread) [aiohomekit.controller.ble.client] Using fragment size: 101
2022-08-08 09:00:23.207 DEBUG (MainThread) [aiohomekit.controller.ble.client] Queuing fragment for write: b"\x00\x02\x1b\t\x00\xd0\x01\t\x01\x01\x01\xff\x06\x01\x03\x03\xff+\x1c\xaaD\xde\xc7\x85+w\xca\xb9\xea2^\x06\x84\xd8t\xb4V\xca\xcfO\xab\x1c\xd8m5\xbd$<\x19\xba%\xefi-V\xec\x9c\xc9F\xebe\x87\xa1Sz\xc8\xad_\xc6\xf1\xb8\xbdV-V\xae\xec'\xef_\xbc\x98\xf1iL'\xa9`\xe9K.\xcf\x03\x86\x8eM\x971\x9d\xa1&"
2022-08-08 09:00:23.207 DEBUG (MainThread) [aiohomekit.controller.ble.client] Queuing fragment for write: b'\x80\x1bv\xcb[$\xc7\xf1J\x19F=h\xd0tah\xc9\xbf\x12\xc1{S\x08H\xd3*\xfdy\xf3\xd0\xa6 #[\x10\x86\xa5z$\xd2X\xb4\xf5U\xef\x907\x0e\xbbo\x9aN\xddB\xfe\xb8\x80\\\x1f\x9c\xc5\xa7\xa2\x0b\x02\xaa\xf8\nu\xb4&\x84\xa7\xd5jX\xd2\x11\x1a\xd1\xce\xf2c\xf1U\x8a\x05\x97\x81\x15\xb5\x00\x99w\xd4\x91\xd6\xd8\xd9V'
2022-08-08 09:00:23.207 DEBUG (MainThread) [aiohomekit.controller.ble.client] Queuing fragment for write: b'\x80\x1b\xe60~\xb4\xa3n\xf2\xc3\xd0\x14\x99\xbc\xdb\xfc\xd9\xa0\xf4~\xcbx\xe4\x88l\xad\x8b\x7f\xb5\xaf"\x84\xb8\xc6\xae\x8dp;\xf0r\xe6~q\xbe\x13\xaa\xac\xb1\xd3\xa5X\xf0\xfe\x02\x9f\xbc\x14\x02\xccI\xc22\x91\x04\x82\x87\x03\xbd\x1f\x01\xcaAFC\xf7\xf8\x03\x81\xd9\x92[\xc6\xc3\xa8\x14\x8d\xc0\xf4]\xfe0n_\xe5\xd9\x01>Z\xac\x96\xca'
2022-08-08 09:00:23.207 DEBUG (MainThread) [aiohomekit.controller.ble.client] Queuing fragment for write: b"\x80\x1b\x89\x8e\xe5Z\x0f\x98\x9c\xb1`F7\x18\xf5%zv!P\xe3\xc09\x94\xd4\x8e\x1c\xf0\xe5\xf84\xde4\xfcJ\xbaJ\x15YD\x9a\xfd\xd3\xa4.\x92u\\K>\x02\x1e\xdc\x92\x13\xab\x1f\x8f\x05\xdd\xec\xdc#\xef\xf5|\x9cdMd\x9d\x0f\xeci\x95c\xa8\x0b\xcea\x0b\xc0\xe0\x8e \x90YrV\x13PzD']\xf1\x10\x91\x96\xb4\xab"
2022-08-08 09:00:23.207 DEBUG (MainThread) [aiohomekit.controller.ble.client] Queuing fragment for write: b"\x80\x1bz\xc7\xaf\x1b\xc8\xfdu\x04@\xaf\x9d\xc9\xee\xc74\xad+e\x81\xe8\x89\xdd\xed\xd3\xa3P\xf7\xb0\xd1\\\xa6\xbe\xa1'\xf0\xb2]\xb8&\xea\xa2\xb7\x89V\xbc2\x020\xa3\xa6\xd6\xc9\x95\x88\x91I}\x90rT\x0c\xbcgk\x06\xc3\xe4\xb3g5$\xe4z"
```

```
2022-08-08 09:00:40.394 DEBUG (MainThread) [aiohomekit.controller.ble.client] Read fragment: bytearray(b'\x02\x1b\x00\x7f\x04\x01\xff\x06\x01\x04\x04@\x01\xe53\xeb\xb5I\xd1;9bvA\xecV;\n\xb0\\\xcb\xaa\xa5\xbe\x14\xe2-\xb6\xd8y3\xa3M\x0f\x7f\xf0L~\xeb\x02\xd87\rB\x93|uF\xdb\xe8YWS\x9e\xcf\x1a\x97<\x90\xf4o\xadk5O\x9e\x05\xff\x8e\xa9\x13\x0bS<\x86\x8d\x9f\x86;\xe8|\x9f\xd3 \xe8\xb5\x85\xe4\xafZ\xcf\'\xce`f\xce\xaf]\t\x9e\xbe\xcc6\x01[r\x96]@\xd4)\xdd9\x0c\xb2r\xc3X\xab\xd9z\x16z\x0e$3\xb15\xca\xd1\x9b\xc2\x9a7<h/\xab`\xab`\x14\x9ef\n}n\x05\xf84\xf2\xe60L\xd7\xc8\xb9W7\x9fU\x8c\x026K7FL\x91\x93\xb9R\x8cELn\x9c&\xe4\xd9Z{\x0c\xd0\x81F\xe4A\xab\xcf\xb2w\xa8F\xc2\x94m\xcar\x8d\xa3Z6\xba\x05\xf4\x9a,.;_\x05bEx\xd3v\xe5\x9c\xd6E\x07\xae\xfa\xb3\xe4\xc1;k\x14Mn\x0e\x0e#\xb1\n\xae_\xf3/\x12D\xca"\x8e\x1dos\x88oU\x01\xff\x06=\nS9W\xa0\x9a>Nc\x14yN\x1b\x1cr\xf3\xba3Y\xcc\x1a\x00\xd6<j\xbb\x13\x82es\x9f_C2\x1c\xd0\x89\x19\xdc\xbd\x9c\xad\xd8jA\xec\xe3\xe5\x95\x12?\x06\x99\xbc1\x9c\xcb\x12yo\x87\xea\xc2\xb2\xf5\x97`\xe6G\x05\xff\x98\xe5\xbe\xcf)\xed9L\xe7P\x04\x13(\xec\xea\xe96};T\x8e>\xffX\xd0\xeb\xba\x85e\t\xc0\xf7\x85\xaf\nc\x86H\x8c\xa3\x8f\x05e\xa6:\x12\x02')

```

It actually takes 17s to respond

closed 2022-08-26 by dlech

cmrenus, 2022-10-31:
@bdraco Question, have the changes for this been merged into HA yet? And if so, do you know which version?

Is there also any reason here to not expose the timeout as an environment variable or so? To make it more configurable? Just curious mainly, but facing similar issues linked through this fix of Bluetooth homekit devices timing out

PR #928 (2022-08-26): Adjust default timeout for ``read_gatt_char()`` with CoreBluetooth to 20s

fixes #926
commit 9a1f31da (2022-08-26): Adjust default timeout for ``read_gatt_char()`` with CoreBluetooth to 20s

fixes #926