hbldh/bleak #929: Issue when using bleak on linux when everything worked well on windows

aefrtyi, 2022-08-09:
* bleak version: 0.15.1
* Python version: 3.9
* Operating System: Debian GNU/Linux 11 (bullseye)
* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.55

### Description

I have a ble peripheral device that has a custom scan response data in EIR 0x42. I previously used bluepy on linux to scan and communicate with it, but I wanted to be able to use my code on both windows and linux, so I switch to bleak.

I managed to scan my device with the expected data on windows and communicate with it successfully, however, when I try to do so on linux, BleakClient.get_services() returns an empty dict, and I am unable to find the adv data I'm looking for.

### What I Did

Scanner:
```python
import asyncio
import json
import platform
import struct
from typing import Dict, List, Tuple

from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData

CURRENT_SYSTEM: str = platform.system()

def __format_adv_data(device: BLEDevice) -> Dict[bytes, bytes]:
        adv_data: dict = dict()
        if CURRENT_SYSTEM == "Windows":
            for detail in device.details:
                # device possess advertising details that are enumerable. In some cases, some of those details may be empty
                if detail is not None:
                    # if detail is not empty, then get the adv data from the details obtained
                    for data in detail.advertisement.data_sections:
                        unpacked_adv_data = [dt[0] for dt in struct.iter_unpack('<B', data.data)]
                        # Convert adv data to match correct format to use later on
                        bytes_adv_data = b"".join([(dt.to_bytes(1, "big")) for dt in unpacked_adv_data])
                        data_type = int(data.data_type).to_bytes(1, 'big')
                        adv_data[data_type] = bytes_adv_data
        elif CURRENT_SYSTEM == "Linux":
            detail = device.details.get('props')
            print(f"details {device.details} | type {type(device.details)}")
                # print(json.loads(detail).get('props'))
        return adv_data

def detection_callback(device: BLEDevice, adv_data: AdvertisementData):
    if device.address.lower() == '53:4e:48:00:00:07':
        if CURRENT_SYSTEM == "Windows":
            _, raw_adv_data = adv_data.platform_data
            new_adv_data, scan_data = raw_adv_data
            if new_adv_data is not None and scan_data is not None:
                formated_adv_data = dict()
                # if detail is not empty, then get the adv data from the details obtained
                for data in new_adv_data.advertisement.data_sections:
                    unpacked_adv_data = [dt[0] for dt in struct.iter_unpack('<B', data.data)]
                    # Convert adv data to match correct format to use later on
                    bytes_adv_data = b"".join([(dt.to_bytes(1, "big")) for dt in unpacked_adv_data])
                    data_type = int(data.data_type).to_bytes(1, 'big')
                    formated_adv_data[data_type] = bytes_adv_data
                for data in scan_data.advertisement.data_sections:
                    unpacked_adv_data = [dt[0] for dt in struct.iter_unpack('<B', data.data)]
                    # Convert adv data to match correct format to use later on
                    bytes_adv_data = b"".join([(dt.to_bytes(1, "big")) for dt in unpacked_adv_data])
                    data_type = int(data.data_type).to_bytes(1, 'big')
                    formated_adv_data[data_type] = bytes_adv_data
                print(formated_adv_data)
        elif CURRENT_SYSTEM == "Linux":
            print(f"local_name: {adv_data.local_name}")
            print(f"manufacturer_data: {adv_data.manufacturer_data}")
[... 173 lines cut ...]
    await bleak_client.disconnect()
    print('Disconnected from bleak client')

if __name__ == '__main__':
    asyncio.run(main())
```

As for the discovery of services, using the example get_services(), just changing the mac address, I obtain the following:
```bash
root@iot-gate-imx8:/home/compulab/code/GatewaySmartphone# pipenv run python get_services.py
Services:

```

I did find when reading the bluetooth service logs
```bash
Aug 09 07:53:36 iot-gate-imx8 bluetoothd[505]: src/device.c:load_gatt_db() No cache for 53:4E:48:00:00:07
```
but I'm unsure if it's the related to my issue.

I'm also attaching the bleak logs I got during scan and get_services() as well as the bluetooth trafic captured with tshark in case it could help.
[bleak_log_comm.log](https://github.com/hbldh/bleak/files/9288824/bleak_log_comm.log)
[wireshark_scan.log](https://github.com/hbldh/bleak/files/9288825/wireshark_scan.log)
[wireshark_comm.log](https://github.com/hbldh/bleak/files/9288826/wireshark_comm.log)
[bleak_log_scan.log](https://github.com/hbldh/bleak/files/9288827/bleak_log_scan.log)

dlech, 2022-08-09:
From the logs, we can see that BlueZ does not add D-Bus objects for the services and characteristics. BlueZ hides certain services, so it could be possible that the device doesn't have any services that are not hidden. Or there could be issues with the BlueZ cache which can be fixed by removing the device from BlueZ and manually deleting a cache file (see troubleshooting section of the docs).

aefrtyi, 2022-08-10:
I've checked but as far as I know, my device does not have hidden services or characteristics.
I followed the troubleshooting section to clear the cache, but it seems my device is not in cache at all:
```bash
[bluetooth]# remove 53:4E:48:00:00:07
Device 53:4E:48:00:00:07 not available
```

```bash
root@iot-gate-imx8:/var/lib/bluetooth/14:F6:D8:45:10:04/cache# ls
00:21:AD:13:13:20  90:78:B2:A8:BB:4E  90:FD:9F:A7:EE:EF  D8:4D:72:F9:22:2D  ED:7B:67:62:26:66
38:18:4C:BF:37:05  90:FD:9F:A7:ED:39  CC:98:8B:E0:6A:3A  E4:B3:66:BE:30:94  F0:2E:89:CE:50:4E
```

aefrtyi, 2022-08-10:
I do not have a GUI on the linux device I used, so I used tshark instead of wireshark. I already attached the results in my first post.
Bellow are the results I got on windows using the same scripts
[wireshark_comm_windows.log](https://github.com/hbldh/bleak/files/9302521/wireshark_comm_windows.log)
[wireshark_scan_windows.log](https://github.com/hbldh/bleak/files/9302522/wireshark_scan_windows.log)
.

dlech, 2022-08-10:
The linux "wireshark" logs don't include the actual packet data, so are as useful as they could be. The windows logs seem to contain TCP data, not bluetooth.

aefrtyi, 2022-08-10:
I'll see if I can get the packet data. For windows, I tried all interfaces shown by wireshark and I used the only one that had data when I ran my scripts. The only other interface that had data was the wifi one

dlech, 2022-08-10:
There is a special external program from Microsoft required on Windows. See the Bleak troubleshooting docs for details.

aefrtyi, 2022-08-10:
My bad, completely missed this part.
Here are the files with packet data
[tshark_comm_linux.log](https://github.com/hbldh/bleak/files/9302798/tshark_comm_linux.log)
[wireshark_scan_windows.log](https://github.com/hbldh/bleak/files/9302799/wireshark_scan_windows.log)
[wireshark_comm_windows.log](https://github.com/hbldh/bleak/files/9302801/wireshark_comm_windows.log)
[tshark_scan_linux.log](https://github.com/hbldh/bleak/files/9302802/tshark_scan_linux.log)
.

dlech, 2022-08-10:
Thanks. From the `tshark_comm_linux.log`:

[image]

BlueZ requests an MTU of 512, but the device replies that it wants to use 517. So BlueZ agrees and asks if the device wants to use 517, but then the device says that is invalid. BlueZ then disconnects. So the problem appears to be with the device.

On Windows, we can see the MTU exchange was successful.

[image]