hbldh/bleak #947: Error while connecting to BLE ESP32 device

vChavezB, 2022-08-19:
* bleak version: 0.15.1
* Python version: 3.92
* Operating System: Linux raspberrypi 5.15.32-v7l+
* BlueZ version (`bluetoothctl -v`) in case of Linux:  5.55

### Description

I am trying to connect to a custom ESP32 BLE device  but I get the following error

[org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

Code 
https://pastebin.com/Dj5PWFSk

What the code does is scans devices, and if it finds that they are compatible with a MAC filter it attempts connection and discovers it services and characteristics. However, when attempting connection (line 52) it fails most of the times.

 Log

```
pi@raspberrypi:~/ble/python_Central $ python example.py
DEBUG:root:start discovery
DEBUG:root:end discovery
DEBUG:root:Checking compatibility device ESP32 [0/24]
attempt connection  ESP32
[Exception BLE connection]: [org.bluez.Error.Failed] Software caused connection abort
DEBUG:root:Checking compatibility device ESP32 [2/24]
attempt connection  ESP32
[Exception BLE connection]: [org.bluez.Error.Failed] Software caused connection abort
DEBUG:root:Checking compatibility device ESP32 [4/24]
attempt connection  ESP32
[Exception BLE connection]: [org.bluez.Error.Failed] Software caused connection abort
DEBUG:root:Checking compatibility device ESP32 [5/24]
attempt connection  ESP32
[Exception BLE connection]: [org.bluez.Error.Failed] Software caused connection abort
DEBUG:root:start discovery
DEBUG:root:end discovery
DEBUG:root:Checking compatibility device ESP32 [0/22]
attempt connection  ESP32
[Exception BLE connection]: [org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

DEBUG:root:Checking compatibility device ESP32 [1/22]
attempt connection  ESP32
[Exception BLE connection]: [org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

DEBUG:root:Checking compatibility device ESP32 [2/22]
attempt connection  ESP32
[Exception BLE connection]: [org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

DEBUG:root:Checking compatibility device ESP32 [4/22]
attempt connection  ESP32
[Exception BLE connection]: [org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

DEBUG:root:Checking compatibility device ESP32 [6/22]
attempt connection  ESP32
[Exception BLE connection]: [org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist
```

dlech, 2022-08-19:
Code copied from pastebin:

```python
import asyncio
import logging
import logging.config
from bleak import BleakScanner, BleakClient

total_conns = 0
MAX_CONNECTIONS = 5
logging.config.dictConfig({
    'version': 1,
    'disable_existing_loggers': True, ##Disable BLEAK logger
})

logging.basicConfig(level=logging.DEBUG)

class Sensor:
    Svc = "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
    Char = "beb5483e-36e1-4688-b7f5-ea07361b26a8"
    ReadyChar = "beb5483e-36e1-4688-b7f5-ea07361b26a7"
    MAC_FILTER = "3C:61:05"
    @staticmethod
    def isCompat(Services):
        compatible = False
        for svc in Services:
            if svc.uuid == Sensor.Svc:
                for char in svc.characteristics:
                    if char.uuid == Sensor.Char:
                        compatible = True
        return compatible
    def __init__(self,client):
        self._client = client

def disconnect_callback(device_client):
    global total_conns
    # fires when a device is disconnected
    total_conns -= 1
    print("Disconnected from {0}".format(device_client.address))

async def discover():
    global total_conns, MAX_DEVICES
    logging.debug('start discovery')
    devices = await BleakScanner.discover(timeout=2.0)
    logging.debug('end discovery')
    total_devices = len(devices)
    for count, dev in enumerate(devices):
        if dev.address[:8] == Sensor.MAC_FILTER and total_conns < MAX_CONNECTIONS:
            msg = "Checking compatibility device %s [%d/%d]" % (dev.name, count, total_devices)
            logging.debug(msg)
            try:
                client = BleakClient(dev)
                client.set_disconnected_callback(disconnect_callback)
                print("attempt connection ", dev.name)
                connected = await client.connect(timeout=2.0)
                if connected:
                    logging.debug("Getting services")
                    services = await client.get_services()
                    if not Sensor.isCompat(services):
                        await client.disconnect()
                    else:
                        total_conns +=1
            except Exception as e:
                print("[Exception BLE connection]:", str(e))
                await asyncio.sleep(3)
                pass

async def main():
    while True:
        await discover()

asyncio.run(main())

```

dlech, 2022-08-19:
>     [Exception BLE connection]: [org.bluez.Error.Failed] Software caused connection abort

These sorts of errors come from BlueZ, not Bleak. Have you tried logging Bluetooth packets as described in the Bleak docs troubleshooting page?

>     [Exception BLE connection]: [org.freedesktop.DBus.Error.UnknownObject] Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist

We should probably make better error message for this. It means that too much time has passed between the end of scanning and the connection attempt and the device has been removed from BlueZ.

Instead of scanning for many devices and then trying to connect to them all later, you may have better results if you use `BleakScanner.find_device_by_filter()` instead to discover devices one at a time, then connect to that one device, then scan again for another device.

vChavezB, 2022-08-19:
Thanks for the info. I will check more in detail the logging I get from the DBUS with the BlueZ tools.

I was not aware of the timeout between scans and connection. I will try the method proposed and post the results.

Another side question that I would like to ask:

Is it possible to detect which devices are already connected and obtain their a Bleak.BLEDevice object without discovery? This could happen if I connected previously with the device outside of Bleak or I interrupted my script. I was looking if there was a __del__()  method for devices connected so they can be disconnected once the script ends.

dlech, 2022-08-19:
> Is it possible to detect which devices are already connected and obtain their a `Bleak.BLEDevice` object without discovery?

Not currently. Please see #367.

> `__del__()`

`__del__()` is not guaranteed to run at exit - it is not the right tool for the job. Using `async with BleakClinet() as client:` in conjunction with `asyncio.run()` will cause devices to disconnect at exit.

vChavezB, 2022-08-19:
This is the log that I got after using `find_device_by_filter` and trying to connect to the device object that was returned by the function. 

```
attempt connection  ESP32
2022-08-19 17:47:19,778 bleak.backends.bluezdbus.client DEBUG: Connecting to device @ 3C:61:05:3E:0D:6E with hci0
DEBUG:bleak.backends.bluezdbus.client:Connecting to device @ 3C:61:05:3E:0D:6E with hci0
2022-08-19 17:47:19,797 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
2022-08-19 17:47:19,803 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_68_72_C3_DC_DF_91', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 68:72:C3:DC:DF:91)>, 'AddressType': <dbus_next.signature.Variant ('s', public)>, 'Alias': <dbus_next.signature.Variant ('s', 68-72-C3-DC-DF-91)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -83)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {117: <dbus_next.signature.Variant ('ay', b'B\x04\x01\x80fhr\xc3\xdc\xdf\x91jr\xc3\xdc\xdf\x90\x01\x00\x00\x00\x00\x00\x00')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_68_72_C3_DC_DF_91', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 68:72:C3:DC:DF:91)>, 'AddressType': <dbus_next.signature.Variant ('s', public)>, 'Alias': <dbus_next.signature.Variant ('s', 68-72-C3-DC-DF-91)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -83)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {117: <dbus_next.signature.Variant ('ay', b'B\x04\x01\x80fhr\xc3\xdc\xdf\x91jr\xc3\xdc\xdf\x90\x01\x00\x00\x00\x00\x00\x00')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-08-19 17:47:19,835 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_3C_61_05_3E_A7_D2): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -41)>, 'TxPower': <dbus_next.signature.Variant ('n', 3)>}, []]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_3C_61_05_3E_A7_D2): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -41)>, 'TxPower': <dbus_next.signature.Variant ('n', 3)>}, []]
2022-08-19 17:47:20,032 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_3C_61_05_30_B6_D2): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -54)>, 'TxPower': <dbus_next.signature.Variant ('n', 3)>}, []]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_3C_61_05_30_B6_D2): ['org.bluez.Device1', {'RSSI': <dbus_next.signature.Variant ('n', -54)>, 'TxPower': <dbus_next.signature.Variant ('n', 3)>}, []]
2022-08-19 17:47:20,034 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_3C_61_05_3E_16_26', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 3C:61:05:3E:16:26)>, 'AddressType': <dbus_next.signature.Variant ('s', public)>, 'Name': <dbus_next.signature.Variant ('s', ESP32)>, 'Alias': <dbus_next.signature.Variant ('s', ESP32)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -38)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', ['4fafc201-1fb5-459e-8fcc-c5c9c331914b'])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'TxPower': <dbus_next.signature.Variant ('n', 3)>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
[... cut ...]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_3C_61_05_3E_0D_6E): ['org.bluez.Device1', {'Connected': <dbus_next.signature.Variant ('b', True)>}, []]
2022-08-19 17:47:20,540 bleak.backends.bluezdbus.client DEBUG: _cleanup_all(/org/bluez/hci0/dev_3C_61_05_3E_0D_6E)
DEBUG:bleak.backends.bluezdbus.client:_cleanup_all(/org/bluez/hci0/dev_3C_61_05_3E_0D_6E)
[Exception BLE connection]: [org.bluez.Error.Failed] Software caused connection abort
```

vChavezB, 2022-08-19:
This is what I got from btmon

```
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                #91 [hci0] 7.070694
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                                                                                                                                #92 [hci0] 7.245839
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Master (0x00)
        Peer address type: Public (0x00)
        Peer address: 3C:61:05:31:62:9E (Espressif Inc.)
        Connection interval: 48.75 msec (0x0027)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Master clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 44                                                                                                                     {0x0001} [hci0] 7.245867
        LE Address: 3C:61:05:31:62:9E (Espressif Inc.)
        Flags: 0x00000008
          Unknown device flag (0x00000008)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): ESP32
        TX power: 3 dBm
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific (4fafc201-1fb5-459e-8fcc-c5c9c331914b)
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2                                                                                                         #93 [hci0] 7.246056
        Handle: 64
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                #94 [hci0] 7.246549
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> HCI Event: Command Complete (0x0e) plen 14                                                                                                                             #95 [hci0] 7.247355
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
        00 00 00 00 00 00 00 00 00 00                    ..........
> HCI Event: LE Meta Event (0x3e) plen 12                                                                                                                                #96 [hci0] 7.580895
      LE Read Remote Used Features (0x04)
        Status: Connection Failed to be Established (0x3e)
        Handle: 64
        Features: 0x3f 0x00 0x00 0x08 0x00 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Slave-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
          Remote Public Key Validation
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                           #97 [hci0] 7.581461
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Failed to be Established (0x3e)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                   {0x0001} [hci0] 7.595317
        LE Address: 3C:61:05:31:62:9E (Espressif Inc.)
        Reason: Unspecified (0x00)

```

vChavezB, 2022-08-19:
After some more debugging I think I found the error. My setup consists of 10 ESP32 MCUs connected in parallel to the same power supply. After connecting each of them to a separate power source it seems that the connection is established in a more stable manner.
So it seems I was having a harware error... Im gonna close this issue

vChavezB, 2022-08-20:
Im reopening as the error still persists. It worked once but not anymore. I will try to dig deeper

closed 2022-08-20 by vChavezB