hbldh/bleak #951: The _remove_device_watcher needs to be canceled when the connection failed

bdraco, 2022-08-20:
```
2022-08-20 11:12:01.209 DEBUG (MainThread) [yalexs_ble.lock] Front Door (M10CK0F): Disconnected from lock callback
2022-08-20 11:12:01.210 DEBUG (MainThread) [yalexs_ble.lock] Front Door (M10CK0F): Disconnected from lock callback
2022-08-20 11:12:01.211 DEBUG (MainThread) [yalexs_ble.lock] Front Door (M10CK0F): Disconnected from lock callback
```

I noticed I was getting the callback multiple times and tracked it down to the `_remove_device_watcher` not being canceled when the connection fails

https://github.com/hbldh/bleak/blob/456cea854823ef13c55efcaa5e0fba380a055d8c/bleak/backends/bluezdbus/client.py#L152

dlech, 2022-08-26:
I don't see how this is possible. Immediately after line 152, there is a try/except block that calls `_cleanup_all` on any exception. `_cleanup_all` calls `self._remove_device_watcher()`.

How exactly is the connection failing?

bdraco, 2022-08-26:
It happens after the connection timed out.  I expect the its raising CancledError but I would have expected BaseException catching would have caught that and triggered the `_cleanup_all`

I have not yet been able to replicate it on demand.

bdraco, 2022-08-26:
So I think part of what's going on is (and this is a tricky race so I may still be missing something):

Disconnected Monitor is started
https://github.com/hbldh/bleak/blob/8515d22dca74a8a6ca6316f4debda1bef933cb2d/bleak/backends/bluezdbus/client.py#L202

`await self.get_services()` is called and the timeout gets hit, raising CanceledError
https://github.com/hbldh/bleak/blob/8515d22dca74a8a6ca6316f4debda1bef933cb2d/bleak/backends/bluezdbus/client.py#L205

`_cleanup_all()` is called
https://github.com/hbldh/bleak/blob/8515d22dca74a8a6ca6316f4debda1bef933cb2d/bleak/backends/bluezdbus/client.py#L209

Device completes connection in the background is is connected but we think it is not connected because the timeout was raised.

The next connection attempt things are in a indeterminate state since the device is actually connected but we don't know that.

bdraco, 2022-08-26:
So I think the race is that the cancel happens during `get_services`

Then `_cleanup_all` runs it calls `self._bus.disconnect()`
https://github.com/hbldh/bleak/blob/8515d22dca74a8a6ca6316f4debda1bef933cb2d/bleak/backends/bluezdbus/client.py#L257

And set `self._bus` to `None`
https://github.com/hbldh/bleak/blob/8515d22dca74a8a6ca6316f4debda1bef933cb2d/bleak/backends/bluezdbus/client.py#L269

After that the `_disconnect_monitor` runs and the bus is already disconnected so it leaves the connection open
https://github.com/hbldh/bleak/blob/8515d22dca74a8a6ca6316f4debda1bef933cb2d/bleak/backends/bluezdbus/client.py#L223

bdraco, 2022-08-26:
And I just realized why I can't replicate it anymore.

Since I have `get_services` wrapped to disconnect on Exception it never happens anymore

https://github.com/Bluetooth-Devices/bleak-retry-connector/blob/acbee3824e4200f4acb43851ff90ccfd60ecb575/src/bleak_retry_connector/__init__.py#L160

PR #968 (2022-08-30): BlueZ: fix not disconnecting if get_services() throws during connect
It is possible for get_services() to raise an exception, e.g. if external code cancels the task. If this happens, we need to be sure
that the device disconnects so the BlueZClient is in a consistent state (if the connect() method raises and exception, is is expected that the device will not be connected).

To fix this, we can just move the call to get_services() inside of the existing try block that will disconnect on any exception, then reraise the exception.

Fixes #951.
commit b93082a6 (2022-08-30): backends/bluezdbus/client: fix not disconnecting if get_services() throws

It is possible for get_services() to raise an exception, e.g. if
external code cancels the task. If this happens, we need to be sure
that the device disconnects so the BlueZClient is in a consistent
state (if the connect() method raises and exception, is is expected
that the device will not be connected).

To fix this, we can just move the call to get_services() inside of the
existing try block that will disconnect on any exception, then reraise
the exception.

Fixes #951.
PR #969 (2022-08-31): release v0.16.0

Added
-----
* Added ``BleakGattCharacteristic.max_write_without_response_size`` property. Fixes #738.

Fixed
-----
* Fixed regression in v0.15 where devices removed from BlueZ while scanning
  were still listed in ``BleakScanner.discovered_devices``. Fixes #942.
* Fixed possible bad connection state in BlueZ backend. Fixes #951.

Changed
-------
* Made BlueZ D-Bus signal callback logging lazy to improve performance. Merged #912.
* Switch to using async_timeout instead of asyncio.wait_for for performance. Merged #916.
* Improved performance of ``BlueZManager.get_services()``. F [... cut]
closed 2022-08-30 by dlech