hbldh/bleak #958: BleakDBusError[org.bluez.Error.InProgress] Operation already in progress

des-gan09, 2022-08-23:
* bleak version: 0.14.3
* Python version: 3.9.2
* Operating System: Raspberry Pi 3B+ (Raspbian Bullseye) 
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description

I am trying to send a command and read from a sensor. I used a modified uart_example which works for the Pi 4s but for the Pi 3s, I sometimes encounter this issue. 
```Traceback (most recent call last):
  File "/home/pi/powersensor-firmware/sampling/debug.py", line 115, in <module>
    asyncio.run(uart_terminal())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/powersensor-firmware/sampling/debug.py", line 49, in uart_terminal
    device = await BleakScanner.find_device_by_address(MAC, timeout=60)
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/scanner.py", line 220, in find_device_by_address
    return await cls.find_device_by_filter(
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/scanner.py", line 253, in find_device_by_filter
    return None
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/scanner.py", line 99, in __aexit__
    await self.stop()
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/scanner.py", line 177, in stop
    assert_reply(reply)
  File "/home/pi/.local/lib/python3.9/site-packages/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
```

Here's my code

```
import asyncio
from gettext import find
from http import client
import sys
import logging
import os

from bleak import BleakScanner, BleakClient
from bleak.backends.scanner import AdvertisementData
from bleak.backends.device import BLEDevice
from datetime import datetime
import time
import queue

MAC = sys.argv[1].upper().rstrip('\n')
now = datetime.now()
dt_string = now.strftime("%Y-%m-%dT%H:%M:%S")
name = 'power/' + dt_string + '.txt'
interval = 'time/' + dt_string + '.txt'

UART_SERVICE_UUID = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
UART_RX_CHAR_UUID = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
UART_TX_CHAR_UUID = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

# All BLE devices have MTU of at least 23. Subtracting 3 bytes overhead, we can
# safely send 20 bytes at a time to any device supporting this service.
UART_SAFE_SIZE = 20
log_data = []
flag = 0

async def uart_terminal():

    disconnected_event = asyncio.Event()
    start = int(round(time.time()))
    device = await BleakScanner.find_device_by_address(MAC, timeout=60)
    
    def handle_rx(_: int, data: bytearray):
        global flag
        byteToString = data.decode("utf-8")
        sep = '\n'
        mag_data = byteToString.split(sep, 1)[0]
        if len(mag_data) == 1:
            disconnected_event.set()
        if len(mag_data) > 1:
            logging.info("%s", mag_data, extra={'simple': True})
            
    while True:
        try: 
            async with BleakClient(device, timeout=60) as client:
                await client.start_notify(UART_TX_CHAR_UUID, handle_rx)

                logging.basicConfig(
                    filename=name,
                    filemode='w',
                    level=logging.INFO,
                    format='%(message)s'
                )

                logging.info("%s", MAC, extra={'simple': True})

                print("Connected to " + MAC)
                data = b'sample 2000\n'

                print("Writing to", MAC)
                await client.write_gatt_char(UART_RX_CHAR_UUID, data)
                
                #print("Disconnecting from device")
                await disconnected_event.wait()
                print("Stopping notification", MAC)
                await client.stop_notify(UART_TX_CHAR_UUID)
                await client.disconnect()
                end = int(round(time.time()))
                print("Disconnecting", MAC)
                timing = str(end - start)
                print(timing,MAC)
                break
        except:
            print("Reconnecting...",MAC)
        

if __name__ == "__main__":
    # while(1):
    try:
        asyncio.run(uart_terminal())
    except asyncio.CancelledError:
        # task is cancelled on disconnect, so we ignore this error
        pass
```

dlech, 2022-08-26:
Since the error comes from BlueZ, I would suggest logging Bluetooth packets using Wireshark to see what is going on.

dlech, 2022-09-03:
Is this still a problem in Bleak 0.16?

des-gan09, 2022-09-06:
Sorry for the late reply. I have managed to fixed it by updating Bluez to 5.64. I'll try out the new bleak version and see there's any improvement

dlech, 2022-09-12:
Closing since this appears to be a BlueZ bug.

closed 2022-09-12 by dlech

wahadatjan, 2024-07-04:
@dlech if i am connected with two or more devices then i need the get data from the different device characterisctics but it gives one at a time and gives error in the rest of the devices