hbldh/bleak #961: Using detection

damianjwilliams, 2022-08-25:
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

dlech, 2022-08-25:
If you have a question, please start a discussion: https://github.com/hbldh/bleak/discussions

damianjwilliams, 2022-08-25:
Oop, I accidentally posted the issue in the middle of writing it. I've updated it now.

dlech, 2022-08-25:
> I know that when using MacOS 12.4, service UUIDs need to be provided to read advertising data (as discussed [extensively](https://github.com/hbldh/bleak/pull/692), but I am still struggling to get it to work.

Apple fixed the bug in 12.3, so it is no longer required.

> Am I missing something obvious?

Not that I can see. I've confirmed it is also not working on 12.5.1. I wonder if Apple introduced a new bug?

As a workaround, you can just leave out the service uuids.

dlech, 2022-08-25:
After trying it a few more times, it is working again (I also started the Packet Logger app in the meantime). Not sure what could have been the difference from the times when I was able to reproduce the issue.

damianjwilliams, 2022-08-25:
I just updated my MacOS from 12.4 to 12.5.1 and it seems to be working now! Thanks for your help.

closed 2022-08-25 by damianjwilliams