hbldh/bleak #970: Can't get data from two devices at the same rate

mulkkyul, 2022-09-01:
* bleak version: 0.15.1
* Python version: 3.8.0
* Operating System: Ubuntu 18.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description
Hi! 
First of all, thank you for the great package! It has helps me a lot working on my project.

I'm trying to get sensor readings from multiple IMU sensors.
I can connect two devices, start notify and receive data from them simultaneously.

The issue I'm having is that I don't get data from the devices at the same rate.
It seems that I get more data from the one that started notify later.

The ratio is roughly 1:3. 
I mean, when I receive 1 sensor reading from the device A, I receive 3 sensor readings from the device B
(when device B started notify later)

I'm wondering anyone can give me some advice.
Thanks!

### What I Did
Here is my code. 
I also tried the code in https://github.com/hbldh/bleak/issues/630, but still I don't get the data with the same rate.

```

import asyncio
from bleak import BleakClient, BleakScanner
import time
from functools import partial
from typing import Sequence
from bleak.backends.device import BLEDevice

NUM_DEVICES = 2
myDevices = [None] * NUM_DEVICES
myDevices[0] = "97:A7:3E:33:F3:D9"
myDevices[1] = "6B:CC:19:37:4A:79"
quaternionCharacteristic = "19b10000-0003-537e-4f6c-d104768a1214"

def callback_imu(client: BleakClient, sender: int, data: bytearray):
    print(int(time.time()*1000), client.address)

async def connect_to_device():
    scanner = BleakScanner()
    list_address = []
    devices: Sequence[BLEDevice] = await scanner.discover(timeout=3.0)
    for d in devices:
        if(d.address in myDevices):
            list_address.append(d.address)

    if(len(list_address) <1):
        print("Found no NICLA SENSE ME. Try reboot them.")
        exit()
    client = [None] * len(list_address)
    for idx in range(len(list_address)):
        client[idx] = BleakClient(list_address[idx])
        try:
            await client[idx].connect()
        except Exception as e:
            print('error: ', e, end='')
            exit()
        await client[idx].start_notify(quaternionCharacteristic, partial(callback_imu, client[idx]))

    for i in range(3):
        await asyncio.sleep(1)

    for idx in range(len(list_address)):
        await client[idx].disconnect()

    print('disconnected.')

async def main():
    futures = [asyncio.ensure_future(connect_to_device())]
    await asyncio.gather(*futures)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    loop.close()

```

dlech, 2022-09-01:
Have you tried logging Bluetooth packets with Wireshark (see troubleshooting page in Bleak docs) to see what the Bluetooth adapter is doing? Since two devices are trying to transmit at the same time, it could be that one device is just winning the race more often than the other and isn't transmitting as often.

mulkkyul, 2022-09-06:
Thank you for the pointer! 

I logged the packets using Wireshark, and as you said one device was "winning the race".

Although two identical IMU devices run the same code, the one that starts notify first tends to lose the race. I guess some packets were lost?

Anyway, I guess this is not bleak issue so I'll close the issue. Thank you!

closed 2022-09-06 by mulkkyul