hbldh/bleak #972: startNotify fails with message 0x0e (Connection Rejected Due To Security Reasons)

fsaris, 2022-09-02:
* bleak version: 0.15.1
* Python version: 3.10
* Operating System: Ubuntu
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description

I'm porting some code to control lights from bluepy to bleak.

I got the connection and controlling the device to work. But when I want to start listening for notifications I get the notification through the callback but it fails after a short time with the message ` [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Connection Rejected Due To Security Reasons)`.

 
### What I Did

A simplified version of the code:

```
        def disconnected_callback(client):
            _LOGGER.debug("Disconnected callback called!")

        self._client = BleakClient(self.mac, disconnected_callback=disconnected_callback)

        # Connect
        await self._client.connect()

        _LOGGER.debug("Connected to %s", self.mac)

        _LOGGER.debug("Sent pair message"
        await self._client.write_gatt_char(PAIR_CHAR_UUID, message)

        _LOGGER.debug("Read pair response")
        value = await self._client.read_gatt_char(PAIR_CHAR_UUID)

... process pair to some session info

        # Tell the device to start sending out notifications
        await self._client.write_gatt_char(STATUS_CHAR_UUID, b'\x01')

        def callback(sender: int, data: bytearray):
            _LOGGER.debug(f"callback: {sender}: {data}")
        
        _LOGGER.debug("Start notify")
        await self._client.start_notify(STATUS_CHAR_UUID, callback)
```

This results in:

```
DEBUG:awoxmeshlight:Connected to a4:c1:38:8a:ba:94
DEBUG:awoxmeshlight:Sent pair message
DEBUG:bleak.backends.bluezdbus.client:Write Characteristic 00010203-0405-0607-0809-0a0b0c0d1914 | /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char001a: bytearray(b'\x0c\r8\x84\x019g\xf2\r\x91\xc4\xa3\xb0,Z\xda$')
DEBUG:awoxmeshlight:Sent status message
DEBUG:bleak.backends.bluezdbus.client:Write Characteristic 00010203-0405-0607-0809-0a0b0c0d1911 | /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011: b'\x01'
DEBUG:awoxmeshlight:Read pair response
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char001a): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\r\xc0$\xad\xc6\tmI\x94W\xea\x1f\x0f\xfc\xf0sh')>}, []]
INFO:bleak.backends.bluezdbus.client:value changed: /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char001a
DEBUG:bleak.backends.bluezdbus.client:Read Characteristic 00010203-0405-0607-0809-0a0b0c0d1914 | /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char001a: bytearray(b'\r\xc0$\xad\xc6\tmI\x94W\xea\x1f\x0f\xfc\xf0sh')
DEBUG:awoxmeshlight:Read 0x0dc024adc6096d499457ea1f0ffcf07368 from characteristic 00010203-0405-0607-0809-0a0b0c0d1914
DEBUG:__main__:connected
INFO:awoxmeshlight:Start notify
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'\r9\x0c\x00\x00\xba\x93\xafnybb\xf1\xd0(\x88\x9f<s\xd8')>}, []]
INFO:bleak.backends.bluezdbus.client:value changed: /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011
DEBUG:awoxmeshlight:callback: 17: bytearray(b'\r9\x0c\x00\x00\xba\x93\xafnybb\xf1\xd0(\x88\x9f<s\xd8')
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'q\xd9#\x00\x00\x9d\xc5\x0f]\x86\xf4\xbcx\xbdbqZ\xa5\xc9\x01')>}, []]
INFO:bleak.backends.bluezdbus.client:value changed: /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011
DEBUG:awoxmeshlight:callback: 17: bytearray(b'q\xd9#\x00\x00\x9d\xc5\x0f]\x86\xf4\xbcx\xbdbqZ\xa5\xc9\x01')
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b"5'8\x00\x00j\xfb\xa5\xbe\xc9\x02.\x10\x03\xa8\x95\x8evEU")>}, []]
INFO:bleak.backends.bluezdbus.client:value changed: /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011
DEBUG:awoxmeshlight:callback: 17: bytearray(b"5\'8\x00\x00j\xfb\xa5\xbe\xc9\x02.\x10\x03\xa8\x95\x8evEU")
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b"\x8f N\x00\x00j\x03t#\x1a\x9df<|\x906's4_")>}, []]
[... 72 lines cut ...]
Looking at the details of the `write` and `startNotify` in wireshark 

**write_gatt_char(STATUS_CHAR_UUID, b'\x01')**
[image]

**startNofity()**
[image]

Main difference in these messages is the mentioned `Handle: 0x001X`. Not sure if that is related. But for the `read` and `write` is mentions `Handle: 0x0012` and for the `notify` `Handle: 0x0013` but they are all 3 the same Characteristic

```
INFO:awoxmeshlight:[Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 7): Device Information
INFO:awoxmeshlight:	[Characteristic] 00002a26-0000-1000-8000-00805f9b34fb (Handle: 8): Firmware Revision String (read), Value: b'2.2.6\x00'
INFO:awoxmeshlight:	[Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 10): Manufacturer Name String (read), Value: b'AwoX\x00'
INFO:awoxmeshlight:	[Characteristic] 00002a24-0000-1000-8000-00805f9b34fb (Handle: 12): Model Number String (read), Value: b'ESpot_120\x00'
INFO:awoxmeshlight:	[Characteristic] 00002a27-0000-1000-8000-00805f9b34fb (Handle: 14): Hardware Revision String (read), Value: b'4.30'
INFO:awoxmeshlight:	[Characteristic] 00010203-0405-0607-0809-0a0b0c0d1911 (Handle: 17): Unknown (read,write,notify), Value: b'\x01'
INFO:awoxmeshlight:		[Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 19): Client Characteristic Configuration) | Value: b'Status'
INFO:awoxmeshlight:	[Characteristic] 00010203-0405-0607-0809-0a0b0c0d1912 (Handle: 20): Unknown (read,write-without-response,write), Value: b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
INFO:awoxmeshlight:		[Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 22): Characteristic User Description) | Value: b'Command'
INFO:awoxmeshlight:	[Characteristic] 00010203-0405-0607-0809-0a0b0c0d1913 (Handle: 23): Unknown (read,write-without-response), Value: b'\xe0\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
INFO:awoxmeshlight:		[Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 25): Characteristic User Description) | Value: b'OTA'
INFO:awoxmeshlight:	[Characteristic] 00010203-0405-0607-0809-0a0b0c0d1914 (Handle: 26): Unknown (read,write), Value: b'\x0f\x98d\x96$\xea\x18F\xe2$N\xcf\xfb6!\xfau'
INFO:awoxmeshlight:		[Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 28): Characteristic User Description) | Value: b'Pair'
```

dlech, 2022-09-02:
Can you share the full wireshark log?

fsaris, 2022-09-02:
sure 
[awox-try-to-startNotify.zip](https://github.com/hbldh/bleak/files/9480419/awox-try-to-startNotify.zip)

dlech, 2022-09-02:
It seems like this device isn't following the Bluetooth spec and uses a characteristic to enable notifications instead of the Client Characteristic Configuration descriptor which is causing BlueZ to complain. If you control the firmware of the device, I would suggest fixing it there. If not, you might be able to hack around the issue by manually adding the notification callback to `self._client._notification_callbacks` instead of calling the `start_notify` method.

fsaris, 2022-09-02:
I checked the Characteristic and the char I use has `write`, `read` and `notify` 

But for enabling the notify it should use a different Handle?  
Is there like `characteristic` also a command to get this info from the device?

I tried pygat with the gattool as backend and there happens the same. Found this in the code:

```
        # Expect notifications on the value handle...
        value_handle = self.get_handle(uuid)

        # but write to the characteristic config to enable notifications
        # TODO with the BGAPI backend we can be smarter and fetch the actual
        # characteristic config handle - we can also do that with gattool if we
        # use the 'desc' command, so we'll need to change the "get_handle" API
        # to be able to get the value or characteristic config handle.
        characteristic_config_handle = value_handle + 1

        return value_handle, characteristic_config_handle
```
https://github.com/peplin/pygatt/blob/master/pygatt/device.py#L176

I tried the "hack" already. But sadly the messages are not put on the bus by bluez when `startNotify` isn't called.

Updating the firmware of the lights is also not a real solution. I'm trying to control some existing lights   https://github.com/fsaris/home-assistant-awox

dlech, 2022-09-02:
Is it possible that the attributes have changed and BlueZ is using old values from the cache? (see troubleshooting page in Bleak docs on how to clear cache).

Can you get a wireshark log of a working connection for comparison?

Or could you just periodically read the characteristic instead of enabling notifications?

dlech, 2022-09-02:
After looking at the wireshark logs some more, it is clear that the device gatt attribute database is just misconfigured.

> I tried the "hack" already. But sadly the messages are not put on the bus by bluez when `startNotify` isn't called.

In the bleak debug log you shared, I can see `value changed: /org/bluez/hci0/dev_A4_C1_38_8A_BA_94/service0010/char0011` so the info *is* being received by Bleak. So, a hack should be possible.

fsaris, 2022-09-07:
> Or could you just periodically read the characteristic instead of enabling notifications?

That's not an option because the content of the notifications differs from the `read` result. `read` only return a 1 or 0 telling that notifications is enabled or disabled.

> After looking at the wireshark logs some more, it is clear that the device gatt attribute database on the device is just misconfigured.

Do you maybe have a link to some docs about the BLE standards? I see that the tools build on Bluez use a different `handle` value when sending the command to enable the notifications. But didn't yet find any documentation that explains this characteristic.

> If you catch and ignore the exception raised by start_notify(), does the device just disconnect?

Bluez disconnects when the error was thrown. Catching the exception didn't help there sadly. But will try some more.

But looks for now no issue of Bleak but of Bluez + devices that do not follow the BLE standards 100%. It would already help if you could just listen to all incoming notifications independent from the `startNotify` call. Just a callback you could register that receives the notifications for all handles like in bluepy

dlech, 2022-09-07:
> Do you maybe have a link to some docs about the BLE standards?

https://www.bluetooth.com/specifications/specs/

The Generic Attribute Profile (GATT) is in Vol 3, Part G of the Core Specification (CS).

---

The problem with this device is that it looks like they meant for this characteristic to have both a Client Characteristic Configuration descriptor and a Characteristic User Description descriptor but they didn't allow for enough attribute handles for both. So the Characteristic User Description descriptor handle is where the Client Characteristic Configuration should be and the Client Characteristic Configuration descriptor handle overlaps with characteristic value handle. This is why reading and writing the characteristic value is actually reading and writing the Client Characteristic Configuration descriptor and the value cannot be read.

---

> Bluez disconnects when the error was thrown.

This seems to be the default BlueZ behavoir when it encounters something unexpected. Other Bluetooth stacks seem to be more forgiving, but BlueZ will just disconnect.

> It would already help if you could just listen to all incoming notifications independent from the `startNotify` call. Just a callback you could register that receives the notifications for all handles like in bluepy

The BlueZ D-Bus APIs don't allow for doing anything low level like this. I think the options are fix the device firmware, patch BlueZ for this specific device or don't use BlueZ. Such workarounds are outside the scope of Bleak.

dlech, 2023-03-31:
> Such workarounds are outside the scope of Bleak.

Based on the number of links to this issue, I might have to eat my words here. It sounds like more than one device has the same sort of problem with a broken client characteristic configuration descriptor.

> It would already help if you could just listen to all incoming notifications independent from the `startNotify` call. Just a callback you could register that receives the notifications for all handles like in bluepy

This sounds reasonable. I think the simplest thing to do would be to add an additional parameter to the `BleakClient` constructor to register the callback similar to what we do currently with the disconnect callback.

SuibianP, 2023-09-14:
I wonder how this additional parameter can be implemented if the D-Bus API just cannot stand this without disconnecting. Are we changing to a native backend like bluepy does?

BrainStone, 2026-10-11:
Looking through this discussion, this change seems to fairly contained. Could this be reconsidered?

dlech, 2026-10-11:
I'm not sure that it is technically possible, but if it is, we can have a look.