hbldh/bleak #992: Device may already be connected when calling connect

bdraco, 2022-09-10:
If the device is connected, and then the docker container is restarted and the device is not disconnected, the next connection attempt will fail because the device is already connected on the host but we think it is not connected and a connect call will fail.

```
      "/org/bluez/hci1/dev_BD_24_6F_85_AA_61": {
        "org.freedesktop.DBus.Introspectable": {},
        "org.bluez.Device1": {
          "Address": "BD:24:6F:85:AA:61",
          "AddressType": "public",
          "Name": "Dream~BD246F85AA61",
          "Alias": "Dream~BD246F85AA61",
          "Appearance": 962,
          "Icon": "input-mouse",
          "Paired": false,
          "Trusted": false,
          "Blocked": false,
          "LegacyPairing": false,
          "RSSI": -36,
          "Connected": true,
          "UUIDs": [
            "00001800-0000-1000-8000-00805f9b34fb",
            "00001801-0000-1000-8000-00805f9b34fb",
            "0000180a-0000-1000-8000-00805f9b34fb",
            "0000ffd0-0000-1000-8000-00805f9b34fb",
            "0000ffd5-0000-1000-8000-00805f9b34fb"
          ],
          "Modalias": "usb:v045Ep0040d0300",
          "Adapter": "/org/bluez/hci1",
          "ManufacturerData": {
            "20808": {
              "__type": "<class 'bytearray'>",
              "repr": "bytearray(b'364656')"
            }
          },
          "ServicesResolved": true
        },
        "org.freedesktop.DBus.Properties": {}
      },
```

bdraco, 2022-09-10:
Need to think about how to fix this one as its not strait-forward.  I wanted to document it so I don't forget about it

bdraco, 2022-09-10:
Maybe skip calling "Connect" if the `Connected` property is already true when we get here https://github.com/hbldh/bleak/blob/aa24332a39952880005705155629df423fbad1aa/bleak/backends/bluezdbus/client.py#L158

bdraco, 2022-09-10:
Maybe could use the same check that ServicesResolved uses

dlech, 2022-09-10:
This is essentially a duplicate of #367.

Instead of hacking the BleakClient connect method, I think we should use the algorithm described in the diagram [here](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/BestPracticesForInteractingWithARemotePeripheralDevice/BestPracticesForInteractingWithARemotePeripheralDevice.html#//apple_ref/doc/uid/TP40013257-CH6-SW9).

The idea I've had in my mind for a while to do this is to create a new BleakAdapter class. This class could provide the "get known peripherals" and the "get connected peripherals" methods. The BleakAdapter class could also be used to solve issues like #320 and #513.

dlech, 2022-09-10:
Also related is https://github.com/bluez/bluez/issues/89. Bluez is the only OS stack that doesn't "release" the device when the app dies.

PR #994 (2022-09-10): Handle a device already being connected with the BlueZ Dbus backend
Fixes #992
PR #1006 (2022-09-12): Release/v0.17.0

Added
-----
* ``AdvertisementData`` class now has an attribute ``tx_power``. Merged #987.

Changed
-------
* ``BleakClient`` methods now raise ``BleakError`` if called when not connected in WinRT backend. Merged #973.
* Extended disconnect timeout to 120 seconds in WinRT backend. Fixes #807.
* Changed version check for BlueZ battery workaround to exclude versions >= 5.55. Merged #976.
* Use Poetry for build system and dependencies. Merged #978.
* The BlueZ D-Bus backend implements a services cache between connections to significancy improve reconnect performance.
  To use the  [... cut]
closed 2022-09-10 by dlech