hbldh/bleak #993: bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local

Arbuzov, 2022-09-10:
* bleak version: ```0.16.0```
* Python version: ```3.10```
* Operating System: Home Assistant docker container ```Linux homeassistant 5.15.32-v8 #1 SMP PREEMPT Tue Aug 16 14:52:58 UTC 2022 aarch64 Linux```
* BlueZ version (`bluetoothctl -v`) in case of Linux: ```5.64```

### Description

Cant connect to device

### What I Did

```
import asyncio
import binascii
import uuid
from bleak import BleakClient, BleakScanner
from bleak.exc import BleakError

MAC = "00:A0:50:82:75:CA"
NAME_CHARACTERISTIC = '00002A00-0000-1000-8000-00805F9B34FB'
GENERAL_SERVICE = '00001801-0000-1000-8000-00805f9b34fb'

async def main(ble_address: str):
    device = await BleakScanner.find_device_by_address(ble_address, timeout=20.0)
    async with BleakClient(device) as client:
        services = await client.get_services()
        print("Services:")
        for service in services:
            print(service)
        value = bytes(await client.read_gatt_char(uuid.UUID(NAME_CHARACTERISTIC))).decode('utf-8')
        print(value)

asyncio.run(main(MAC))

```
The stack-trace is:
```
Traceback (most recent call last):
  File "/config/./debug_bleak.py", line 29, in <module>
    asyncio.run(main(MAC))
  File "/usr/local/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/config/./debug_bleak.py", line 21, in main
    async with BleakClient(device) as client:
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/client.py", line 61, in __aenter__
    await self.connect()
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/client.py", line 167, in connect
    assert_reply(reply)
  File "/usr/local/lib/python3.10/site-packages/bleak/backends/bluezdbus/utils.py", line 24, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local
```

Same code at the native Linux works correctly

dlech, 2022-09-10:
> Same code at the native Linux works correctly

I'm not sure I understand what this means. Can you explain more?

Since this is a BlueZ error, we would need to look into the BlueZ source code and/or log Bluetooth packets to see what could cause the error.

Slluxx, 2022-09-14:
I have the same issue on a RPi4 running Ubuntu 22.04.1.

```py
        async with BleakClient("XX:XX:XX:XX:XX:XX") as client:
            try:
                await client.write_gatt_char("00000000-0000-0000-0000-000000000000", packet)
            except Exception as e:
                print(e)
                break
```
`> [org.bluez.Error.Failed] le-connection-abort-by-local`

I changed MAC and GATT id. 
I know that the LED strip works cause i can connect to it via an official app aswell as via "nRF Connect" and send stuff to it.

dlech, 2022-09-14:
the best way to troubleshoot this is to capture bluetooth packets with wireshark for both the working and not working cases and compare the two to see what is different

Slluxx, 2022-09-14:
@Arbuzov and i have the same error message with the same stacktrace, on the same version of bluez, python and bleak with different devices (and at least one of them works the intended way). I higly doubt its a user issue and you should be able to reproduce the issue on your end. I am sadly not familiar enough with wireshark and dont have enough free time to learn it right now. I understand if you dont have the time either and dont want to just dump this on you and then leave but i am unable to check this out for an unforeseeable amount of time.

Arbuzov, 2022-09-15:
@Slluxx Interesting point. I have not motioned it worked in the native linux RPI3 and it does not inside the Docker on RPI4
I will try to deploy Docker at RPI3 to confirm/decline hardware dependency

Slluxx, 2022-09-15:
I am very interested in your findings. Please keep me up to date!

Arbuzov, 2022-09-19:
> I am very interested in your findings. Please keep me up to date!

Got the same error at RPI3 Model B
uname -a
Linux a0d7b954-ssh 5.15.61-v8 #1 SMP PREEMPT Tue Sep 13 15:24:53 UTC 2022 aarch64 Linux

Arbuzov, 2022-09-19:
@dlech I have no UI at that hosts. So if you can advice how to get dump/dbus log/bleak log. I will be very appreciated.

dlech, 2022-09-19:
You can capture Bluetooth packets from a command prompt with `hcidump` or `btmon`. 

In all cases, the logs can be redirected to a file (i.e. `$command > $file`) and then the files transferred to another computer for inspection.

Arbuzov, 2022-09-19:
Got btmon dump. Cut messages including `00:A0:50:82:75:CA` this is my device this is a faulty case:
`
> HCI Event: LE Meta Event (0x3e) plen 43                  #73 [hci0] 19.281249
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
        RSSI: -81 dBm (0xaf)
@ MGMT Event: Device Found (0x0012) plen 45           {0x0001} [hci0] 19.283011
        LE Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        RSSI: -81 dBm (0xaf)
        Flags: 0x00000000
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
< HCI Command: LE Create Connection (0x08|0x000d) plen 25  #76 [hci0] 19.283095
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: Accept list is not used (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Own address type: Public (0x00)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: LE Meta Event (0x3e) plen 19                  #94 [hci0] 21.375700
      LE Connection Complete (0x01)
        Status: Unknown Connection Identifier (0x02)
        Handle: 64
        Role: Central (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Connection interval: 48.75 msec (0x0027)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Central clock accuracy: 0x00
> HCI Event: LE Meta Event (0x3e) plen 43                  #97 [hci0] 22.542550
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
        RSSI: -63 dBm (0xc1)
@ MGMT Event: Device Found (0x0012) plen 45           {0x0001} [hci0] 24.854706
        LE Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        RSSI: -63 dBm (0xc1)
        Flags: 0x00000000
        Data length: 31
[... 88 lines cut ...]
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
        RSSI: -79 dBm (0xb1)
@ MGMT Event: Device Found (0x0012) plen 45           {0x0001} [hci0] 46.026176
        LE Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        RSSI: -79 dBm (0xb1)
        Flags: 0x00000000
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
`

dlech, 2022-09-19:
```
> HCI Event: Disconnect Complete (0x05) plen 4            #167 [hci0] 43.783462
        Status: Success (0x00)
        Handle: 64 Address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Reason: Connection Failed to be Established (0x3e)
```

The device is initiating the disconnection.

It does this after sending the connection complete:

```
> HCI Event: LE Meta Event (0x3e) plen 19                 #162 [hci0] 43.468453
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Central (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:A0:50:82:75:CA (OUI 00-A0-50)
        Connection interval: 48.75 msec (0x0027)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Central clock accuracy: 0x00
```

So I'm not sure why the device would be doing this.

It could be helpful to capture a working connection to the same device to compare the difference.

Arbuzov, 2022-09-19:
Here is the dump from the successful connection:
```
> HCI Event: LE Meta Event (0x3e) plen 43                   #10 [hci0] 6.097522
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific (00035b03-58e6-07dd-021a-08123a000300)
        RSSI: -75 dBm (0xb5)
> HCI Event: LE Meta Event (0x3e) plen 12                   #11 [hci0] 6.097776
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        Data length: 0
        RSSI: -75 dBm (0xb5)
@ MGMT Event: Device Found (0x0012) plen 45            {0x0001} [hci0] 6.097813
        LE Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        RSSI: -75 dBm (0xb5)
        Flags: 0x00000000
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific (00035b03-58e6-07dd-021a-08123a000300)
> HCI Event: LE Meta Event (0x3e) plen 43                  #48 [hci0] 17.108122
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific (00035b03-58e6-07dd-021a-08123a000300)
        RSSI: -69 dBm (0xbb)
@ MGMT Event: Device Found (0x0012) plen 45           {0x0001} [hci0] 17.109749
        LE Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        RSSI: -69 dBm (0xbb)
        Flags: 0x00000000
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific (00035b03-58e6-07dd-021a-08123a000300)
> HCI Event: LE Meta Event (0x3e) plen 12                  #59 [hci0] 17.292271
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Scan response - SCAN_RSP (0x04)
        Address type: Public (0x00)
        Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        Data length: 0
        RSSI: -71 dBm (0xb9)
@ MGMT Event: Device Found (0x0012) plen 14           {0x0001} [hci0] 17.292307
        LE Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
[... 139 lines cut ...]
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 64
        Role: Master (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        Connection interval: 48.75 msec (0x0027)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Master clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 44       {0x0001} [hci0] 27.846446
        LE Address: 00:A0:50:82:75:CA (CYPRESS SEMICONDUCTOR)
        Flags: 0x00000008
          Unknown device flag (0x00000008)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        Name (complete): D1703592
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific (00035b03-58e6-07dd-021a-08123a000300)
```
[btmon.bad.txt](https://github.com/hbldh/bleak/files/9601668/btmon.bad.txt)
[btmon.good.txt](https://github.com/hbldh/bleak/files/9601669/btmon.good.txt)
The good and the bad logs are attached

dlech, 2022-09-19:
Also, searching for the error may turn up some results, like: https://stackoverflow.com/questions/68426984/bluetooth-connection-failed-from-raspberry-pi-at-le-read-remote-used-features

Slluxx, 2022-09-20:
> Also, searching for the error may turn up some results, like: https://stackoverflow.com/questions/68426984/bluetooth-connection-failed-from-raspberry-pi-at-le-read-remote-used-features

Sadly thats not the solution, as mine was already on "dual" and setting it to "LE" doesnt change anything.

Arbuzov, 2022-09-20:
Well, downgrade bluez to 5.55 makes connection more reliable.  Not a first try connection, but it`s possible to use.

deepcoder, 2022-10-21:
Same issue on x86 Ubuntu 22.04.1
```
user@quieter3-01:~/bleak-ble$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.1 LTS"
user@quieter3-01:~/bleak-ble$ uname -a
Linux quieter3-01 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
user@quieter3-01:~/bleak-ble$ bluetoothctl -v
bluetoothctl: 5.64
user@quieter3-01:~/bleak-ble$ pip3 show bleak
Name: bleak
Version: 0.19.0
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: /home/user/.local/lib/python3.10/site-packages
Requires: async-timeout, dbus-fast
Required-by:
```

bdraco, 2022-10-31:
Is this by chance an rtl8761bu chip?

deepcoder, 2022-10-31:
machine I see the issue on:
MeLE Fanless Mini PC Quieter3Q Celeron N5105 8GB 256GB
```
user@quieter3-01:~$ inxi -Eaz
Bluetooth:
  Device-1: Intel AX201 Bluetooth type: USB driver: btusb v: 0.8 bus-ID: 1-8:5 chip-ID: 8087:0026
    class-ID: e001
  Report: hciconfig ID: hci0 rfk-id: 1 state: up address: <filter> bt-v: 3.0 lmp-v: 5.2
    sub-v: 237e hci-v: 5.2 rev: 237e
  Info: acl-mtu: 1021:4 sco-mtu: 96:6 link-policy: rswitch sniff link-mode: peripheral accept
    service-classes: rendering, capturing, audio, telephony
```

bdraco, 2022-10-31:
When my RTL8761BU based ZEXMTE BT-505 gets overloaded I get this error.

I replaced it with a Feasycom FSC-BP119 which is CSR8510A10 and seems to handle the bluetooth density quite a bit better.

pedroishimaru, 2023-10-04:
> Well, downgrade bluez to 5.55 makes connection more reliable. Not a first try connection, but it`s possible to use.

I faced the same error using Ubuntu 22.04 with **BlueZ 5.64** on a Raspberry PI 4. I spent a couple days trying to figure out how to get rid of it, but the only solution that seems to work was to downgrade BlueZ to 5.53.

topraveenpaul, 2023-10-05:
> I faced the same error using Ubuntu 22.04 with **BlueZ 5.64** on a Raspberry PI 4. I spent a couple days trying to figure out how to get rid of it, but the only solution that seems to work was to [...]

@pedroishimaru After downgrade, were you able to connect within atleast 3 to 4 tries? Or it still took more tries to connect?

topraveenpaul, 2023-10-05:
> > > > Well, downgrade bluez to 5.55 makes connection more reliable. Not a first try connection, but it`s possible to use.
> > > 
> > > 
> > > I faced the same error using Ubuntu 22.04 with **BlueZ 5.64** on a Raspberry PI 4. I spent a couple days trying to figure out how to get rid of it, but the only solution that seems to work was to downgrade BlueZ to 5.53.
> > 
> > 
> > @pedroishimaru After downgrade, were you able to connect within atleast 3 to 4 tries? Or it still took more tries to connect?
> 
> @topraveenpaul I'm still testing it, but I'd say that at least at 80% of the times the device connects successfully within the first 2 or 3 tries, and even when it fails, i'm getting a different error message now:
> 
> `bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Software caused connection abort`
> 
> Btw, for some reason it seems to work even better when running on **Ubuntu 20.04**.

Great. Thank you for the update. 

In my case, On my Rpi 4 on ubuntu 22.04 LTS, bluez version 5.64, the same error is thrown by the chip-tool that I use to connect with matter iot devices as you can see from the attached logs. 

I will try downgrading the bluez as you suggested.

Upon going through the code, the comment says below.  I never understood how wifi radio interferes BLE connection.  
    "_{
        ChipLogError(DeviceLayer, "FAIL: ConnectDevice : %s (%d)", error->message, error->code);
        // Due to radio interferences or Wi-Fi coexistence, sometimes the BLE connection may not be
        // established (e.g. Connection Indication Packet is missed by BLE peripheral). In such case,
        // BlueZ returns "Software caused connection abort error", and we should make a connection retry._
"

[bluez_error.txt](https://github.com/hbldh/bleak/files/12819631/bluez_error.txt)

topraveenpaul, 2023-10-05:
> Since Bleak 0.20.0 we automatically retry to connect if `le-connection-abort-by-local` is seen. Is this not working?
> 
> I guess it kind of is based on seeing it multiple times in `bluez_error.txt` and the fact that a different error (`[org.bluez.Error.Failed] Software caused connection abort`) is reported now.
> 
> If there was a regression in BlueZ, I would suggest reporting the error there so we can better understand why the change was made.

No, the retry is happening and in my case it retries 5 times and then quit. 
What i dont understand is the reason for the error "le-connection-abort-by-local(36)". 

b'[1696247711.592396][2863:2864] CHIP:BLE: BLE initiating scan.\n'
b'[1696247711.631431][2863:2864] CHIP:BLE: Device 5F:52:45:A8:CB:C7 does not look like a CHIP device.\n'
b'[1696247711.650659][2863:2864] CHIP:BLE: New device scanned: D5:86:EA:1C:4A:81\n'
b'[1696247711.650697][2863:2864] CHIP:BLE: Device discriminator match. Attempting to connect.\n'
b'[1696247712.007194][2863:2864] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)\n'
b'[1696247712.410912][2863:2864] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)\n'
b'[1696247712.869689][2863:2864] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)\n'
b'[1696247713.253387][2863:2864] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)\n'
b'[1696247713.684620][2863:2864] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: le-connection-abort-by-local (36)\n'
b'[1696247713.684749][2863:2865] CHIP:DL: HandlePlatformSpecificBLEEvent 16386\n'
b'[1696247713.684777][2863:2865] CHIP:CTL: Commissioning discovery over BLE failed: src/platform/Linux/bluez/Helper.cpp:1772: CHIP Error 0x000000AC: Internal error\n'

Arbuzov, 2023-11-10:
I have updated my HA to the latest-greatest.
Looks like it was fixed.

closed 2023-11-10 by Arbuzov