hbldh/bleak #998: AttributeError: 'NoneType' object has no attribute 'max_pdu_size'

snowuyl, 2022-09-12:
* bleak version: 0.16.0
* Python version: 3.10.7
* Operating System: Windows 10

### Description
python3 main.py
C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyhrv\utils.py:367: UserWarning: Signal duration is to short for segmentation into 300000s. Input data will be returned.
  warnings.warn("Signal duration is to short for segmentation into %is. Input data will be returned." % duration)
C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyhrv\time_domain.py:303: UserWarning: Signal duration too short for SDANN computation.
  warnings.warn("Signal duration too short for SDANN computation.")
C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyhrv\time_domain.py:576: UserWarning: CAUTION: The TINN computation is currently providing incorrect results in the most cases due to a malfunction of the function. This function will be reviewed over the next updates to solve this issue
  warnings.warn('CAUTION: The TINN computation is currently providing incorrect results in the most cases due to a '
C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\pyhrv\nonlinear.py:308: UserWarning: Not enough NNI samples for Detrended Fluctuations Analysis.
  warnings.warn("Not enough NNI samples for Detrended Fluctuations Analysis.")
10
BleDeviceStruct(name='ANCS3', address='E8:20:9F:75:90:08', rssi=-35)
_run_battery E8:20:9F:75:90:08
_run_uart E8:20:9F:75:90:08
Exception ignored in: <function BleakClientWinRT.connect.<locals>.max_pdu_size_changed_handler at 0x0000020E13AD0C10>
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 292, in max_pdu_size_changed_handler
    logger.debug("max_pdu_size_changed_handler: %d", self._session.max_pdu_size)
AttributeError: 'NoneType' object has no attribute 'max_pdu_size'

closed 2022-09-12 by dlech

snowuyl, 2022-09-12:
Which version of bleak solve this issue?

dlech, 2022-09-12:
v0.17.0. This information can always be found in the [changelog](https://github.com/hbldh/bleak/blob/develop/CHANGELOG.rst).

snowuyl, 2022-09-13:
Thanks for your reply!

commit 05c8b0d2 (2022-09-12): backends/winrt/client: fix crash in max_pdu_size_changed_handler

It is possible that self._session is set to None when
max_pdu_size_changed_handler is called. So we can use the sender arg
instead to avoid a possible AttributeError.

Fixes #998.
PR #1001 (2022-09-12): backends/winrt/client: fix crash in max_pdu_size_changed_handler
It is possible that `self._session` is set to `None` when `max_pdu_size_changed_handler` is called. So we can use the `sender` arg instead to avoid a possible `AttributeError`.

Fixes #998.
commit 21e72cb0 (2022-09-12): backends/winrt/client: fix crash in max_pdu_size_changed_handler

It is possible that self._session is set to None when
max_pdu_size_changed_handler is called. So we can use the sender arg
instead to avoid a possible AttributeError.

Fixes #998.