hbldh/bleak #1028: BleakScanner.find_device_by_filter() signature changed in v0.18.0

jeffsf, 2022-09-24:
Between v0.17.0 and v0.18.0, a named parameter changed from `filterfunc` to `filter_func`

Calls to `find_device_by_filter` that identify the parameter by name will fail in one version.

Workarounds:
* Detect the version and branch between two call signatures
* Convert to a positional argument list
* Require v0.17.0 or later and hope that the user is using pip or equivalent

In v0.17.0, in `bleak/backends/scanner.py`
```
    234     @classmethod
    235     async def find_device_by_filter(
    236         cls, filterfunc: AdvertisementDataFilter, timeout: float = 10.0, **kwargs
    237     ) -> Optional[BLEDevice]:
```

In v0.18.0, in `__init__.py``
```
    227     @classmethod
    228     async def find_device_by_filter(
    229         cls, filter_func: AdvertisementDataFilter, timeout: float = 10.0, **kwargs
    230     ) -> Optional[BLEDevice]:
```

dlech, 2022-09-24:
Bah, I should know better than to have changed that.

jeffsf, 2022-09-24:
It also looks like there is an inconsistency between `device_or_address` in `__init__.py` and `address_or_ble_device` in the backends.

```
    301     def __init__(
    302         self,
    303         device_or_address: Union[BLEDevice, str],
    304         disconnected_callback: Optional[Callable[[BleakClient], None]] = None,
    305         *,
    306         timeout: float = 10.0,
    307         winrt: WinRTClientArgs = {},
    308         backend: Optional[Type[BaseBleakClient]] = None,
    309         **kwargs,
    310     ):
```

```
bleak/backends/bluezdbus/client.py:    def __init__(self, address_or_ble_device: Union[BLEDevice, str], **kwargs):
bleak/backends/client.py:    def __init__(self, address_or_ble_device: Union[BLEDevice, str], **kwargs):
bleak/backends/corebluetooth/client.py:    def __init__(self, address_or_ble_device: Union[BLEDevice, str], **kwargs):
bleak/backends/p4android/client.py:    def __init__(self, address_or_ble_device: Union[BLEDevice, str], **kwargs):
bleak/backends/winrt/client.py:        address_or_ble_device: Union[BLEDevice, str],
```

PR #1030 (2022-09-25): Release/0.18.1

Fixed
-----
* Reverted unintentional breaking parameter name changes. Fixes #1028.
commit e0cb89fd (2022-09-25): revert breaking name changes

When we added new concrete BleakClient and BleakScanner classes a few
parameter names were changed. But these can be used as keyword arguments
so changing the name was an unintentional breaking change.

Fixes #1028.
closed 2022-09-25 by dlech