hbldh/bleak #1036: bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress is occur.

KukhoHeo, 2022-09-28:
* bleak version: 0.18.1
* Python version:3.10.6
* Operating System: Ubuntu 22.04.1 LTS (Raspberry 4b)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

### Description
when I do the program, the error occur.

import asyncio
import os
from bleak import *
import time

CHAR_UUID="4eab0101-0001-0001-0003-1001200252b9"

def notify_callback(sender, data):
    print(sender, data)

def disconnected_callback(client):
    print(f'Device {client.address} disconnected')

async def work( address):
    async with BleakClient(address) as client:
        client.set_disconnected_callback(disconnected_callback)
        await client.start_notify(CHAR_UUID, notify_callback)
    print(f'Work task for {address} end')

async def main():
    task_list=[]
    target_devices=["E2:A5:A4:C3:4C:C0","DC:8B:F0:17:5C:5A"]
    await asyncio.gather(work(target_devices[0]),work(target_devices[1]))
    print("Main task end")

if __name__ == "__main__":
    asyncio.run(main())

this is my simple program that connects two device in same time.

what should I do for fix the program?

manschwetus, 2022-09-28:
I had this problem with pybricksdev, it seems that it was the adapter, once I disabled and enabled it, this started working.

dlech, 2022-09-28:
Connecting to two devices at the same time is problematic for many people. Try connecting to one first and then the other.

Amar-Judo, 2022-12-30:
I have this same issue, do you know the solution to solve this issue?
I am so much struggling to solve.

Many Thanks 

[image]

buganini, 2023-04-29:
bleak version: 0.20.2
Python version: 3.10.6
Operating System: Ubuntu 22.04.2 LTS (x64)
BlueZ version (bluetoothctl -v) in case of Linux: 5.64

I got the same error, stop scanner before connect and start scanner later works for me.

datk1912, 2024-12-09:
I guess that you can replace the addresses in `target_devices` with a list of BLEDevice objects, which are the return values of BleakScanner, It might work.