hbldh/bleak #1038: Fatal error while scanning under windows

zapta, 2022-09-29:
* bleak version:

pip.exe freeze
async-timeout==4.0.2
bleak==0.16.0
bleak-winrt==1.1.1
cycler==0.11.0
fonttools==4.37.1
kiwisolver==1.4.4
matplotlib==3.5.3
numpy==1.23.2
packaging==21.3
Pillow==9.2.0
pyparsing==3.0.9
PyQt6==6.3.1
PyQt6-Qt6==6.3.1
PyQt6-sip==13.4.0
pyqtgraph==0.12.4
python-dateutil==2.8.2
six==1.16.0
typing_extensions==4.3.0
  
* Python version: Python 3.10.7

* Operating System: Windows
 
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

I have a python/bleak application that has two python mains, (i) Scanning for BLE devices and (ii) connecting to a BLE device, reading data points and display them in real time.  The two programs work on my windows machine but a beta user that tried it report that both fail for him and adding debug messages suggest that it's bleak related.  I will provide here information for the first program which is significantly simpler and will deal with the other one seperatly.

Source code https://github.com/zapta/ble_stepper_motor_analyzer/blob/main/python/scanner_main.py

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

> py scanner_main.py
Scanning (5 sec)...
Traceback (most recent call last):
  File "C:\Users\James\Documents\Github\ble_stepper_motor_analyzer\python\scanner_main.py", line 20, in <module>
    asyncio.run(scan())
  File "C:\Program Files\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\Users\James\Documents\Github\ble_stepper_motor_analyzer\python\scanner_main.py", line 13, in scan
    devices = await discover(timeout=5)
TypeError: discover() got an unexpected keyword argument 'timeout'

dlech, 2022-09-29:
Please try with the current version of Bleak (v0.18.1). The `discover()` method has been deprecated, use `BleakScanner.disocver()` instead.

closed 2022-09-29 by dlech

zapta, 2022-10-02:
That worked. Thanks.

PR #1042 (2022-09-29): fix deprecated discover method
This was broken in 825df5dcdcdc234edcd7b33086056d7edc198916.

Fixes #1038.
commit 17d9e7ab (2022-09-29): fix deprecated discover method

This was broken in 825df5dcdcdc234edcd7b33086056d7edc198916.

Fixes #1038.