hbldh/bleak #1039: Exception ignored in: function BleakClientWinRT.connect.<locals>.max_pdu_size_changed_handler

snowuyl, 2022-09-29:
* bleak version: 0.17.0
* Python version: 3.10.7
* Operating System: Windows 10

### Description

python3 main.py
Exception ignored in: <function BleakClientWinRT.connect.<locals>.max_pdu_size_changed_handler at 0x0000021B434DD6C0>
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 282, in max_pdu_size_changed_handler
    logger.debug("max_pdu_size_changed_handler: %d", sender.max_pdu_size)
OSError: [WinError -2147417848]

dlech, 2022-09-29:
Some interesting reading on the race condition that may contribute to this: https://devblogs.microsoft.com/oldnewthing/20190521-00/?p=102505

closed 2022-09-29 by dlech

snowuyl, 2022-09-30:
Which version of bleak solved this issue?

dlech, 2022-09-30:
Please refer to the changelog for status.

PR #1041 (2022-09-29): winrt/client: handle exception in max_pdu_size_changed_handler
This catches a possible OSError when max_pdu_size_changed_handler is called after the GATT session ends. It can safely be ignored.

Fixes #1039.
commit 3f483441 (2022-09-29): winrt/client: handle exception in max_pdu_size_changed_handler

This catches a possible OSError when max_pdu_size_changed_handler
is called after the GATT session ends. It can safely be ignored.

Fixes #1039.
PR #1081 (2022-10-14): Release/0.19.0

Added
-----
* Added support for Python 3.11. Merged #990.
* Added better error message for Bluetooth not authorized on macOS. Merged #1033.
* Added ``BleakDeviceNotFoundError`` which should is raised if a device can not
  be found by ``connect``, ``pair`` and ``unpair``. Merged #1022.
* Added ``rssi`` attribute to ``AdvertisementData``. Merged #1047.
* Added ``BleakScanner.discovered_devices_and_advertisement_data`` property. Merged #1047.
* Added ``return_adv`` argument to ``BleakScanner.discover`` method. Merged #1047.
* Added ``BleakClient.unpair()`` implementation for BlueZ [... cut]