hbldh/bleak #1043: Android examples not showing

AgustinCFZ, 2022-09-29:
Hello!,
Please, move this question if this is not the correct place. 
I'm trying to use bleak from android, but the example page is not working:
https://bleak.readthedocs.io/en/examples/kivy

Please, could you fix this?. 

Many thanks!

dlech, 2022-09-30:
It is in the source tree, not the docs. https://github.com/hbldh/bleak/tree/develop/examples/kivy

Where did you get the link?

AgustinCFZ, 2022-09-30:
Thanks!. I found the link here:
https://bleak.readthedocs.io/en/latest/backends/android.html

I'm looking a way to use bleak from an android phone to upload a code to the ESP32, but i can't find information on how install bleak on android.
I downloaded an app called "pydroid 3" and tried to execute there a bleak python code, with no success.
If you have a link or any information, i would appreciate it very much!.

Many thanks!

AgustinCFZ, 2022-09-30:
There is a missing example in the website, where Buildozer will compile an app and upload it to a device
Is the first link here:
https://bleak.readthedocs.io/en/latest/backends/android.html

dlech, 2022-09-30:
Fixed in #1045

PR #1045 (2022-09-30): Update android.rst
Fixes #1043
commit 822c3edb (2022-09-30): Update android.rst (#1045)

Fixes #1043
commit b5361077 (2022-09-30): Update android.rst (#1045)

Fixes #1043
closed 2022-09-30 by dlech