hbldh/bleak #1049: Support for typed access to characteristic values in a more Pythonic way

jackjansen, 2022-10-01:
I have code to allow accessing characteristic values (for both reading and writing) based on 2904 descriptors.

But this is really only a stepping stone, because what I really want eventually is to be able to have accessor objects, so I can do things like

```
async with BleakClientAccessor(...) as mydev:
    myservice = mydev[service_uuid]
    mychar = myservice[char_uuid]
    mychar.value = mychar.value + 1
    mydev[service_uuid][char_uuid].value = 42
```

So that would require accessor objects for services and characteristics, but that would all be fairly easy to implement.

So my question is really whether this is a direction that is worth pursuing. If you agree I'll first create a pull request for the typed access (using the 2904 descriptors), and then later for the accessors. The 2904 support is actually pretty useful by itself already.

jackjansen, 2022-10-01:
Oh yeah: the naming used in the sample code above suggests that it would be only UUIDs to index the accessor, but I envision that it can actually also be `BleakGATTCharacteristics`, etc.

dlech, 2022-10-02:
For those that don't know what 2904 is, it is the "Characteristic Presentation Format" descriptor which describes the data format of the characteristic value. So, if (and only if) a characteristic has this descriptor, Bleak could decode the value to an int or float or string for you instead of having to decode the bytes yourself. It additionally includes the units and namespace/description of the values. This is primarily useful for generic clients that want to display data from an unknown device. Also the namespace/description is the recommended way to differentiate two characteristics with the same UUID.

> mychar.value = mychar.value + 1

Wouldn't this require I/O to read from and write to the device? How could that be done in a non-blocking way without `await`?

jackjansen, 2022-10-02:
> mychar.value = mychar.value + 1

You're right, I'm running but first I should be walking. For now replace by

```
v = await mychar.get()
await mychar.set(v+1)
```

jackjansen, 2022-10-02:
>  So, if (and only if) a characteristic has this descriptor, Bleak could decode the value to an int or float or string for you instead of having to decode the bytes yourself.

I _think_ that it should also be possible to obtain the information from the [bleak-sigspec](https://bleak-sigspec.readthedocs.io/en/latest/) module, if it is installed. And I'm hoping that @Carglglz can help me with glueing that together after the 2904 support is in place. (personally I'm most interested in 2904, because its the only way to advertise my characteristics short of shelling out the money to become a BLE SIG member and officially registering my UUIDs).

Carglglz, 2022-10-02:
@jackjansen I did mentioned this in [bleak-sigspec](https://bleak-sigspec.readthedocs.io/en/latest/futurework.html) 

> there is a Characteristic Presentation Format Descriptor which defines the format of the Characteristic Value. So if this Descriptor is present it is possible to read it and get the characteristic value format, exponent, unit, name space and description.

And I've just remembered that there is a XML file for this descriptor:

```
Descriptor Metadata:
Descriptor: Characteristic Presentation Format

    - NAME: Characteristic Presentation Format
    - ABSTRACT: The Characteristic Presentation Format descriptor defines the format of the Characteristic Value.
    - SUMMARY: One or more Characteristic Presentation Format descriptors may be present. If multiple of these descriptors are present, then a
                   Aggregate Formate descriptor is present. This descriptor is read only and does not require authentication or authorization to
                   read. This descriptor is composed of five parts: format, exponent, unit, name space and description. The Format field determines
                   how a single value contained in the Characteristic Value is formatted. The Exponent field is used with integer data types to
                   determine how the Characteristic Value is further formatted. The actual value = Characteristic Value * 10^Exponent.
    - UUID: 2904
    - INFO TEXT: None
    - Format:
        - Requirement: Mandatory
        - Format: 8bit
        - Ctype: B
        - Minimum: 0
        - Maximum: 27
        - Enumerations:
            - 0: Reserved For Future Use
            - 1: Boolean
            - 2: unsigned 2-bit integer
            - 3: unsigned 4-bit integer
            - 4: unsigned 8-bit integer
            - 5: unsigned 12-bit integer
            - 6: unsigned 16-bit integer
            - 7: unsigned 24-bit integer
            - 8: unsigned 32-bit integer
            - 9: unsigned 48-bit integer
            - 10: unsigned 64-bit integer
            - 11: unsigned 128-bit integer
            - 12: signed 8-bit integer
            - 13: signed 12-bit integer
            - 14: signed 16-bit integer
            - 15: signed 24-bit integer
            - 16: signed 32-bit integer
            - 17: signed 48-bit integer
            - 18: signed 64-bit integer
            - 19: signed 128-bit integer
            - 20: IEEE-754 32-bit floating point
            - 21: IEEE-754 64-bit floating point
            - 22: IEEE-11073 16-bit SFLOAT
            - 23: IEEE-11073 32-bit FLOAT
            - 24: IEEE-20601 format
            - 25: UTF-8 string
            - 26: UTF-16 string
            - 27: Opaque Structure
    - Exponent:
        - Requirement: Mandatory
        - Format: sint8
        - Ctype: b
    - Unit:
        - InformativeText: The Unit is a UUID.
        - Requirement: Mandatory
        - Format: uint16
        - Ctype: H
    - Namespace:
        - InformativeText: The Name Space field is used to identify the organization that is responsible for defining the enumerations for the
         description field.
        - Requirement: Mandatory
        - Format: 8bit
        - Ctype: B
        - Minimum: 0
        - Maximum: 1
        - Enumerations:
            - 1: Bluetooth SIG Assigned Numbers
    - Description:
        - InformativeText: The Description is an enumerated value from the organization identified by the Name Space field.
        - Requirement: Mandatory
        - Format: 16bit
        - Ctype: h
```

So If I'm not wrong Descriptor 2904 value should follow

  | FORMAT   | EXPONENT  |      UNIT      | NAMESPACE | DESCRIPTION |
  |-----------|------------- |------------|--------------|---------------|
  |   B (8bit)   |    b (sint8)    |  H (uint16)  |          B(8bit)   |       h(16bit)     |

e.g in binary

```
0b00000000 00000000 0000000000000000 00000000 0000000000000000
```
So basically bit masking +`struct` and / or `bleak_sigspec.SuperStruct` should do the trick. 👍🏼

dlech, 2022-10-03:
It sounds like including this feature `bleak-sigspec` instead of `bleak` might be a better fit since `bleak-sigspec` already has all of the decoding functions.

jackjansen, 2022-10-05:
Hmm, I see your point... But `bleak-sigspec` is only half of the solution (only for reading, not for writing), and it actually doesn't solve the bit of the puzzle that I'm most interested in (using 2904 introspection in stead of a database). The alternative is that I create an add-on package (`pybleak` or something) that just depends on both `bleak` and `bleak-sigspec`.