hbldh/bleak #1060: add BleakAdapter class

dlech, 2022-10-03:
This is a feature request to add a new `BleakAdapter` class to provide access to the system Bluetooth adapter.

This class should have a property for getting the adapter state (powered on, powered off, not present, not authorized) and allow registering a callback for when the adapter state changes. This could fix #320.

It could also have a method to get already connected Bluetooth devices. This could fix #367.

Additionally, it could provide an eaiser way to specify which adapter to use on Linux (instead of passing the adapter arg to each BleakScanner and BleakClient).

Creation of this object will necessarily be async due to the OS APIs that would need to be used, so this should have an async `default_adapter()` class (or static) method to get the adapter (currently all OSes other than Linux only allow one Bluetooth adapter) instead of an `__init__` method.

JPHutchins, 2022-12-01:
This seems to be a requirement in the case that Windows 10 has bonded with the device, I will be investigating.

bdraco, 2023-01-05:
Feel free to lift any code that might be useful from https://github.com/bluetooth-devices/bluetooth-adapters

JPHutchins, 2023-01-05:
Another resource for future reference https://github.com/IanSavchenko/BleLab.  I've confirmed that I can (only) access bonded devices with this tool.