hbldh/bleak #1065: Advertisements only seldom received/displayed

capiman, 2022-10-05:
* bleak version: 0.18.1
* Python version: Python 3.9.2
* Operating System: Linux test 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description

I have a device which is sending a lot advertisements. I said it to do in maximum speed.

test@test:~/bleak $ sudo hcitool lescan --duplicate
LE Scan ...
E4:5F:01:BA:05:2D (unknown)
E4:5F:01:BA:05:2D
E4:5F:01:BA:05:2D (unknown)
E4:5F:01:BA:05:2D
E4:5F:01:BA:05:2D (unknown)
E4:5F:01:BA:05:2D

I can see it advertising at a high speed (around 40 advertisements per second).

Then I use (from example):

test@test:~/bleak $ cat detection.py
"""
Detection callback w/ scanner
--------------
Example showing what is returned using the callback upon detection functionality
Updated on 2020-10-11 by bernstern <bernie@allthenticate.net>
"""

import asyncio
import logging
import sys

from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData

logger = logging.getLogger(__name__)

def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    logger.info(f"{device.address}: {advertisement_data}")

async def main(service_uuids):
    scanner = BleakScanner(simple_callback, service_uuids)

    while True:
        print("(re)starting scanner")
        await scanner.start()
        await asyncio.sleep(5.0)
        await scanner.stop()

if __name__ == "__main__":
    logging.basicConfig(
        level=logging.INFO,
        format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
    )
    service_uuids = sys.argv[1:]
    asyncio.run(main(service_uuids))

test@test:~/bleak $

It gives the following output:

$ python3 detection.py
(re)starting scanner
2022-10-05 13:16:40,784 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\x98\xc3\x01'})
2022-10-05 13:16:45,775 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\x98\xc3\x01'})
(re)starting scanner
2022-10-05 13:16:45,798 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\xb1\xc3\x01'})
2022-10-05 13:16:50,799 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\xb1\xc3\x01'})
(re)starting scanner
2022-10-05 13:16:50,819 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\xca\xc3\x01'})
2022-10-05 13:16:55,823 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\xca\xc3\x01'})
(re)starting scanner
2022-10-05 13:16:55,838 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00:\xe3\xc3\x01'})

It gives only seldom data, far away from 40 per second.
I assume the checked fields remain equal, but there shall be at least a counter inside, which increases every 200ms.
So I assume at least I should see an output every 200ms?

Could I tell Bleak to display every received advertisement?
Why is stop/restart needed? Just that duplicates are reported again? 
Or do I somehow need to tell BlueZ under Bleak to use something like lescan with --duplicate?

Or could it be that problem is on sender side? Advertisement data not correct? Or only sometimes correct?

dlech, 2022-10-05:
What is the output with `BLEAK_LOGGING=1`?

capiman, 2022-10-05:
```
pilabs@pilabs-iotgateway:~/bleak $ python3 detection.py
(re)starting scanner
2022-10-05 16:49:32,813 bleak.backends.bluezdbus.manager DEBUG: initial properties: {'/org/bluez': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.AgentManager1': {}, 'org.bluez.ProfileManager1': {}, 'org.bluez.HealthManager1': {}}, '/org/bluez/hci0': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'E4:5F:01:BA:08:45', 'AddressType': 'public', 'Name': 'pilabs-iotgateway', 'Alias': 'pilabs-iotgateway', 'Class': 0, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001801-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0537', 'Roles': ['central', 'peripheral']}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.GattManager1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 5, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name']}, 'org.bluez.Media1': {}, 'org.bluez.NetworkServer1': {}}}
2022-10-05 16:49:32,813 bleak.backends.bluezdbus.manager DEBUG: initial properties: {'/org/bluez': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.AgentManager1': {}, 'org.bluez.ProfileManager1': {}, 'org.bluez.HealthManager1': {}}, '/org/bluez/hci0': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'E4:5F:01:BA:08:45', 'AddressType': 'public', 'Name': 'pilabs-iotgateway', 'Alias': 'pilabs-iotgateway', 'Class': 0, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001801-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0537', 'Roles': ['central', 'peripheral']}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.GattManager1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 5, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name']}, 'org.bluez.Media1': {}, 'org.bluez.NetworkServer1': {}}}
2022-10-05 16:49:32,818 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
2022-10-05 16:49:32,818 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', True)>}, []]
2022-10-05 16:49:32,824 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_E4_5F_01_BA_05_2D', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', E4:5F:01:BA:05:2D)>, 'AddressType': <dbus_fast.signature.Variant ('s', public)>, 'Alias': <dbus_fast.signature.Variant ('s', E4-5F-01-BA-05-2D)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -70)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {65535: <dbus_fast.signature.Variant ('ay', bytearray(b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00\x02\xb0\xc3\x01'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-10-05 16:49:32,824 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_E4_5F_01_BA_05_2D', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', E4:5F:01:BA:05:2D)>, 'AddressType': <dbus_fast.signature.Variant ('s', public)>, 'Alias': <dbus_fast.signature.Variant ('s', E4-5F-01-BA-05-2D)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -70)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {65535: <dbus_fast.signature.Variant ('ay', bytearray(b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00\x02\xb0\xc3\x01'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
[... cut ...] /bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {65535: <dbus_fast.signature.Variant ('ay', bytearray(b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00\x03F\xc3\x01'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-10-05 16:50:02,996 __main__ INFO: E4:5F:01:BA:05:2D: AdvertisementData(manufacturer_data={65535: b'\xbe\xac\x13\xb7\xcbV\x81ZH\xec\xa0\x1f0!\xabI\xa1U\x00\x00\x03F\xc3\x01'})

```

dlech, 2022-10-05:
I'm not sure why each d-bus signal is being logged twice, but Bleak is only receiving a few update from BlueZ. So the next thing to check would be to log Bluetooth packets with Wireshark to see if the adapter is actually receiving advertisements more frequently or not.

capiman, 2022-10-05:
Would hcidump also work?

dlech, 2022-10-05:
yes

capiman, 2022-10-05:
It contains 4 files: 
1 bleak log file, suitable hcidump trace (slow)
1 lescan log output, suitable hcidump trace (fast)

I see hcidump also has only less activity, compared to fast case.

capiman, 2022-10-05:
Slow:
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
    
Fast:
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x00 (disabled)

Can I somehow tell bleak to not filter duplicates?

dlech, 2022-10-05:
Yes, you can add `bluez=dict(filters=dict(DuplicateData=True))` to the `BleakScanner` args. 

It looks like we need to fix the [docs](https://bleak.readthedocs.io/en/latest/backends/linux.html#bleak.backends.bluezdbus.scanner.BlueZDiscoveryFilters) to show the members of `BlueZDiscoveryFilters` dictionary.

dlech, 2022-10-05:
I think DuplicateData needs to be `False` instead of `True`.

https://github.com/bluez/bluez/blob/master/doc/adapter-api.txt#L112

capiman, 2022-10-05:
No matter if I use True or False, the parameter "filter duplicates" remains 0x01 (enabled).
There is a "filter duplicates" with value 0x00, when Python app is closed.

capiman, 2022-10-05:
test@test:~/bleak $ grep Duplicate detection.py
    scanner = BleakScanner(simple_callback, bluez=dict(filters=dict(DuplicateData=True)))
test@test:~/bleak $ python3 detection.py
{'Transport': <dbus_fast.signature.Variant ('s', le)>, 'DuplicateData': <dbus_fast.signature.Variant ('b', True)>}
(re)starting scanner

When I change to False, printed text also changes to False.

capiman, 2022-10-05:
Could this be the code on the "other" side?

https://github.com/bluez/bluez/blob/61f4f2895882a19c402e93ef4ba7bb6b802015a4/client/main.c#L1520

dlech, 2022-10-05:
No, that is the code for `blueoothctl`. The D-Bus API for the adapter is handled in `src/adapter.c`.

dlech, 2022-10-05:
Could be related to #235

capiman, 2022-10-05:
I can confirm, that the (bad) workaround from

https://github.com/hbldh/bleak/issues/235#issuecomment-708670983

is working also on my side. I get fast advertisement messages for a short time, till it gets slow again.

dlech, 2022-10-05:
I found https://github.com/torvalds/linux/commit/abfeea476c68afea54c9c050a2d3b19d5d2ee873 that indicates the behavoir was changed in Linux kernel 5.17 to not filter duplicates by default. So maybe upgrading the kernel is a possibility?

capiman, 2022-10-05:
We are using a Raspberry Pi. Upgrading the kernel is not so easy. We get it as binary only.
So you think, trying to compile and install a newer BlueZ would not help, if a change in the kernel was needed.

I think I must sleep over it (already late evening here in Germany) and think tomorrow how it can be solved.

I would like to send you a BIG THANK YOU for all your help!

dlech, 2022-10-05:
I tried it on a newer kernel (5.17 and 5.19) and it doesn't seem to have changed. I was able to come up with a different workaround though. If you enable experimental features in `bluetoothd` and add an advertisement watcher (currently this is how "passive" advertising is implemented in Bleak) then active scanning will disable the duplicate filter when started.

closed 2025-11-09 by github-actions[bot]