hbldh/bleak #1075: bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error)

ujur007, 2022-10-09:
* bleak version: latest, installed with pip
* Python version:Python 3.8.10
* Operating System:  Linux Mint-64bit
* BlueZ version (`bluetoothctl -v`) in case of Linux: bluetoothctl: 5.53

### Description

I am trying a python script to read & write BLE characteristic values. The issue that I am facing is that while writing to the characteristics. The following output from the terminal.

What I was trying was to write to the specific characteristic with command(write_gatt_char) such as the following. I also tried to change to int:handle instead of string:UUID. I am able to read the characteristic. I am sure that the characteristic is writable, as I have tried it with mobile application.

await client.write_gatt_char("c66f3301-33b3-4687-850a-d52b0d5d1e3e", b'5')

Traceback (most recent call last):
  File "testApp.py", line 64, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "testApp.py", line 46, in main
    print(await client.write_gatt_char("c66f3301-33b3-4687-850a-d52b0d5d1e3e", b'5'))
  File "/home/ujjval/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 720, in write_gatt_char
    assert_reply(reply)
  File "/home/ujjval/.local/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error)

### Logs
I am new to Wireshark and update the log in this issue.

ujur007, 2022-10-11:
I tried to find the solution and what I found was some suggestions.

What I have tried is to downgrade/upgrade the bluez versions but the error still remains the same.
There are some threads also suggesting to increase the packet size on the peripheral device. Which I think should not be an issue, in my case.

dlech, 2022-10-11:
Logging Bluetooth packets is really the best way to understand what is going on.

ujur007, 2022-10-11:
The write data seems to be not showing up in the traffic on wireshark itself. Here is the Wireshark log for bluetooth0 interface.

[Log_ble.log](https://github.com/hbldh/bleak/files/9758651/Log_ble.log)

dlech, 2022-10-11:
Everything is from host to controller, so the device is not responding. It appears to be a problem with the device.

ujur007, 2022-10-11:
Ok. I will check with other BLE devices perhaps. Currantly, I have Nimble stack port for esp-idf running on ESP32-pico.

ujur007, 2023-01-10:
I have this issue still present in my application. Actually I am simultaneously writing to three characteristics. 
What I see happening is that only first write to the characteristic is happening while the latter two are indeed failing.
I have the wireshark log for more information.
[wireshark_bluetooth-monitor_20230110174326_mxdjgs.pdf](https://github.com/hbldh/bleak/files/10382722/wireshark_bluetooth-monitor_20230110174326_mxdjgs.pdf)

await self.client.write_gatt_char(first, ([0xFF]))
await self.client.write_gatt_char(second, ([0x00]))
await self.client.write_gatt_char(third, ([0x00]))
        
print(await self.client.read_gatt_char(first))
print(await self.client.read_gatt_char(second))
print(await self.client.read_gatt_char(third))`
`

ujur007, 2023-01-11:
okey, seems the error was in my device code and not with Bleak. So now it works. Thanks..!

closed 2023-01-11 by dlech

smalik007, 2023-12-08:
@ujur007, what was the mistake in your code? I am experiencing the same issue when I stress the BLE notification packets too frequently, it suddenly disconnects with device after notifying for few seconds, the issue however is not there when I reduce the messages' frequency.  @dlech is there an upper limit in terms of timing we should be aware of, not go beyond to have stable connection?

dlech, 2023-12-08:
> is there an upper limit in terms of timing we should be aware of, not go beyond to have stable connection?

Not a hard limit. It will depend on the device you are connecting to and the Bluetooth adapter you are using.

smalik007, 2023-12-11:
The exception is thrown on from the bleak module, which I am running on a Rasberry Pi 2 board which connects to a low energy embedded chip. I suspect the problem more on the Rasberry Pi2 board.