hbldh/bleak #1076: micro:bit disconnected after pairing (In JustWorks setting) and cannot connect to micro:bit again

BitaMatt, 2022-10-11:
* bleak version: 0.18.1
* Python version: 3.10.7
* Operating System: Windows10

### Description

I want to connect to micro:bit using bleak's api in windows 10 OS. And use the

>  BleakClient.write_gatt_char 

to  send messge to the micro:bit Event Service (UUID: E95D93AF-251D-470A-A062-FA1922DFA9A8) (Which have Characteristic: 'E95D5404-251D-470A-A062-FA1922DFA9A8' to trigger the event I've been written and downloaded to micro:bit). Unfortunately I found out that I must paired with microbit in order to send message from my computer. So I used:
`
async with BleakClient(device,disconnected_callback=disconnected_callback,timeout=15.0) as client:
        print("Pairing Client")
        await client.pair()
        ....
        codes that use to send message
`
to pair with micro:bit.  I found out that micro:bit is successfully connect and pair with my computer. But after pairing finished, disconnected_callback called out (Though I saw it is still connected On Windows Bluetooth setting) 
[image]
After that I add connect command after paired, and it always failed with **TimeoutError**:
`   
 async with BleakClient(device,disconnected_callback=disconnected_callback,timeout=15.0) as client:
        print("Pairing Client")
        await pairClient(client)
        await asyncio.sleep(10)
        print("Reconnect if necessary")
        print(f'is connect? {client.is_connected}')
        if not client.is_connected:
            await client.connect()
`

### What I Did
This is an example code that micro:bit used
https://makecode.microbit.org/_1yFbXM6TyPxT

This is an brief python code (minimal code that can run) that I used to connect and pair with micro:bit :
```
import time
import asyncio
import logging
from bleak import BleakClient, BLEDevice, BleakGATTCharacteristic
from bleak import BleakScanner

#device pair tag
deviceChar = "zipeg"

disconnected_event = asyncio.Event()

def disconnected_callback(client):
    print("Disconnected callback called!")
    disconnected_event.set()

async def scanWithNamePart(wanted_name_part):
    '''
    Find device by specific string
    :param wanted_name_part:
    :return:
    '''
    device = await BleakScanner.find_device_by_filter(
        lambda d,ad: (wanted_name_part.lower() in d.name or wanted_name_part.lower() in d.name.lower()) if d.name is not None else False
    )
    print(device)
    device_data = device
    return device_data

async def mainCheck():
    #main function
    print("Discover device by name:")
    device: BLEDevice = await scanWithNamePart(deviceChar)
[... 234 lines cut ...]
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\**...**\microbit_control.py", line 271, in <module>
    asyncio.run(mainCheck())
  File "C:\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\**...**\microbit_control.py", line 221, in mainCheck
    await client.connect()
  File "C:\Python310\lib\site-packages\bleak\__init__.py", line 392, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 330, in connect
    async with async_timeout.timeout(timeout):
  File "C:\Python310\lib\site-packages\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "C:\Python310\lib\site-packages\async_timeout\__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
`

I almost done with using computer to control micro:bit with BluetoothLE! I would be very happy if this problem is solved!

dlech, 2022-10-11:
Can you share logs of the Bluetooth packets captured with Wireshark as described in the troubleshooting page of the Bleak docs?

BitaMatt, 2022-10-12:
> 

OK I will try it

BitaMatt, 2022-10-12:
> Can you share logs of the Bluetooth packets captured with Wireshark as described in the troubleshooting page of the Bleak docs?

I followed the instruction of troubleshooting page and get this log file. 
[wireshark_log.zip](https://github.com/hbldh/bleak/files/9761203/wireshark_log.zip)

dlech, 2022-10-12:
It looks like the MicroBit keeps sending the "Service Changed" indication, so Windows keeps re-enumerating the services. So I suspect a bug in the MicroBit firmware.

BitaMatt, 2022-10-13:
> 

roger that , I will try to use another micro:bit for code testing

BitaMatt, 2022-10-13:
> It looks like the MicroBit keeps sending the "Service Changed" indication, so Windows keeps re-enumerating the services. So I suspect a bug in the MicroBit firmware.

I finally find out the solution, just cancel the JustWork pairing mode and than I can control micro:bit without pairing with it! Thank you for your help!

closed 2022-10-13 by BitaMatt