hbldh/bleak #1077: The ble HID services cannot be enumerated

SOYOJOE, 2022-10-12:
* bleak version: 0.18.1
* Python version: 3.96
* Operating System: win10 21H2
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

The ble HID services cannot be enumerated

### What I Did

using get_services.py example

```
 & C:/Users/Admin/AppData/Local/Programs/Python/Python39/python.exe c:/Users/Admin/Desktop/bleak-0.18.1/bleak-0.18.1/examples/get_services.py

```
### Logs
python out put

Services:
00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
00001801-0000-1000-8000-00805f9b34fb (Handle: 8): Generic Attribute Profile
0000180a-0000-1000-8000-00805f9b34fb (Handle: 12): Device Information
0000180f-0000-1000-8000-00805f9b34fb (Handle: 72): Battery Service
00010203-0405-0607-0809-0a0b0c0d1912 (Handle: 76): Unknown

in NRF connect APP

[image]

SOYOJOE, 2022-10-12:
I found the logs about this ,so is there no any solution?
[image]

dlech, 2022-10-12:
Correct. You will need to use Windows HID APIs instead.

closed 2022-11-07 by dlech