hbldh/bleak #1079: Custom service not found

profff, 2022-10-13:
bleak version:
bleak 0.18.1
bleak-winrt 1.2.0
Python version:
Python 3.10.6
Operating System:
Windows 10 Professionnal x64

### Description
when listing service to found the one I need it don get listed 
but service exists anyway as it's well seen by any android thrd party BLE tool (NRFconnect) 

## What I Did
here's the source code we assume `dev` is device object returned by bleakscanner
```
print("connecting to : ",dev)
async def main(dev):
    async with BleakClient(dev.address) as client:
        print("Services:")
        for service in client.services:
            print(service)
            for char in service.characteristics:
            	print("  ",char)
asyncio.run(main(dev))
```
here the result 
```
C:\src\btserial> python .\btserial.py
connecting to :  58:BF:09:30:2F:FE: INF-BS
Services:
00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
   00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name
   00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance
   00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters
   00002aa6-0000-1000-8000-00805f9b34fb (Handle: 8): Central Address Resolution
   00002ac9-0000-1000-8000-00805f9b34fb (Handle: 10): Resolvable Private Address Only
00001801-0000-1000-8000-00805f9b34fb (Handle: 12): Generic Attribute Profile
0000180a-0000-1000-8000-00805f9b34fb (Handle: 13): Device Information
   00002a23-0000-1000-8000-00805f9b34fb (Handle: 14): System ID
   00002a24-0000-1000-8000-00805f9b34fb (Handle: 16): Model Number String
   00002a25-0000-1000-8000-00805f9b34fb (Handle: 18): Serial Number String
   00002a26-0000-1000-8000-00805f9b34fb (Handle: 20): Firmware Revision String
   00002a27-0000-1000-8000-00805f9b34fb (Handle: 22): Hardware Revision String
   00002a28-0000-1000-8000-00805f9b34fb (Handle: 24): Software Revision String
   00002a29-0000-1000-8000-00805f9b34fb (Handle: 26): Manufacturer Name String
   00002a2a-0000-1000-8000-00805f9b34fb (Handle: 28): IEEE 11073-20601 Regulatory Cert. Data List
   00002a50-0000-1000-8000-00805f9b34fb (Handle: 30): PnP ID
00001809-0000-1000-8000-00805f9b34fb (Handle: 32): Health Thermometer
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 33): PCB Temperature probe
   00002b18-0000-1000-8000-00805f9b34fb (Handle: 36): Ref Voltage
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 39): Sensor Raw OUT
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 42): Sensor int temp
0000180f-0000-1000-8000-00805f9b34fb (Handle: 45): Battery Service
   00002a19-0000-1000-8000-00805f9b34fb (Handle: 46): state of chargeCurrent draw
   00002b18-0000-1000-8000-00805f9b34fb (Handle: 49): Voltage
   00002aee-0000-1000-8000-00805f9b34fb (Handle: 52): Current draw
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 55): Temperature
00000001-0000-1000-8000-00805f9b34fb (Handle: 58): SDP
   00000002-0000-1000-8000-00805f9b34fb (Handle: 59): Vendor specific
   00000003-0000-1000-8000-00805f9b34fb (Handle: 62): RFCOMM
```
and here you can see screen shot of NRF Connect to same dev

[image]

[image]

as you can see at python script name I try to do something similar to Nordic UART Service and I whant to use bleek to make a small terminal app on pc

### Logs

I still have isue with BTP :(

profff, 2022-10-13:
same issue with a 128 bit UUID

profff, 2022-10-13:
```
connecting to :  58:BF:09:30:2F:FE: INF-BS
2022-10-13 17:09:55,187 bleak.backends.winrt.scanner DEBUG: Received 0F:80:B9:43:DD:3D: Unknown.
2022-10-13 17:09:55,200 bleak.backends.winrt.scanner DEBUG: Received 58:BF:09:30:2F:FE: INF-BS.
2022-10-13 17:09:55,202 bleak.backends.winrt.scanner DEBUG: 2 devices found. Watcher status: 3.
2022-10-13 17:09:55,202 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ 58:BF:09:30:2F:FE
2022-10-13 17:09:56,149 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x0000023FB3B469B0>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
2022-10-13 17:09:56,149 bleak.backends.winrt.client DEBUG: Get Services...
2022-10-13 17:09:56,352 bleak.backends.winrt.client INFO: Services resolved for BleakClientWinRT (58:BF:09:30:2F:FE)
Services:
00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
   00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name
   00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance
   00002a04-0000-1000-8000-00805f9b34fb (Handle: 6): Peripheral Preferred Connection Parameters
   00002aa6-0000-1000-8000-00805f9b34fb (Handle: 8): Central Address Resolution
   00002ac9-0000-1000-8000-00805f9b34fb (Handle: 10): Resolvable Private Address Only
00001801-0000-1000-8000-00805f9b34fb (Handle: 12): Generic Attribute Profile
0000180a-0000-1000-8000-00805f9b34fb (Handle: 13): Device Information
   00002a23-0000-1000-8000-00805f9b34fb (Handle: 14): System ID
   00002a24-0000-1000-8000-00805f9b34fb (Handle: 16): Model Number String
   00002a25-0000-1000-8000-00805f9b34fb (Handle: 18): Serial Number String
   00002a26-0000-1000-8000-00805f9b34fb (Handle: 20): Firmware Revision String
   00002a27-0000-1000-8000-00805f9b34fb (Handle: 22): Hardware Revision String
   00002a28-0000-1000-8000-00805f9b34fb (Handle: 24): Software Revision String
   00002a29-0000-1000-8000-00805f9b34fb (Handle: 26): Manufacturer Name String
   00002a2a-0000-1000-8000-00805f9b34fb (Handle: 28): IEEE 11073-20601 Regulatory Cert. Data List
   00002a50-0000-1000-8000-00805f9b34fb (Handle: 30): PnP ID
00001809-0000-1000-8000-00805f9b34fb (Handle: 32): Health Thermometer
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 33): PCB Temperature probe
   00002b18-0000-1000-8000-00805f9b34fb (Handle: 36): Ref Voltage
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 39): Sensor Raw OUT
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 42): Sensor int temp
0000180f-0000-1000-8000-00805f9b34fb (Handle: 45): Battery Service
   00002a19-0000-1000-8000-00805f9b34fb (Handle: 46): state of chargeCurrent draw
   00002b18-0000-1000-8000-00805f9b34fb (Handle: 49): Voltage
   00002aee-0000-1000-8000-00805f9b34fb (Handle: 52): Current draw
   00002a6e-0000-1000-8000-00805f9b34fb (Handle: 55): Temperature
00000001-0000-1000-8000-00805f9b34fb (Handle: 58): SDP
   00000002-0000-1000-8000-00805f9b34fb (Handle: 59): Vendor specific
   00000003-0000-1000-8000-00805f9b34fb (Handle: 62): RFCOMM
2022-10-13 17:09:56,352 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...
2022-10-13 17:09:56,505 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x0000023FB3B47410>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
```

with debug lines
nothing looks bad ... :(

dlech, 2022-10-13:
It's hard to say what is going on without seeing a Bluetooth packet log. Maybe a GATT service cache issue (i.e. try `use_cached_services=False`)?

dlech, 2022-10-13:
> like that ?

No, like this:

```python
async with BleakClient(dev.address,winrt=dict(use_cached_services=False)) as client:
```

> it doesn't fix issue but with un binding and rebinding the device under windows it looks to work

Yes, this would have the same effect.

profff, 2022-10-14:
ok thanks

profff, 2022-10-24:
heres what 've done 
```
print("connecting to : ",TexDev)
		#async with BleakClient(TexDev.address) as client :
		async with BleakClient(TexDev.address,winrt=dict(use_cached_services=False)) as client:
			for service in client.services:
				if get16bitsUUID(service.uuid)==NUS_SERV_UUID :
					for ch in service.characteristics :
						print(ch)
```
and here's the result (without `use_cached_services` it works but if something change in caracteristics since bind it's not refreshed ) 

```
PS C:\src\btserial> python.exe .\btserial.py
connecting to :  5A:0F:E6:F0:F6:65: INF-BS
Traceback (most recent call last):
  File "C:\src\btserial\btserial.py", line 98, in <module>
    searchDeviceByName(DEVICE_NAME)
  File "C:\src\btserial\btserial.py", line 97, in searchDeviceByName
    asyncio.run(run())
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\src\btserial\btserial.py", line 63, in run
    async with BleakClient(TexDev.address,winrt=dict(use_cached_services=False)) as client:
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 354, in __aenter__
    await self.connect()
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 392, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 339, in connect
    await self.get_services()
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 531, in get_services
    await self._requester.get_gatt_services_async(*args),
OSError: [WinError -2147418113] Défaillance irrémédiable
```

dlech, 2022-10-24:
> but if something change in characteristics since bind it's not refreshed

If the attribute database changes, then the device needs to implement the GATT Database Hash characteristic or Services Changed characteristic to notify Windows that the services have changed. 

You could also try logging Bluetooth packets with Wireshark to try to figure out why it is not working with `use_cached_services=False`.

profff, 2022-10-27:
ok I finaly get wireshark working with Ble 
here's what happens when setting `use_cached_service=False`
heres the code : 
```
print("connecting to : ",TexDev)
		#async with BleakClient(TexDev.address) as client :
		async with BleakClient(TexDev.address,winrt=dict(use_cached_services=False)) as client:
			for service in client.services:
				print(service)
				for ch in service.characteristics :
					if 'notify' in ch.properties:
						print("  ",ch,ch.properties)
```
heres shell result 
```
PS C:\src\btserial> python.exe .\btserial.py
connecting to :  6D:AF:BF:A2:0C:F0: INF-BS
Traceback (most recent call last):
  File "C:\src\btserial\btserial.py", line 155, in <module>
    searchDeviceByName(DEVICE_NAME_PREFIX)
  File "C:\src\btserial\btserial.py", line 154, in searchDeviceByName
    asyncio.run(run())
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\src\btserial\btserial.py", line 127, in run
    async with BleakClient(TexDev.address,winrt=dict(use_cached_services=False)) as client:
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 354, in __aenter__
    await self.connect()
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 392, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 339, in connect
    await self.get_services()
  File "C:\Users\o.gaste\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 531, in get_services
    await self._requester.get_gatt_services_async(*args),
OSError: [WinError -2147418113] Défaillance irrémédiable
```
and I'm attaching the wireshark capture
[noncached bleak error.zip](https://github.com/hbldh/bleak/files/9876926/noncached.bleak.error.zip)

dlech, 2022-10-27:
It looks like the Bluetooth chip is still scanning after connecting to the remote device and the remote device is still advertising during this time, neither of which should be happening.

Telling your device to stop advertising after connecting could be a good first step.

JohnsonChouCrestDiving, 2022-11-09:
Hi @dlech,
I have same issue here, when I upgrade to bleak-0.19.4, [WinError -2147024874] start to occur.
In bleak-0.17.0, I got occasionally success, about 3 times out of 10, the other fail will be timeout.
In bleak-0.19.4, fail case **Red mark** ATT command start to different from success case.

I need help

## Error 
bleak version:
bleak 0.19.4
bleak-winrt 1.2.0
Python version:
Python 3.9.6
Operating System:
Windows 10 Home, Version 10.0.19044 Build 19044
```sh
DEBUG:bleak.backends.winrt.client:Connecting to BLE device @ C2:5E:51:CF:C5:2E 
DEBUG:bleak.backends.winrt.client:Get Services...
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000146E5CB8270>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
DEBUG:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event
DEBUG:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event
DEBUG:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='Task-1' coro=<run() done, defined at d:\Git\python\debug_tool\crest_debugger\interface\ble.py:144> exception=PermissionError(13, 'The device does not recognize the command', None, -2147024874, None)>
Traceback (most recent call last):
  File "d:\Git\python\debug_tool\crest_debugger\interface\ble.py", line 155, in run
    async with BleakClient(address,winrt=dict(use_cached_services=False)) as client:
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\__init__.py", line 433, in __aenter__
    await self.connect()
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\backends\winrt\client.py", line 380, in connect
    await asyncio.gather(wait_connect_task, wait_get_services_task)
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\backends\winrt\client.py", line 627, in get_services
    await service.get_characteristics_async(*args),
PermissionError: [WinError -2147024874] The device does not recognize the command
```
[image]

## Success
bleak version:
bleak 0.17.0
bleak-winrt 1.2.0
Python version:
Python 3.9.6
Operating System:
Windows 10 Home, Version 10.0.19044 Build 19044
[image]

dlech, 2022-11-09:
> I have same issue here, when I upgrade to bleak-0.19.4, [WinError -2147024874] start to occur.

This is not the same error, so I started a new issue for this in https://github.com/hbldh/bleak/issues/1123

closed 2025-10-26 by github-actions[bot]