hbldh/bleak #1084: Sometimes write characteristic works sometimes not

renatotu, 2022-10-19:
* bleak version: 0.19.0
* Python version:3.10.7
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.

I want to test a Bluetooth (BLE) board, by sending a command in the write characteristic and then reading in another characteristic.
I already have something working with a mobile app, so the python script would be just to have something faster and connected to PC.
With the script bellow sometimes it works, sometimes it doesn´t send the data, it seems some garbage remain from another session, I don´t know exactly.
As I'm not a python expert I also don´t know if everything is correct.

The logs seems the same when it works and when it doesn´t, at least until the part that it will send the data.

### What I Did
`import asyncio
import time
from bleak import BleakClient
from bleak import BleakScanner
from bleak.backends.characteristic import BleakGATTCharacteristic

address = "80:1F:12:B7:CC:45"
MODEL_NBR_UUID = "00002a24-0000-1000-8000-00805f9b34fb"

UART_SERVICE_UUID = "49535343-FE7D-4AE5-8FA9-9FAFD205E455"
UART_RX_CHAR_UUID = "49535343-8841-43F4-A8D4-ECBE34729BB3"
UART_TX_CHAR_UUID = "49535343-1E4D-4BD9-BA61-23C647249616"

list = []
received_data = bytearray(6)
expected_response = bytearray([0xA3, 0x00, 0x00, 0x00, 0x3C, 0x33])
        
async def main():

    def handle_rx(_: BleakGATTCharacteristic, data: bytearray):
        #print("received:", data)
        received_data[:] = data

    #def notification_handler(handle: int, data: bytes) -> None:
        #logger.info(f'Received response at {handle=}: {hexlify(data, ":")!r}')
        #print(data)
        
      
    devices = await BleakScanner.discover()
    if devices:
        print("\nList of Devices:\n-------------------------------------------------\n")
    else:
        print("\nNo devices were found\n")
        sys.exit()
    for index, item in enumerate(devices):
        print(index, item)
        list.append(item)
    option = input('\n-------------------------------------------------\nWhat is the device index?\n')
#    await operations(d)   #TODO: IDENTIFY AUTOMATICALLY THE BLEBOARD
    address = list[int(option)]

    BleakClient.pair(address)
    client = BleakClient(address)
 
    await client.connect(timeout=60)
    print("BLE is connected:",{client.is_connected})
    #async with BleakClient(address) as client:

    #for service in client.services:
    #    for char in service.characteristics:
    #        if "notify" in char.properties:
    #            await client.start_notify(char, notification_handler)
[... 388 lines cut ...]
2022-10-19 14:08:19,519 bleak.backends.winrt.client INFO: Services resolved for BleakClientWinRT (80:1F:12:B7:CC:45)
BLE is connected: {True}
2022-10-19 14:08:29,861 bleak.backends.winrt.client DEBUG: Read Characteristic 0013 : bytearray(b'RN487x')
-------------------------------------------------

Test Result:

Failed!!!

BLE response not received, Something went wrong, try again!!!
-------------------------------------------------

2022-10-19 14:09:10,635 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...
2022-10-19 14:09:13,738 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 23
2022-10-19 14:09:13,739 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000197123F73B0>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED

C:\Projects\BTDHWTest\Python>pause`

Include any relevant logs here.

Logs are essential to understand what is going on behind the scenes.

See https://bleak.readthedocs.io/en/latest/troubleshooting.html for information on how to collect debug logs.

If you receive an OS error (`WinError`, `BleakDBusError`, `NSError`, etc.), Wireshark logs of Bluetooth packets are required to understand the issue!

dlech, 2022-10-19:
For starters, `time.sleep()` is blocking, you need to use `await asyncio.sleep()` instead. Otherwise async events can't be handled.

renatotu, 2022-10-19:
@dlech thanks for updating my post
Just changed to asyncio.sleep, same result

closed 2022-10-19 by renatotu

dlech, 2022-10-19:
I would try logging with wireshark to see if the device is actually sending a response. My guess is that is is not.