hbldh/bleak #1085: AttributeError: 'android.bluetooth.BluetoothDevice' object has no attribute 'getRssi'

icore-clarience, 2022-10-19:
* bleak version: 0.19.9
* Python version: 3.10
* Operating System: Android 12

### Description
Previously functional kivy Android app using bleak 0.18.1 doesn't work in a new build environment with bleak 0.19.0

### What I Did

try:
				
	device = await BleakScanner.find_device_by_filter(lambda device, advertising_data: device.name == self.connect_to)
						
	if device is not None:
						
		async with BleakClient(device,disconnection_callback) as client:
							
			while not self.client.is_connected:
				await asyncio.sleep(0.1)
								
			connection_callback(self.connect_to)

			await client.start_notify(UART_RX_UUID, packet_recieved_callback)

			while client.is_connected:
				await asyncio.sleep(0.1)

except BleakError as e:
				
	report_error(e)

**see traceback in Logs section below

### Logs
adb log is very very long, here is a specific segment, please let me know if there is anything specific I could find for you

10-19 11:12:28.787 11726 11833 I python  :  --- Logging error ---
10-19 11:12:28.787 11726 11833 I python  :  Traceback (most recent call last):
10-19 11:12:28.787 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/events.py", line 80, in _run
10-19 11:12:28.787 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/rrdemo/arm64-v8a/bleak/backends/p4android/scanner.py", line 259, in _handle_scan_result
10-19 11:12:28.787 11726 11833 I python  :  AttributeError: 'android.bluetooth.BluetoothDevice' object has no attribute 'getRssi'
10-19 11:12:28.787 11726 11833 I python  :  
10-19 11:12:28.787 11726 11833 I python  :  During handling of the above exception, another exception occurred:
10-19 11:12:28.787 11726 11833 I python  :  
10-19 11:12:28.787 11726 11833 I python  :  Traceback (most recent call last):
10-19 11:12:28.787 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 1083, in emit
10-19 11:12:28.787 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 927, in format
10-19 11:12:28.787 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/rrdemo/arm64-v8a/kivy/logger.py", line 291, in format
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 172, in deepcopy
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 270, in _reconstruct
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 146, in deepcopy
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 230, in _deepcopy_dict
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 146, in deepcopy
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 210, in _deepcopy_tuple
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 210, in <listcomp>
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/copy.py", line 161, in deepcopy
10-19 11:12:28.788 11726 11833 I python  :  TypeError: cannot pickle 'traceback' object
10-19 11:12:28.788 11726 11833 I python  :  Call stack:
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/app/main.py", line 131, in <module>
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/runners.py", line 44, in run
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 629, in run_until_complete
10-19 11:12:28.788 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 596, in run_forever
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 1890, in _run_once
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/events.py", line 94, in _run
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 1764, in call_exception_handler
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 1738, in default_exception_handler
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 1475, in error
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 1589, in _log
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 1599, in handle
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 1661, in callHandlers
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 952, in handle
10-19 11:12:28.789 11726 11833 I python  :    File "/home/icore/rr-kivy/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/logging/__init__.py", line 1091, in emit
10-19 11:12:28.789 11726 11833 I python  :  Message: 'Exception in callback BleakScannerP4Android._handle_scan_result(<android.blue...0x7b42ef5b50>>)\nhandle: <Handle BleakScannerP4Android._handle_scan_result(<android.blue...0x7b42ef5b50>>)>'
10-19 11:12:28.789 11726 11833 I python  :  Arguments: ()

dlech, 2022-10-19:
Can you please try #1086 to see if it fixes the problem?

icore-clarience, 2022-10-24:
Thanks for the quick response, unfortunately I couldn't get back to this issue until today. 

I tried to test your fix but have run into a different issue: I use Buildozer to compile for Android, and every time I build the app it appears to overwrite the modification to the scanner.py file by pulling down a fresh copy of bleak version 0.19.0 via pip.

I haven't found an easy workaround for that, so for now I've just rolled back bleak to 0.18.1. But if you know of any resources for how to work around this I'll gladly give it another try.

dlech, 2022-10-24:
Can you tell Buildozer to use `https://github.com/hbldh/bleak/archive/refs/heads/android-fix-rssi.zip` instead of a version?

icore-clarience, 2022-10-25:
That does appear to work. I replaced "bleak" in the requirements with that URL and Buildozer installed bleak 0.20.0a1, and the bluetooth now works in my app.

dlech, 2022-10-25:
Great, thanks for testing!

PR #1086 (2022-10-25): backends/p4android/scanner: fix crash when getting rssi
In #1047, the call to getRssi() was broken by using the wrong object.

This fixes the mistake.

Fixes #1085.
commit b48e4cf0 (2022-10-25): backends/p4android/scanner: fix crash when getting rssi

In #1047, the call to getRssi() was broken by using the wrong object.

This fixes the mistake.

Fixes #1085.
PR #1097 (2022-10-29): Release/0.19.1

Fixed
-----
* Fixed crash in Android backend introduced in v0.19.0. Fixes #1085.
* Fixed service discovery blocking forever if device disconnects in BlueZ backend. Merged #1092.
* Fixed ``AttributeError`` crash when scanning on Windows builds < 19041. Fixes #1094.
closed 2022-10-25 by dlech