hbldh/bleak #1093: HaBleakScannerWrapper' has no attribute 'find_device_by_address'

and7ey, 2022-10-27:
* bleak version: (don't know how to check at Home Assistant)
* Python version: 3
* Operating System: [Home Assistant OS](https://github.com/home-assistant/operating-system) core-2022.10.5
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.65

### Description

I am debugging https://github.com/roslovets/SP110E-HASS Home Assistant (HA) integration. 
It uses SP110E library, Bleak is used there - https://github.com/roslovets/SP110E/blob/main/sp110e/driver.py.

`devices = await bleak_discover()` there shows me the devices list, including my device.
But when I try to connect with particular device with `device = await BleakScanner.find_device_by_address(mac_address, timeout=timeout)`, I am getting the following error:

```
Traceback (most recent call last):
File "/config/sp107e/controller.py", line 213, in _connect_with_retries
await self._driver.connect(self._mac, timeout=self._timeout)
File "/config/sp107e/driver.py", line 53, in connect
device = await BleakScanner.find_device_by_address(mac_address, timeout=10.0)
AttributeError: type object 'HaBleakScannerWrapper' has no attribute 'find_device_by_address'
```
(line numbers could be different from the URL above since I've added logging to my local version of the library)

dlech, 2022-10-27:
`HaBleakScannerWrapper` is not part of Bleak. I would suggest asking about it in the Home Assistant support channels.

and7ey, 2022-10-29:
Sorry, didn't notice Ha there.

closed 2022-10-29 by and7ey