hbldh/bleak #1099: data to remote device won't send on Linux?

David-Afik, 2022-10-30:
Hello, 
I'm trying to use the same code on 2 devices:
Windows 10 with Python 3.10
PI zero running Raspbian GNU/Linux 10 (buster) with Python 3.7.3
the code is simple 
reading 1 CHR 
and sending data "0" \ "1" to another CHAR 

on windows - everything is working great 
on Linux - the reading part is working , but when I try to send data - it doesn't send me (but I don't get any errors)

this is the code:

```
import asyncio
from bleak import BleakScanner, BleakClient

MAC_Espruino = "E1:75:AE:71:3F:88"
temperature_UUID = "3e440001-f5bb-357d-719d-179272e4d4d9"
Led_Char = '3e440002-f5bb-357d-719d-179272e4d4d9'
temperature_Char = "3e440008-f5bb-357d-719d-179272e4d4d9"

async def main():
    New_Led_Status = input('''
     Enter new led status:
1 - Led on ,  will turn on the  relay 
0 - led off , will turn off the relay 
'''
                           )

    DeviceOnline = True

    if DeviceOnline:
        try:
            async with BleakClient(MAC_Espruino) as client:
                Espruino_Temperature = await client.read_gatt_char(temperature_Char)  # get Unit Temperature
                print('Unit temp is : ', int.from_bytes(Espruino_Temperature, byteorder='big'))
                try:
                    value = await client.read_gatt_char(Led_Char)  # read old value
                    print("Pre-Write Value: {0}".format(value))
                    if New_Led_Status == "1":
                        await client.write_gatt_char(Led_Char, b'\01')  # send new value
                    elif New_Led_Status == "0":
                        await client.write_gatt_char(Led_Char, b'\00')  # send new value
                    else:
                        await client.write_gatt_char(Led_Char, b'\07')  # send new value
                    value = await client.read_gatt_char(Led_Char)
                    print("Post-Write Value: {0}".format(value))
                # print('-----', whatisit)
                except Exception as e:
                    print(e)
                else:
                    await client.disconnect()
                    print('finish')
        except asyncio.exceptions.TimeoutError as e1:
            print(e1)
            print('TIME OUT ERROR?')
        except Exception as e:
            print(e)
    else:
        print("Espruino not on line! ")

asyncio.run(main())
```

on windows this is what I'm getting:
```
     Enter new led status:
1 - Led on ,  will turn on the  relay 
0 - led off , will turn off the relay 
1
Unit temp is :  29
Pre-Write Value: bytearray(b'\x00')
Post-Write Value: bytearray(b'\x01')
finish
```

on LInux this is the what I'm getting:
```
     Enter new led status:
1 - Led on ,  will turn on the  relay
0 - led off , will turn off the relay
0
Unit temp is :  30
Pre-Write Value: bytearray(b'\x01')
Post-Write Value: bytearray(b'\x01')
finish
```
what is wrong?
as you can see the reading is working on both OS 

Thanks ,

dlech, 2022-10-30:
Bleak does write without response by default. If the device does not support it, you need to add a `response=True` argument.

dlech, 2023-03-17:
Closing as hopefully solved by #1236.

closed 2023-03-17 by dlech